prenticedavid/MCUFRIEND_kbv #1: Upgrade to Teensy

Thebobcachelot, 2016-08-24:
Hi David, I was working with your library with my lcd screen , working verywell on my UNO . I was trying to make it work on my Teensy 3.1, but There is a lot of thing that seems to be problematic, i am not even able to compile in the controller, do you have an idea of what should be modified to get it work with it except the pinout? 

I'm using mcufriend_kbv.h

thank you

prenticedavid, 2016-08-24:
The Teensy has got a MK20DX128 or MK20DX256VLH7.

The library is currently working on a MKL25Z128VLK4 i.e. a FRDM-KL25Z 
I have similar code running on a MK20DX128VLH5 i.e. a FRDM-K20D50M board

As far as I can see,   the Teensy does not accept regular Shields.   And no one has ever asked for Teensy support.    There does seem to be a Sparkfun Adapter shield.    But it looks a bit of a Heath Robinson arrangement.

If you are using the Sparkfun Adapter Shield,  I will add the support to mcufriend_shield.h
If you tell me what non-standard pins you intend to use,  I will add the appropriate support to mcufriend_special.h

I can only test on the FRDM-K20D50M.  So I would require feedback from you.

David.

Thebobcachelot, 2016-08-25:
I was Expecting to make my own adapter, i have bought the shield version, but a breakout version would do the same job. for the SPI Pin: it seems to be the same thing as UNO, 13 sck(Port C 5), 12DIn(Port C7), 11 DOUT (Port C6), 10 CS(Port C4). for the analog pin I would probably use the same ReSet A4(B3), 
CS A3(B1) , CD A2(B0) and Wr  A1(C0) and RD A0(D1). the processor is the MK20DX256VLH7.

prenticedavid, 2016-08-25:
It appears that my FRDM-K20 board uses the same mcu as the Teensy3.0.

If I can run the Teensyduino software on the FRDM board,  it would be easy to support.

Your Teensy3.1 uses a different mcu to the FRDM-K20
I can add your configuration to the mcufriend_special.h
You have explained which Port bits are used for control signals: LCD_RD etc  
You have not said which Port bits are used for data bus: LCD_D0 .. LCD_D7

Has anyone got Teensyduino working on a FRDM board?

David.

Thebobcachelot, 2016-08-25:
pin 2,A6-A7,5,6,7,8,9 will be easy to use so D0, D5,D6,D7,D4,D2,D3,C3 .  as I know about the teensy, people devop alot with this but stay with this size of board because you can compile it as an arduino but with a lot more ram and flash and everything else. almost every information possible is availlable on PJRC.com

prenticedavid, 2016-08-25:
You do not make any sense.   I want to know which port pin is used for
LCD_D0
LCD_D1
LCD_D2
LCD_D3
LCD_D4
LCD_D5
LCD_D6
LCD_D7

LCD_RD PTD.1
LCD_WR PTC.0
LCD_RS  PTB.0
LCD_CS  PTB.1
LCD_RESET PTB.3

The reason that I support "shields" is because I know what pins go to which port.    I have no idea what Teensy uses. 

David.

prenticedavid, 2016-08-25:
I have installed v1.6.9 of the Arduino IDE
I have installed Teensyduino
I have selected Teensy3.0

The library and sketch builds successfully.   The IDE creates a BIN file in the sketch directory.
I do not know how to upload the compiled BIN into the FRDM.   Is there a third party programmer that can upload the BIN via the CMSIS-DAP ?

Everything works fine with Keil.

David.

Thebobcachelot, 2016-08-25:
With teensy the arduino ide upload directly in the mcu without third party
. I comeback with the pin selection as soon as i go home. Thank you for
your support!

Thebobcachelot, 2016-08-26:
Shield    |  Teensy
LCD_D0 | Port D0, 
LCD_D1 | Port C3
LCD_D2 |  Port D2
LCD_D3 |  Port D3
LCD_D4 |  Port D4
LCD_D5 |  Port D5
LCD_D6  |  Port D6
LCD_D7  |  Port D7

this will be perfect for the pin selection.
If your Code & sketch compile, I'll have to try on my side when you are ready.
Thank you again, contributing to make this world better!

prenticedavid, 2016-08-26:
I have added a branch called "test_frdm"

Please plug your display shield into its proper Teensy Adapter.   i.e. **standard** Arduino Pins
Please run the graphictest_kbv sketch.

Does it report the correct ID to the Serial Terminal?
Do the graphictests work?   I have made no attempt to slow down the RD_STROBE or WR_STROBE.    They work fine on a 48MHz FRDM (compiled with Keil)

If everything works ok,  you can #define USE_SPECIAL and USE_BOBCACHEALOT_TEENSY macros.
Rewire for your custom data bus.   And test.    This may also need slowing down.

David.

Thebobcachelot, 2016-08-26:
Thank you i'll give you new a the end of the week end!

SeeedMonkey, 2016-09-15:
hi,
i came across this thread after i got the same initial question as thebobchelot:
I got a 3.97" Touch LCD from mcufriend, it works perfectly with my Arduino Uno. Now i want to change to a Teensy 3.2, to build a project with it. I tried the test_frdm branch, the code compiles and uploads fine.
However, i just get a white Screen on the LCD, the Serial Monitor stops after "ID = 0x" with the graphictest_kbv sketch...
I got the Panel connected via jumper links to the same pins as they where connected to on the UNO, i also tried the alternative wiring from Thebobcachelot  in mcufriend_special.h with the same result.
LCD_ID_readreg seams to work, here is what i got in the Serial monitor:

`Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 77 93	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Powsr Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
`
Here is a pic of the statistic screen from the graphic test on the UNO:
[image]

I would appreciate your help, a big thanks anyway for the awesome library!
Greetings from Germany,

prenticedavid, 2016-09-15:
First off.   You know how everything works on a Uno.
I presume that you have plugged the Shield into the Sparkfun Adapter.   Built and uploaded graphictest_kbv.ino from the examples.

What does the Serial Terminal say?   What ID does it report?
Does the Teensy require you to wait for Serial to start?   The example sketch could wait a little longer. 
You could force tft.begin(0x7793) if it did not read the ID correctly.  But if it read wrong, it probably writes wrong too.

I have heard nothing from Mr bobcachealot.    So I do not know if I have got the macros correct.
Your "build" should say _warning regular UNO shield on a Teensy 3.x_ if you have set verbose compilation in IDE Preferences.

David.

SeeedMonkey, 2016-09-16:
Hi,
i do not own a Sparkfun adapter, so i use some breadboard jumper cables / links to connect the LCD to the teensy. But i took the schematics of the shield as a reference... 

The Serial Terminal just says:

_Serial took 5000ms to start
ID = 0x0_

and thats it.
To me it seams that the processor can not read the ID and therefore also can not initialize the LCD.
I tried to force the ID, but that does not make a difference.
I do get the warning "warning regular UNO shield on a Teensy 3.x" during compilation.

Could the problem got something to do with the 3.3V logic of the Teensy? Does the LCD require 5V logic signals?

Could you please post a list of the pins you assigned for the lcd_d0 to lcd_d7 connections in the mcufriend_special.h? I don`t understand that part, and i could check my wiring again :)

Greetings

prenticedavid, 2016-09-16:
I have double-checked my macros.   They look ok to me but obviously I can't test them in real life.
LCD_D0 PTD3
LCD_D1 PTC3
LCD_D2 PTD0
LCD_D3 PTA12
LCD_D4 PTA13
LCD_D5 PTD7
LCD_D6 PTD4
LCD_D7 PTD2

LCD_RD PTD1
LCD_WR PTC0
LCD_RS PTB0
LCD_CS PTB1
LCD_RST PTB3

3V3 3V3
5V   5V
GND GND

Quite honestly,  you should be able to map the Teensy A0 pin with the Uno A0 pin (LCD_RD)
Likewise the Teensy D9 pin with the Uno D9 (LCD_D1).    
The Display Shield has all the pins clearly marked.

I strongly advise that you buy a Sparkfun Adapter.    Then at least I know that your Shield has mated with the correct pins.   The display logic is happy with 3.3V.  Only the Backlight uses 5V (to feed its own LDO regulator.)

David.

SeeedMonkey, 2016-09-20:
hi,
so i ordered a Sparkfun Adapter, witch arrived today. Unfortunately, i still get no image on the lcd. Only the white screen :( The LCD_ID_readreg and the graphicstest_kbv still output the same to the serial monitor.
You mentioned to slow down the rd and wd strobe, how could i do that? Maybe that would help...
Thanks in advance, Greetings

prenticedavid, 2016-09-20:
I don't know what the Teensy does by way of initialising GPIO pins.    Freescale MCUs do not start up with GPIO,  but I would guess that the Arduino startup would enable the peripherals.
The Atmel SAM3X and SAMD21 chips in Due and Zero enter setup() with GPIO enabled.

If the Teensy does not,  please just call pinMode() before the first tft call i.e. tft.readID()   
`
for (int i = 2; i < 10; i++) pinMode(i, OUTPUT);

for (int i = A0; i < A5; i++) pinMode(i, INPUT_PULLUP);
`
Let me know if readID() works.

David.

stevstrong, 2016-09-21:
I had similar problems, the root cause was that the reset signal was not output from Arduino board to the LCD module.
S0 please check the reset line.

prenticedavid, 2016-09-21:
The Mcufriend boards MUST have LCD_RST connected.   The Library makes this pin an output,  performs a hardware reset with an active-low pulse.

Modern MIPI controllers have a "Software-Reset" command (0x01).   This is always used on MIPI controllers.    So the LCD_RST pin is superfluous.   Just connect a 10k pull-up resistor to LCD_RST pin.

@stevstrong,
Have you tried the Teensy?

David.

prenticedavid, 2016-09-21:
ST, Atmel, NXP enable GPIO as part of the startup (as far as I can remember)
Freescale does not.   To use MCUFRIEND_kbv on MBED with FRDM boards,  I have to tell MBED to create some dummy GPIO.

My own Freescale projects initialise GPIO as part of its regular initialisation,
I do the same with ST but it is superfluous.

If you are using an ST board with Arduino headers,  it would be wise to add it as 'standard' in mcufriend_shield.h.

David.

prenticedavid, 2016-09-26:
I did not hear back from anyone.  So I ordered a Teensy3.2 which arrived today.

The Arduino startup does not put the digital or analog pins into GPIO mode.   So a manual pinMode() loop was necessary in the sketch.
The Driver required a little "slowing down" of the write cycle.   And a big "slowing down" of the Read cycle.

The Teensy is certainly impressive.   The Adafruit tests take about 3.5 secs on a MK20D5 @ 48MHz but only 1.84 secs with the MK20D7 @ 96MHz.

I will update "mcufriend_shield.h" later this evening (probably).

David.

stevstrong, 2016-09-26:
You mean all the tests together take only 1.8 secs? Wow. Do you have a youtube video?

prenticedavid, 2016-09-27:
I have updated the test_frdm branch.   I have only tested with a ILI9341.  Obviously a bigger display takes longer to run the Adafruit tests.    Other controllers may not be happy with the speed.

I don't have a video.  One day I will learn how to do videos.

I have no intention of buying a STM32duino and wiring up a shield adapter.   But I would be happy to buy a NUCLEO-F103RB **if** it is integrated with the Arduino IDE.    A 72MHz M3 should run just as well as any other Cortex.    In practice the GPIO speed is too fast.    The overall time depends on how complex the pin mapping is for the data bus.

David.

stevstrong, 2016-09-27:
Here you can see what one can achieve with STM32:
https://www.youtube.com/watch?v=V8fU9Ve4Xp0
Is your version much faster than that?

SeeedMonkey, 2016-09-27:
hi,
thank you very much for the update!
It works perfectly now! With my 3.95" screen it finishes in 2.27 sec :)
Reading the ID also works now, its 7793. Touch works fine too.
Thank you very much for your help again, i had almost given up on getting it to work :)
Greetings, Samuel

W8TEE, 2016-10-15:
I just got my 3.5 and 3.6 Teensy...amazing stuff. Any plans to support these monsters?

PeteJohno, 2017-01-04:
Thanks David for your library! 
I have just got my Teensy 3.5 running on a 2.4" mcufriend UNO shield (ILI9325)
Graphics loads in 1.35sec!
Thanks, Pete

prenticedavid, 2017-01-04:
The "Adafruit Tests" take 1.53sec on a 32MHz Xmega.   1.22sec with overclocked 40MHz.
It is faster with a 16-bit bus.   But you are limited by the controller rather than the AVR/ARM.
From memory,   the best that I can achieve on an 8-bit bus is 0.89sec with a 48MHz Xmega.  But some controllers are faster than others e.g. 66ns tWC.   I have found the ST7789V is a bit fussy.

The main idea behind the library was to work with ALL Mcufriend Shields.   I could optimise for a specific controller and go for speed, size, ...

David.

prenticedavid, 2017-01-05:
The idea is to support Shields.   So if it is a commercial Shield like Sparkfun,  it should go in mcufriend_shield.h.    Yes,   I could detect Teensy 3.2, 3.5, 3.6 chips.   And just allow different delays for chip and F_CPU.  

If you have some special wiring,  it should go in mcufriend_special.h
What adjustments did you make to the write timing,  and what to the read timing?

Incidentally,  tWC for an ST7789S or ST7789V is supposed to be 66ns.   My display is not happy with anything less than 250ns.

David.

shsg2003, 2017-05-23:
Hi prenticedavid!
I use Teensy 3.2 with 2.8" display with driver st7783 and yours lib MCUFRIEND on Arduino 1.8.1.
Could you kindly explain to me how to change control pins on Teensy from A0,A1,A2,A3,A4 to other, for example to A10,A11,A12,A13,A14.
I try use example "graphictest_kbv", there change to other but unlucky.
Or may be you have library only for ST7783 driver?
Thanks in advance.

prenticedavid, 2017-05-23:
I provide a driver for "Sparkfun Adapter with Teensy3.x".   You just plug and go.

If you want to use different pins,  you have to write a suitable "SPECIAL".    It is not a question of changing constructor arguments.   (they are all dummies)
Life is much simpler if you use the Sparkfun Adapter wiring and the existing library.

If you produce a compelling reason for different wiring,   I will write the SPECIAL for you.
e.g.  you are going to make/sell a "better" Adapter than Sparkfun.

David.

closed 2018-05-16 by prenticedavid