prenticedavid/MCUFRIEND_kbv #9: How to move data pins from interrup pins 2 and 3

stojanos, 2017-01-30:
Hi,

All these shields use digital interrupt pins 2 and 3 as part of data pins. Moving these pins to (e.g. 9 and 10) that are often used by SD card would free up interrupt fields for use with e.g. encoders. 

I can physically rewire this shield and UNO board but I will have to change this code to properly use new 9 and 10 pins instead of 2 and 3 and this is the problem. 

I have looked through the code and there is no definition for data pins that can be easily changed.  I suspect that good starting point for code change is in mcufrined_shiled.sh line 25 but it is too complex to update without spending several hours first understanding complex byte shifting that is happening there. 

Can somebody confirm that I am on right track before I do this or let me know better way of doing it ? I promise that I will publish all my work here.

Thanks!

prenticedavid, 2017-01-30:
A Shield has got fixed wiring.   If it did not,  I would not have written the library.

Yes,  it is a little fiddly to create a SPECIAL and put it in the mcufriend_special.h.
If you tell me your exact wiring,  I could do it for you.

Quite honestly,  you use up almost all GPIO on a Uno.    You could use a pin-change on A5 or the D10-D13 if you are not using SPI.

David.

stojanos, 2017-01-30:
Thanks David. Yes I fully agree that all shields have fixed wiring and that the only way to free up some dig pins are to stop using SD card pins 10-13. I don't need SD card from shield and I need 2 and 3 for interrupts.. Therefore I thought to move 2 an 3 to 10 and 11 doing the following :
unsolder dig pins 2 and 3 , 10, 11, 12 and 13 from the shield so that they do not connect to UNO
solder wire to 2 and 3 shield and connect wire to UNO pins 10 and 11.  so that data will work. 

SD card will be lost but interrupt pins can be used. This is what I found very important. Often we use encoders which values need to be presented on screen. No matter how efficient is the code for screen rendering if we do not visit encoder code on time it can lose the measurement . Having encoder triggered by with interrupt will help a lot here and get smooth user interface.

So if you can help me to update code to use pins 10 and 11 instead of 2  and 3 that would be great.

Thanks a lot !
Srdjan

stojanos, 2017-01-30:
I forgot to mention - Thanks for such a great library that you built !!!

prenticedavid, 2017-01-30:
edit mcufriend_special.h
`
#define USE_STOANOS
...
#elif defined(__AVR_ATmega328P__)  && defined(USE_STOJANOS)      // 
#warning SPECIAL for USE_STOJANOS
#define RD_PORT PORTC
#define RD_PIN  0
#define WR_PORT PORTC
#define WR_PIN  1
#define CD_PORT PORTC
#define CD_PIN  2
#define CS_PORT PORTC
#define CS_PIN  3
#define RESET_PORT PORTC
#define RESET_PIN  4

#define BMASK         0x0F
#define DMASK         0xF0
#define write_8(x)    { PORTB = (PORTB & ~BMASK) | ((x) & BMASK); PORTD = (PORTD & ~DMASK) | ((x) & DMASK); }
#define read_8()      ( (PINB & BMASK) | (PIND & DMASK) )
#define setWriteDir() { DDRB |=  BMASK; DDRD |=  DMASK;  }
#define setReadDir()  { DDRB &= ~BMASK; DDRD &= ~DMASK;  }
#define write8(x)     { write_8(x); WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

#define PIN_LOW(p, b)        (p) &= ~(1<<(b))
#define PIN_HIGH(p, b)       (p) |= (1<<(b))
#define PIN_OUTPUT(p, b)     *(&p-1) |= (1<<(b))

`

Untested.   Typed into the Browser.   Enable by editing mcufriend_sheld.h
#define USE_SPECIAL

At least you have chosen a "sensible" alternate pinout.
You just need to snip the shield D2, D3 pins off the shield.  And solder thin wires from the top of D2 to D10, D3 to D11.   You can access D2, D3 with some angle header with the Shield in place.

I still reckon you should use Pin-Change interrupts.

David.

stojanos, 2017-01-30:
David, you are star. I will test this when I come home tonight and report back. If it works I will send you the tested mcufriend_special.h file.

Yes, I have thought about pin change interrupts as well. They are not that elegant as external interrupts because they are triggered for set of pins and not just for a single one. Furthermore default Encoder library is not using it so I would need to implement it myself. This is more elegant for me.

Thanks a lot.

stojanos, 2017-01-30:
Eh, it did not work. At least now I know were to focus to try to get it working.
Thanks anyway. 
Srdjan

prenticedavid, 2017-01-30:
I don't believe you.    I just tried on a regular Uno with:
`
#define BMASK         0x03 //0x0F
#define DMASK         0xFC //0xF0
#define write_8(x)    { PORTB = (PORTB & ~BMASK) | ((x) & BMASK); PORTD = (PORTD & ~DMASK) | ((x) & DMASK); }
#define read_8()      ( (PINB & BMASK) | (PIND & DMASK) )
#define setWriteDir() { DDRB |=  BMASK; DDRD |=  DMASK;  }
#define setReadDir()  { DDRB &= ~BMASK; DDRD &= ~DMASK;  }
`
And it worked just fine for regular Shield.   Your BMASK, DMASK values are all that I need to change.
I presume that you have LCD_D0 - LCD_D3 on D8-D11 (with no microSD in socket)

I still reckon that you should use Pin-Change.   It does everything that INT0, INT1 do.

David.

stojanos, 2017-01-30:
Yes you are right. Something is wrong with my code. It is not picking up mcufriend_special.h  file. I have done change in mcufriend_shiled.h to define USE_SPECIAL

stojanos, 2017-01-30:
Your code is correct. I have tested it when updated directly inside mcufrined_shield.h file. I don't know at the moment why mcufrined_special.h was not picked up even when having #define USE_SPECIAL   as a first line in the shield file. I will investigate further tomorrow but the most important thing is that this now works !!!
Thanks David for all your effort !!!.

porkyhttp, 2017-07-01:
Hi all,
I need to change the LCD_D0 to other pin using a Mega 2560 board, because i nedd the standard D8 pin for the FreqMeasure library, and also I would use the SDcard to store some infos. Could you help me with this issue? 
Best regards

prenticedavid, 2017-07-01:
A shield has fixed pins.   Every male pin has a corresponding female header socket on the Arduino.

The only way that you can change pins is to physically mutilate the male pins on the Shield.

If you tell me which Port pin you want to use for LCD_D0,   I will post a SPECIAL for you.

Digital#8 is PORTH.5 whose only special function is OC4C.   There are other PWM pins on the 18x2 header.   There would be no need to mutilate your Shield.

David.

porkyhttp, 2017-07-01:
Hi David,
I am using cables to connect the Mega board to the LCD, and other devices like GPS, rain, dht22 and other sensors.
So because I want to use D10 to D13 for the SDcard, could you write a special that use any pin from D22 to D50?

prenticedavid, 2017-07-01:
I strongly advise against using cables.   God invented Shields and they were good.

Using a commercial Adapter Shield or custom Protoshield gives you robust mechanical support and reliable electrical connections.    

I made my own adapter some years ago.   It has always been available as a SPECIAL.   i.e. USE_MEGA_8BIT_PROTOSHIELD which uses D22-D29 for the databus and the regular A0-A4 for the control signals.
You can write your own SPECIAL.   Note that the Touch Panel needs two Analog pins.

David.

porkyhttp, 2017-07-01:
Ok the PCB is next step, but for now I am in testing e developement mode :)

For switch to the USE_MEGA_8BIT_PROTOSHIELD pinout I must ad it in my code, like  #define 
USE_MEGA_8BIT_PROTOSHIELD ??

prenticedavid, 2017-07-01:
Read the how_to file.   It explains how to use a SPECIAL.

porkyhttp, 2017-07-01:
Ok as described I uncommented the #define USE_SPECIAL  in the sheild file and uncommented #define USE_MEGA_8BIT_PROTOSHIELD on the top of the special file, but nothing works

prenticedavid, 2017-07-01:
Well,  you have to connect all the wires to the appropriate pins.

It is always a good idea to test a wiring scheme with the LCD_ID_readreg.ino sketch.
Edit the #defines to match your new wiring.   If the readreg sketch works,  your wiring is correct.

As I said earlier.   The library is designed for Shields that are intimately plugged into Arduino(s).

porkyhttp, 2017-07-01:
//-- Arduino UNO or Mega 2560 Plugged as shield
#define LCD_RST A4
#define LCD_CS A3
#define LCD_RS A2
#define LCD_WR A1
#define LCD_RD A0

#define LCD_D0 28
#define LCD_D1 29
#define LCD_D2 22
#define LCD_D3 23
#define LCD_D4 24
#define LCD_D5 25
#define LCD_D6 26
#define LCD_D7 27

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 66 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 02 04 94 81 FF	ILI9481, HX8357-B
reg(0x00C0) 00 10 3B 00 02 11 00 00 00	Panel Control
reg(0x00C8) 00 00 44 06 11 44 55 0A 08 17 33 77 11	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 43	Power Control
reg(0x00D2) 00 01 22 00 00	NVM Read
reg(0x00D3) 00 01 22 00	ILI9341, ILI9488
reg(0x00D4) 00 01 22 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

Are you sure that USE_MEGA_8BIT_PROTOSHIELD uses the pin from D22 to D29 linear?

prenticedavid, 2017-07-01:
You appear to have got a correct output with some unusual defines.

I would expect the defines to be:
#define LCD_D0 22
#define LCD_D1 23
#define LCD_D2 24
#define LCD_D3 25
#define LCD_D4 26
#define LCD_D5 27
#define LCD_D6 28
#define LCD_D7 29

David.

porkyhttp, 2017-07-06:
Ok now it works correctly.
SDcard functionality remain on standard pins?
Many thanks

prenticedavid, 2017-07-06:
Only **you** know how **you** have wired things.    If you have chosen to use custom data pins that suit the Mega2560,   you might just as well wire the SD pins to suit the Mega2560.

God invented Shields for Arduino and they were good.
The SD pins are correct for the Uno.   You must use Software SPI  if you are using Mega or Due.

David.

porkyhttp, 2017-07-06:
Ok I wll check later for the sd. Now I am trying to replace xbee serial communication with NRF24 modules and MCU to communicate with a pc with a NRF24L01 USB adapter

prenticedavid, 2017-07-18:
I have no idea what you are trying to say.
You can draw any graphics wherever you want. 
You can print the value of any expression at any cursor position on the screen.

I suggest that you ask simple programming questions on a forum like arduino.cc
GitHub is better suited to technical suggestions, questions, code, ...

David.

emaddoost, 2019-12-04:
> 
> 
> edit mcufriend_special.h
> `
> #define USE_STOJANOS
> ...
> #elif defined(**AVR_ATmega328P**) && defined(USE_STOJANOS) //
> #warning SPECIAL for USE_STOJANOS
> #define RD_PORT PORTC
> #define RD_PIN 0
> #define WR_PORT PORTC
> #define WR_PIN 1
> #define CD_PORT PORTC
> #define CD_PIN 2
> #define CS_PORT PORTC
> #define CS_PIN 3
> #define RESET_PORT PORTC
> #define RESET_PIN 4
> 
> #define BMASK 0x0F
> #define DMASK 0xF0
> #define write_8(x) { PORTB = (PORTB & ~BMASK) | ((x) & BMASK); PORTD = (PORTD & ~DMASK) | ((x) & DMASK); }
> #define read_8() ( (PINB & BMASK) | (PIND & DMASK) )
> #define setWriteDir() { DDRB |= BMASK; DDRD |= DMASK; }
> #define setReadDir() { DDRB &= ~BMASK; DDRD &= ~DMASK; }
> #define write8(x) { write_8(x); WR_STROBE; }
> #define write16(x) { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
> #define READ_8(dst) { RD_STROBE; dst = read_8(); RD_IDLE; }
> #define READ_16(dst) { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }
> 
> #define PIN_LOW(p, b) (p) &= ~(1<<(b))
> #define PIN_HIGH(p, b) (p) |= (1<<(b))
> #define PIN_OUTPUT(p, b) *(&p-1) |= (1<<(b))
> 
> `
> 
> Untested. Typed into the Browser. Enable by editing mcufriend_sheld.h
> #define USE_SPECIAL
> 
> At least you have chosen a "sensible" alternate pinout.
> You just need to snip the shield D2, D3 pins off the shield. And solder thin wires from the top of D2 to D10, D3 to D11. You can access D2, D3 with some angle header with the Shield in place.
> 
> I still reckon you should use Pin-Change interrupts.
> 
> David.

How can I modify the SPECIAL code which you sent , so that I can switch from 2 ,3 pins on MEGA2560 to pins 10 and 11?  (Beginner's guide , sorry)

prenticedavid, 2019-12-04:
God invented Shields.   The Shield plugs into the Mega2560.

If you want to use an INTn interrupt,  the Mega2560 has several INTn pins.   And you can access them with the Shield in place.
Alternatively use PCINT interrupts.

David.

emaddoost, 2019-12-04:
> 
> 
> God invented Shields. The Shield plugs into the Mega2560.
> 
> If you want to use an INTn interrupt, the Mega2560 has several INTn pins. And you can access them with the Shield in place.
> Alternatively use PCINT interrupts.
> 
> David.

Tnx, I need 4 interrupts . I am using a clock so I can't use pins 20 and 21 (pins are SCL & SDA on MEGA2560) as interrupts. The only solution is to use pins 2,3,18,19 . Pins 2,3 are already used for LCD and I want to use them as interrupts.
 I uncommented #define USE_MEGA_8BIT_PROTOSHIELD in mcufriend_special.h
then uncommented #define USE_SPECIAL  in mcufriend_shield.h
and changed pins D2-D9 to D22-D29 
Now the LCD is working good but the touch does not work and the analoge pins are connected as follows:
RD---> A0
WR---> A1
RS---> A2
CS---> A3
What should I change to fix the touch?

prenticedavid, 2019-12-04:
I presume that you calibrated your Touch when it was a regular shield.
If Touch is on LCD_D6 and you have moved LCD_D6 from 6 to 28,  you change the Touch constructor. from 6 to 28.

emaddoost, 2019-12-04:
> 
> 
> I presume that you calibrated your Touch when it was a regular shield.
> If Touch is on LCD_D6 and you have moved LCD_D6 from 6 to 28, you change the Touch constructor. from 6 to 28.

Solved! , Thanks so much

nobodyuknow, 2020-01-05:
Hi David,
I have a similar problem on a breadboarded Atmega1284P.  I really need the external interrupt pins for use with several external devices.  I currently have the Kuman 3.5" 320x480 ILI9486 UNO shield display hooked up according to the Bobuino pinout and it works fine --- I just need those interrupts. I've attached an image showing the current hookup for the display.   Used pins are marked with a red X and open (unused) pins are marked with a green circle.  The interrupt pins I want are marked with a red circle --- .D6/PB2,  D2/PD2,  D3/PD3,
[image]
  I've looked at the special.h files and the how_to , but can't really figure out how to go about changing anything.  Do you have any sort of guide for making these changes?  Or any advice?  Seems like you always ask for the regs, so following is the output of LCD_ID_readnew.
Thank you so much for any help!  Robby R

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00 ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66 Manufacturer ID
reg(0x0009) 00 00 61 00 00 Status Register
reg(0x000A) 00 08 Get Powsr Mode
reg(0x000C) 00 66 Get Pixel Format
reg(0x0030) 00 00 00 00 00 PTLAR
reg(0x0033) 00 00 00 00 00 00 00 VSCRLDEF
reg(0x0061) 00 00 RDID1 HX8347-G
reg(0x0062) 00 00 RDID2 HX8347-G
reg(0x0063) 00 00 RDID3 HX8347-G
reg(0x0064) 00 00 RDID1 HX8347-A
reg(0x0065) 00 00 RDID2 HX8347-A
reg(0x0066) 00 00 RDID3 HX8347-A
reg(0x0067) 00 00 RDID Himax HX8347-A
reg(0x0070) 00 00 Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30 RD_DDB SSD1963
reg(0x00B0) 00 00 RGB Interface Signal Control
reg(0x00B3) 00 00 11 11 11 Frame Memory
reg(0x00B4) 00 00 Frame Mode
reg(0x00B6) 00 02 02 3B 3B Display Control
reg(0x00B7) 00 06 Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00 ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E Panel Control
reg(0x00C1) 00 44 00 00 Display Timing
reg(0x00C5) 00 00 Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00 GAMMA
reg(0x00CC) 00 04 Panel Control
reg(0x00D0) 00 00 00 00 Power Control
reg(0x00D1) 00 00 00 00 VCOM Control
reg(0x00D2) 00 00 00 Power Normal
reg(0x00D3) 00 00 94 86 ILI9341, ILI9488
reg(0x00D4) 00 00 00 00 Novatek
reg(0x00DA) 00 54 RDID1
reg(0x00DB) 00 80 RDID2
reg(0x00DC) 00 66 RDID3
reg(0x00E0) 00 0F 21 1C 0B 0E 08 49 98 38 09 11 03 14 10 00 GAMMA-P
reg(0x00E1) 00 0F 2F 2B 0C 0E 06 47 76 37 07 11 04 23 1E 00 GAMMA-N
reg(0x00EF) 00 80 00 10 60 40 ILI9327
reg(0x00F2) 00 18 A3 12 02 B2 12 FF 10 00 00 00 Adjust Control 2
reg(0x00F6) 00 54 80 66 Interface Control

Thanks again, RR

Emiliop89, 2020-02-24:
Hi! your library is great! 
I pilot a display 230x320 whit arduino, the chip on the tft is 6809. I started using arduino UNO and after i used MEGA 2560, but they are too slowly for my application. i bought an arduino DUE. 
i can't use a shield to connect arduino and the tft so i wiring them manually.
I want to ask you if the way to make the tft work faster, without flashing, is to connect the tft's pins from d0 to d7 to arduino's pins from 2 to 9 or other connection. also for rd, wr, rs, cs and rst
thanks!

prenticedavid, 2020-02-24:
There is a SPECIAL already written in mcufriend_special.h i.e. USE_DUE_8BIT_PROTOSHIELD
The data bus is wired to PD0..PD7 and the control pins look like the regular Analog i.e. PA16, PA24, PA23, PA22, PA6

Yes,  the Protoshield goes much faster.   Might be too fast for a RM68090

David.

Emiliop89, 2020-02-25:
PD0..PD7 is from 22 to 29 on arduino DUE?
nothing works.. -.-

prenticedavid, 2020-02-25:
PIOD.0 .. PIOD.7 are on random digital pins e.g. 25-29, ...
Look at a pinout diagram for a Due.

If you want to wire your data to 22-29 like a Mega and controls to 38-42,   you should use USE_MEGA_8BIT_SHIELD

closed 2021-08-24 by prenticedavid