prenticedavid/MCUFRIEND_kbv #10: Teensy 3.6

victorbramhall, 2017-02-11:
Hello David,
I am trying to use MCUFRIEND_kbv with a teensy 3.6. I am using a 2.8" TFT LCD Shield connected as follows:
   2.4" 8-Bit TFT LCD ILI9341 Display

  LCD         Teensy3.6
   ---             -------
LCD_DB0          7
LCD_DB1          8
LCD_DB2          2
LCD_DB3          3
LCD_DB4          4
LCD_DB5          5
LCD_DB6          6
LCD_DB7          7

#define LCD_CS   A3
#define LCD_CD  A2
#define LCD_WR  A1
#define LCD_RD  A0
#define LCD_RST A4

#include <SPI.h> 
#include "Adafruit_GFX.h"
#include <MCUFRIEND_kbv.h>
MCUFRIEND_kbv tft;

The error I get is #error MCU unsupported. Is there an updated library that will support the Teensy 3.6?
Thank you,
Victor

prenticedavid, 2017-02-11:
No,   I do not own a Teensy3.6.   I might buy a Teensy 3.5 one day.   I have no intention of buying a 3.6

The mcufriend_shield.h or mcufriend_special.h test for the specific chip macro.  e.g.  defined(__MK20DX128__)
The 3.5 and 3.6 use different NXP chips.    Both will be faster than the MK20DX used by 3.2.

If you assign "sensible" port pins for the signals AND test with LCD_ID_readreg.ino sketch, I will write the SPECIAL for you.    
At the moment you have LCD_DB7 and LCD_DB0 both connected to D7 (I presume digital#7) pin

David.

victorbramhall, 2017-02-11:
Thank you for your quick response. I will continue trying to solve my
problem.

 

Thanks,

Victor

prenticedavid, 2017-02-11:
Go on.  It is pretty simple to just correct your unusual "defines":
LCD_DB0 7
LCD_DB1 8

Then test whether your "wiring" works by running the LCD_ID_readreg sketch.
The library is designed to work with physical Shields that are intimately plugged into standard Arduinos.

I know the pinout of an official Arduino.   If punters want to connect differently,   it is their responsibility.
But I am happy to help if they can confirm their wiring first.   e.g. by pasting the defines that they used in LCD_ID_readreg.

David.

victorbramhall, 2017-02-12:
I will be happy to send you a Teensy 3.6 (configured for breadboard) if you will help get the MCUFRIEND_kbv library to support and compile for the board. If you are interested.

Thank you,
Victor Bramhall

victorbramhall, 2017-02-12:
[image]

prenticedavid, 2017-02-13:
Ah-ha.  I see that you have edited your pinout in the first message.
That looks exactly like the regular Shield pinout.

I have added Teensy3.5, Teensy3.6 to the current Master Branch.
I am fairly confident that it will "work" out of the box.

If you follow PeteJohno's procedure with graphictest_kbv.ino sketch,  you can reduce the write delay until you get glitches in Rounded Rectangles.   You can reduce the read delay until you get glitches in Software Scroll.

It looks as if you have a Samsung S6D0154 controller.   This should work 100% with every test.
Are all twenty boards identical?

David.

victorbramhall, 2017-02-13:
Thank you. I will look at making the changes today. Yes, all twenty boards
are the same and have the 0x0154 chip. Are you interested in the teensy 3.6?

prenticedavid, 2017-02-13:
No,  I have far too many boards.

Please try the Teensy3.6 for yourself and report back with the timing results.

closed 2018-05-16 by prenticedavid