prenticedavid/MCUFRIEND_kbv #11: HX8347G problem

wallace09, 2017-03-09:
I'm trying to use a 2.8 "TFT screen shield with arduino uno, 
 but I could not write anything on the screen, I tried the library of MCU_KBV and I am waiting for your help 
Screen: MCUFRIEND HX8347G
Note: the SD card on the screen is working

[image]
[image]

prenticedavid, 2017-03-09:
Woo-hoo.  Mcufriend is "almost" becoming helpful.    e.g. you probably have HX8347-G 
Other Shields are appearing with a sticky paper label with a controller number.

In MCUFRIEND_kbv.cpp uncomment the SUPPORT_8347D macro.

David.

wallace09, 2017-03-09:
Thank you so much.
You're king

closed 2017-03-09 by wallace09