prenticedavid/MCUFRIEND_kbv #12: HX8357B problem

SeanShaoCN, 2017-05-16:
I'm trying to use a 3.2 "TFT screen shield with arduino mega 2560,
but I could not write anything on the screen, I tried the library of MCU_KBV and I am waiting for your help
Screen: MCUFRIEND HX8357B

prenticedavid, 2017-05-16:
You have a genuine Himax HX8357-B controller.
Which version of the library are you using?   e.g. v2.9.5 Release or Master Branch of v2.9.6-Beta

I have tried to buy a HX8357-B Shield without success.
I rely on feedback from users.

What happens if you force **tft.begin(0x9090);**   i.e. say that it is a HX8357-D

David.

SeanShaoCN, 2017-05-16:
when i force tft.begin(0x9090); It works fine.
Thank you so much.

prenticedavid, 2017-05-16:
Please wait for a reply before closing the Issue.

I have updated the Master Branch to initialise HX8357-B as a HX8357-D.
Please can you test graphictest_kbv.ino with the new version.

Colours, directions, penguin,  software scroll, inversion should all work 100%.
If you are unsure,  please ask.   If you can link to a video,   I can see for myself.

I will make an official Release when I get your report.   And close the Issue properly.

David.

SeanShaoCN, 2017-05-16:
VIDEO:    [https://youtu.be/x2HM4c8aqbg](url)

prenticedavid, 2017-05-16:
Thanks for the video.  Everything is working except for "Software Scroll"

It looks as if the 8357B reads GRAM memory 16 bits at a time.   I have put the edit onto Master.
There is no need to post another video.   Just tell me if the Software Scroll works.

David.

SeanShaoCN, 2017-05-16:
"Software Scroll" is working  properly! and  where are you ?

prenticedavid, 2017-05-16:
I am in Wormshill in England.

You do not realise just how grateful I am for your help.    I have been unable to buy an HX8357-B Shield. 
Members of the Arduino Forum report that they have received an HX8357-B from a specific Ebay shop.
I try to buy from the same shop.   Received different controllers after 3 attempts from 3 Ebay shops.

Arduino Forum members tend to say "it does not work".    They are not prepared to answer any questions or try any code.    

You have demonstrated just how well the Internet can work.    I presume that you are in China.    I am in the UK.    Different countries.   Different languages.    Ok,  it would have been nice if the first edit had worked 100%.    But two edits is not bad.

Incidentally,   I notice that you have a Mega2560.   The Adafruit tests take about 36 seconds on a Mega.   15 seconds on a Uno. 
They take about 2 seconds on an overclocked Xmega.

Thanks again.

David.

SeanShaoCN, 2017-05-16:
你好，David
       我来自于中国上海，非常高兴在帮到你的同时也帮到了我，说起来也很奇妙，我本不打算购买HX8357B的驱动板，而是要买ILI9431的驱动板，但是商家却给我发了HX8357B的货，同时，他们也没有相关的测试程序，所以我才来到这里寻求帮助，正巧你需要做相关的测试，我就试着给你留言寻求帮助，没想到你也正需要一个HX8357B的驱动板来做测试，合作愉快！

Can you read the above information? ^-*

stevstrong, 2017-05-16:
[Everyone can read it](https://translate.google.com/#auto/en/%E4%BD%A0%E5%A5%BD%EF%BC%8CDavid%0A%E6%88%91%E6%9D%A5%E8%87%AA%E4%BA%8E%E4%B8%AD%E5%9B%BD%E4%B8%8A%E6%B5%B7%EF%BC%8C%E9%9D%9E%E5%B8%B8%E9%AB%98%E5%85%B4%E5%9C%A8%E5%B8%AE%E5%88%B0%E4%BD%A0%E7%9A%84%E5%90%8C%E6%97%B6%E4%B9%9F%E5%B8%AE%E5%88%B0%E4%BA%86%E6%88%91%EF%BC%8C%E8%AF%B4%E8%B5%B7%E6%9D%A5%E4%B9%9F%E5%BE%88%E5%A5%87%E5%A6%99%EF%BC%8C%E6%88%91%E6%9C%AC%E4%B8%8D%E6%89%93%E7%AE%97%E8%B4%AD%E4%B9%B0HX8357B%E7%9A%84%E9%A9%B1%E5%8A%A8%E6%9D%BF%EF%BC%8C%E8%80%8C%E6%98%AF%E8%A6%81%E4%B9%B0ILI9481%E7%9A%84%E9%A9%B1%E5%8A%A8%E6%9D%BF%EF%BC%8C%E4%BD%86%E6%98%AF%E5%95%86%E5%AE%B6%E5%8D%B4%E7%BB%99%E6%88%91%E5%8F%91%E4%BA%86HX8357B%E7%9A%84%E8%B4%A7%EF%BC%8C%E5%90%8C%E6%97%B6%EF%BC%8C%E4%BB%96%E4%BB%AC%E4%B9%9F%E6%B2%A1%E6%9C%89%E7%9B%B8%E5%85%B3%E7%9A%84%E6%B5%8B%E8%AF%95%E7%A8%8B%E5%BA%8F%EF%BC%8C%E6%89%80%E4%BB%A5%E6%88%91%E6%89%8D%E6%9D%A5%E5%88%B0%E8%BF%99%E9%87%8C%E5%AF%BB%E6%B1%82%E5%B8%AE%E5%8A%A9%EF%BC%8C%E6%AD%A3%E5%B7%A7%E4%BD%A0%E9%9C%80%E8%A6%81%E5%81%9A%E7%9B%B8%E5%85%B3%E7%9A%84%E6%B5%8B%E8%AF%95%EF%BC%8C%E6%88%91%E5%B0%B1%E8%AF%95%E7%9D%80%E7%BB%99%E4%BD%A0%E7%95%99%E8%A8%80%E5%AF%BB%E6%B1%82%E5%B8%AE%E5%8A%A9%EF%BC%8C%E6%B2%A1%E6%83%B3%E5%88%B0%E4%BD%A0%E4%B9%9F%E6%AD%A3%E9%9C%80%E8%A6%81%E4%B8%80%E4%B8%AAHX8357B%E7%9A%84%E9%A9%B1%E5%8A%A8%E6%9D%BF%E6%9D%A5%E5%81%9A%E6%B5%8B%E8%AF%95%EF%BC%8C%E5%90%88%E4%BD%9C%E6%84%89%E5%BF%AB%EF%BC%81) :)

SeanShaoCN, 2017-05-16:
case 0x1581:
        _lcd_capable = AUTO_READINC | MIPI_DCS_REV1 | MV_AXIS | **READ_BGR** | READ_24BITS;

"Software Scroll" is working wrong !

R61581

prenticedavid, 2017-05-16:
No,  I can't read Chinese.

Google Translate told me:

> Hello, David
I am from Shanghai, China, very pleased to help you at the same time also helped me, it is also very wonderful, I did not intend to buy HX8357B driver board, but to buy ILI9481 driver board, but the business gave me HX8357B goods, at the same time, they do not have the relevant testing procedures, so I came here to seek help, happened that you need to do the relevant test, I will try to give you a message for help, did not think you are also need a HX8357B Drive board to do the test, happy cooperation!

I only have a faulty R61581 board.    It does not read its ID very well.   Sometimes it works.   It crashes after a while.    

Anyway,  I forced ID=0x1581.   The Software Scroll was not working.    I have removed the READ_BGR attribute and updated the Master Branch.

If you have a genuine R61581 please tell me whether it works reliably.   This one gets warm to the touch.   It might be wise to reduce the Power setting and adjust VCOM.

Which company do you work for?  My email is: david.prentice@farming.co.uk

David.

SeanShaoCN, 2017-05-16:
Yes, I have a genuine R61581. After you remove the READ_BGR attribute ,R61581 "Software Scroll" is working properly!

prenticedavid, 2017-05-25:
Both HX8357-B and R61581 Shields are working 100% in the v2.9.6 Release

So I will Close this issue.

David.

closed 2017-05-25 by prenticedavid