prenticedavid/MCUFRIEND_kbv #15: [SOLVED] Support for 3.5 inch ILI9481

Johi-b, 2017-08-15:
Hello David,
Thanks for the nice LIB.
I have a 3.5 inch display (Aliexpress / Kuman).
You might want to add this to the supported boards:
(I am not too familiar yet with git, so i use this way to contribute)

// These are the pins for some typical shields!
// S6D0154: YP=A1, XM=A2, YM=7, XP=6
// ST7783 : YP=A2, XM=A1, YM=6, XP=7
// ILI9320: YP=A2, XM=A3, YM=8, XP=9
// ILI9325: YP=A1, XM=A2, YM=7, XP=6
// ILI9325BG: YP=A2, XM=A1, YM=6, XP=7
// ILI9341: YP=A2, XM=A1, YM=7, XP=6
// ILI9488: YP=A1, XM=A2, YM=7, XP=6
// R65109V: YP=A2, XM=A1, YM=6, XP=7
****// ILI9481: YP=A3, XM=A2, XM=9, XP=8**** 

  } else if (identifier == 0xB509) {
        name = "R61509V";
        TS_LEFT = 889; TS_RT = 149; TS_TOP = 106; TS_BOT = 975;
        SwapXY = 1;
    } else if (identifier == 0x9481) {
        **name = "ILI9481";
        TS_LEFT = 180; TS_RT = 910; TS_TOP = 950; TS_BOT = 158;**
 }

Best Regards,
John.

closed 2018-05-16 by prenticedavid

famousfish101, 2018-06-30:
I'm sorry I'm such a begginer that none of what you said made any sense to me? I tried once by uaing Google and messed everything up. Anyway you could poijt me in the right direction. I'm new to all this stuff.

prenticedavid, 2018-07-01:
This particular Issue was closed.

Install Adafruit_GFX library and MCUFRIEND_kbv library with the Library Manager.
Plug the Shield into your Uno.
Run the examples. 

Ask Arduino questions on the Arduino.cc Forum under Displays Topic. 

David.

famousfish101, 2018-07-01:
Thanks for the email. I still don't know how to even do that. I thought I
knew but I dont understand how to make directories for the installs to
work. Whenever I use apt-get always end up receiving an error that file
doesn't exist. I don't expect you to teach me but if you could point me in
the right direction of up to date tutorials would be greatly
appreciated.Thanks David,
Ray

prenticedavid, 2018-07-01:
God gave you the Arduino IDE.    Use it.   Do not try to do things by hand.

Sketch->Include Library->Manage Libraries 
The Library Manager will install or update libraries properly.

Tools->Board Manager can update your IDE.
There is no need to use the most recent version.   Just make a note of the version (it is printed on every Edit pane)

If you have an Arduino problem ask on  the Arduino Forum.   Not GitHub.

David.

famousfish101, 2018-07-01:
If I only knew what Adruino was. Sorry to bother you. Just been trying this
out as a hobby kind of thing. Ill look mlre into it. Just so many
conflicting and outdated things on the web. Thanks again David for being
patient and helpful with a beginner.
Ray

prenticedavid, 2018-07-01:
You managed to find GitHub.

You managed to buy an Mcufriend Display Shield.
I presume that you can manage to buy an Arduino Uno to plug it into.

There are tutorials and videos on the Internet.   
God invented Google too.

David.

famousfish101, 2018-07-01:
I'm using pi3. Sorry to bother you. Believe me I have been googling but it
just shows codes but its all good thanks for the help again. Just trying to
get into the world of coding.
Ray

famousfish101, 2018-07-01:
> Go on. An Arduino is a different shape to a Pi. An Arduino Display Shield
> will NOT mate with a Pi.
>
> I bet that you own a real PC with a real USB port to plug the Arduino into.
> Follow Internet tutorials and you will soon be able to run examples and
> even write your own simple programs on the Arduino.
>
> David.

I know that what I was trying to find was the driver for my kuman 3.5 inch
touch screen display to connect to my pi. I just don't know exactly where
to start o guess. The last time I tried had to start all over by formatting
the card and burning the new kali linux version to the card and just don't
want go have to do that again. You have no obligation to help me I do
appreciate it though. If I'm too dumb to understand no hard feelings if you
don't reply. Thanks
Ray

prenticedavid, 2018-07-01:
There are _kuman 3.5 inch touch screen display_ versions for Arduino and for Pi.

Arduino version plugs into Arduino.   And Pi version plugs into Pi.

You don't need to be a rocket scientist to see which one will plug into which.
You will get far more cooperation if you provide the proper information in the first place.    Otherwise you just waste people's time.

David.

famousfish101, 2018-07-01:
Like I've said I'm a beginner and I didnt even know that Adruenos was a
thing. I have a rasperry pi3 running kali linux. Trying to get Kuman 3.5 to
work on the pi3. Like I've said I'm a beginner and if you don't feel like
helping me then just don't. I was just told that the coding community is a
very helping community. Sorry to bother you

famousfish101, 2018-07-01:
And yes the kuman touchscreen I have does fit the pi3.