prenticedavid/MCUFRIEND_kbv #17: Weird graphical effect on R61580

CrazyVito11, 2017-09-02:
Hi!
So i got a TFT LCD with the R61580 chip ID. 
When i run a sketch it works, but it looks HORRIBLE.
Colors look incorrect and you can see like a scanline pattern over the screen.
Black looks for example gray and looking at the display can cause a bit of discomfort due to the scanline continuously moving. Is there something wrong with the display or is it because the display isn't tested yet since it says that in the code. Any help would be appreciated! :)

prenticedavid, 2017-09-02:
Please post a link to your actual display.  e.g. Ebay sale 

Run the graphictest_kbv.ino from the Examples.
Either post photos, video of problems.
Or describe in words.    e.g. BLUE Band is Red, Grey, Striped, ... .   GREEN Band text is mirrored...

The first step is to check that every part of graphictest_kbv works.   i.e. colours, directions, shapes, ...
Adjusting scanlines,  power settings,  VCOM, ... really needs photo or video, 

You can always post on the Displays Topic of the Arduino.cc Forum.

Yes,  I would appreciate any testing.   I can always create a "test_1580" Branch on GitHub.

David.

CrazyVito11, 2017-09-02:
Tonight i'll make a video of it and send a link to the display, i can not garantee you That you will receiver the exact same as mine since i ordered it from Aliexpress and there was Almost no information gives about the display. They didnt even said what chip it Used so i had to figure That out first. Great library btw! I have Used it before and it is really nice! :) 
Oh and i have no problems with testing it for you!

CrazyVito11, 2017-09-02:
i made 3 videos, one is stable but sometimes looses focus but you can really see the scan lines there and the other one was from my phone since it can lock AF and AE you can see there really good that stuff sometimes gets scrambled and one where the picture quality is compared to a ILI9325. Also something you sadly couldn't see on the video is that black kinda looks more like grey. Link 1: https://drive.google.com/file/d/0B8-gcM32QVWoS0VNTzRpeWtxZTA/view?usp=sharing
Link 2: https://drive.google.com/open?id=0B8-gcM32QVWoY2k1QlhHU3RYenM
Link 3: https://drive.google.com/open?id=0B8-gcM32QVWoYXd3aUxISzZOWE0

Link to the display i bought: http://s.aliexpress.com/2iyuU3Mn (please note that if you buy this you have no idea if you'll receive the exact same one, the chip ID was never ever given so i had to figure that out on my own. i bought a pair these displays before and one of them was a ILI9325 and the other one a ILI9320. So the chance you'll also receive a R61580 is small.

prenticedavid, 2017-09-02:
The graphic tests all work as expected except for readPixel()
The testCard comparison is very useful.   The power, VCOM, Frame rates are all wrong.

I will have a think.   And post a test_1580 Branch on Sunday or Monday.

David.

prenticedavid, 2017-09-04:
I have created a test_1580 Branch.   Please compare with the previous one.
You just need to comment / uncomment the required init_table16(R61580_ ... ) line.

Please try forcing tft.begin(ID) with 0x9320, 0x9325 and see if the display is better.   
Do NOT worry about colours or directions.  Just the steadiness, quality of display.

David.

CrazyVito11, 2017-09-04:
Hey! it looks really good now! :)
i didn't understand what you meant with "You just need to comment / uncomment the required init_table16(R61580_ ... ) line." so i skipped it and also i think you meant 0x1580 instead of 0x9320 in "Please try forcing tft.begin(ID) with 0x9320, 0x9325"
But it works! and looks really good! Thanks for the help man!
I really like this library that you made, its easy to use and was the only library that worked with my ILI9325. So again thanks! :) 👍

prenticedavid, 2017-09-04:
I had configured the Branch especially for you.

You still have the readPixel() error.    Please remove the READ_BGR attribute.  
Are all the colours and directions still correct? 

Please do not close the issue until we have resolved it properly.

David.

CrazyVito11, 2017-09-06:
Oops! Sorry for closing it to quickly and for not responding in a few days :/
i've been busy the last few days so i'll go and check it out now

prenticedavid, 2017-09-11:
I would like to merge any improvements for R61580 to the Master and close this test branch.
1.  does the Software Scroll work now?  Or do you still get a readPixel() error?
2.  are the colours good?  e.g. do the graduation in Colour BANDs look smooth?

CrazyVito11, 2017-09-11:
1. I am not sure, i do still get the message readPixel() should be 24 bits, but i have included a link to the video.
2. The colors look really good! The Colour BANDs did look good to me, even tho the video doesn't really show that. so you'll just have to take my word for it

Link to video: https://drive.google.com/open?id=0B8-gcM32QVWoOXljX1NtazhkNmM

prenticedavid, 2017-09-11:
Please can you 
1.  Run the readpixel_kbv example.   Take a photo of one black background screen.  And one blue background screen.

2. then edit your MCUFRIEND_kbv.cpp file:
`
#ifdef SUPPORT_1580
    case 0x1580:
        _lcd_capable = 0 | REV_SCREEN | READ_NODUMMY | INVERT_GS; //thanks vanhan123

`

Run the graphictest_kbv with the NODUMMY attribute.   Report the readPixel() message.
Run the readpixel_kbv and take another photo.

David.

prenticedavid, 2017-09-14:
Bump.   Please try those two steps.   Or ask if you don't understand.

I would like to resolve any remaining R61580 problems and close this Branch.
Then I can merge into the Master Branch for a new Release.

David.

CrazyVito11, 2017-09-14:
I am receiving an error while compiling :/
`C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp: In member function 'void MCUFRIEND_kbv::begin(uint16_t)':

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1132:9: error: expected primary-expression before '}' token

         };

         ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1195:9: error: break statement not within loop or switch

         break;

         ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1401:5: error: case label '17717' not within a switch statement

     case 0x4535:

     ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1445:9: error: break statement not within loop or switch

         break;

         ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1448:5: error: case label '21264' not within a switch statement

     case 0x5310:

     ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1458:9: error: break statement not within loop or switch

         break;

         ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1461:5: error: case label '26644' not within a switch statement

     case 0x6814:

     ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1471:9: error: break statement not within loop or switch

         break;

         ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1475:5: error: case label '30595' not within a switch statement

     case 0x7783:

     ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1520:9: error: break statement not within loop or switch

         break;

         ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1523:5: error: case label '30601' not within a switch statement

     case 0x7789:

     ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:1555:9: error: break statement not within loop or switch
[... 247 lines cut ...]
     case 0xB509:

     ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:2671:9: error: break statement not within loop or switch

         break;

         ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp: At global scope:

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:2708:5: error: '_lcd_rev' does not name a type

     _lcd_rev = ((_lcd_capable & REV_SCREEN) != 0);

     ^

C:\Users\Sander\Documents\Arduino\libraries\MCUFRIEND_kbv-test_1580\MCUFRIEND_kbv.cpp:2709:5: error: expected unqualified-id before 'if'

     if (table8_ads != NULL) {

     ^

exit status 1`

CrazyVito11, 2017-09-14:
Also i have 1 display that is acting, strange...
When the arduino is writing to the display everything is fine, but suddenly everything just starts flipping when it is done writing.
I have included a few links to the pictures and a video.
https://drive.google.com/file/d/0B8-gcM32QVWoTUVuZGV6VkEzMWc/view?usp=sharing
https://drive.google.com/file/d/0B8-gcM32QVWoVTcyeWt0cVJOczQ/view?usp=sharing
https://drive.google.com/file/d/0B8-gcM32QVWoWEhLRUNoSTc3RzA/view?usp=sharing

prenticedavid, 2017-09-14:
I have just built from the test_1580 Branch as it appears on GitHub.    It compiled without Errors.    There are always some Warnings.

I suggest that you close the IDE, delete the current MCUFRIEND_kbv directory from your libraries folder.
Download ZIP from GitHub.   Open the IDE.   Install Library from ZIP.

This should cure your Build "problems".    Run the readpixel_kbv sketch again.    Does it become unstable?

Which statement gives a steady display?
`
        init_table16(R61580_DEM240320C, sizeof(R61580_DEM240320C));

//        init_table16(R61580_regValues, sizeof(R61580_regValues));
`
i.e. swap the double-slashes from one line to the other.    This makes one statement active.  And the other statement a comment.

Your Black Background photo shows the correct output,
The Blue Background is reading the Blue pixels as 0x0703 instead of 0x001F.    

Add the  READ_NODUMMY attribute.   Hint.   The following _case 0x1963:_ controller has this attribute.

Ask if you are unsure.   Oh, if you get hundreds of errors,  you only need to post the first one or two.

David.

CrazyVito11, 2017-09-14:
it only went wrong when i did step 2 of what you said
"then edit your MCUFRIEND_kbv.cpp file:
`
#ifdef SUPPORT_1580
case 0x1580:
_lcd_capable = 0 | REV_SCREEN | READ_NODUMMY | INVERT_GS; //thanks vanhan123"

prenticedavid, 2017-09-14:
I have put the suggested edit on GitHub.   Please compare with what edit(s) you did.

I can't believe that adding the attribute would make your controller unstable.
Likewise,  your photos show a bad screen.   i.e. before you made the edit.

When you attach a photo or video,  please describe exactly what you did.
i.e. Build directly from GitHub.   Run readpixel_kbv.   Take photo.  Unstable screen.
i.e. Download from GitHub. Make READ_NODUMMY edit.  Build.  Run readpixel_kbv.   Take photo.  Unstable screen.

David.

CrazyVito11, 2017-09-14:
It probably is a bad screen since the other one doesn't have that issue, but it is weird that it only glitches when it isn't updating the screen.

CrazyVito11, 2017-09-14:
it reports 0xF8F0 on the first scroll, the other ones after that just say that it should be a 24 bit value
> Run the graphictest_kbv with the NODUMMY attribute. Report the readPixel() message.

prenticedavid, 2017-09-19:
Please can you provide an accurate report.   e.g. 
1.  what GitHub Branch and date 
2.  any edits you have made
3.  what you expected.   what result you actually got.

If you have cloned the GitHub project,  you can see exactly what edits have been made.
If you are just copy-pasting from the GitHub website,  you need to "remember" what you have done.  And most importantly "tell other members"
The whole purpose of GitHub is to automate the procedure.

I would like to get this R61580 problem resolved.   Then I can make a new Release.

David.

unclef1, 2018-10-10:
Hello! Id=0x1602. Manufacturer 0x3880 All test, color and scroll OK. readPixel=0x00. WritePixel- command ili9341 OK.
in .cpp im write:
uint16_t Adafruit_TFTLCD::readPixel(int16_t x, int16_t y) {
CS_ACTIVE;  
   uint8_t hi, lo;
    int16_t t;
 setAddrWindow(x, y, _width-1, _height-1);
      CD_COMMAND;
 write8(0x2E);
      setReadDir();  // Set up LCD data port(s) for READ
      CD_DATA;
 setReadDir();  
delayMicroseconds(50);setReadDir();  
      read8(hi);     // First 2 bytes back are a dummy read
      read8(hi);
      read8(hi);     // Bytes 3, 4 are actual pixel value
      read8(lo);
      setWriteDir(); // Restore LCD data port(s) to WRITE configuration
    CS_IDLE;
    return ((uint16_t)hi << 8) | lo;
}

prenticedavid, 2018-10-10:
I am not sure which shield or controller you are talking  about.

I have never owned or even seen an ID=0x1602.   I have to rely on accurate reports from users.

Please state which Branch or Tag version of the library you are using.   e.g. Release v2.9.8 from IDE Library Manager
Then run graphictest_kbv example on a regular Uno, Mega, Zero, ...
Report anything that does NOT work properly.  e.g. Band Scroll directions,  Software Scroll, ...
Do you get a readPixel() error ?

David.

unclef1, 2018-10-11:
MCUFRIEND 2.99 UNO  oll tesl OK.
PORTRAIT_REV 
VERTICAL SCROLL UP  
VERTICAL SCROLL DN
PORTRAIT
ONLY THE COLOR BAND
readPixel() read 0x0
LANDSCAPE_REV 
SCROLL UP 
SCROLL DN 
LANDSCAPE
SCROLL DESABLED
readPixel() read 0x0

unclef1, 2018-10-11:
All Test OK

unclef1, 2018-10-11:
v 2.98 v 2.99 All Test OK. no readPixel() error. Arduino UNO.  readPixel() read 0x0. 

> I am not sure which shield or controller you are talking  about.
> I have never owned or even seen an ID=0x1602.   I have to rely on accurate reports from users.
> Please state which Branch or Tag [...]

prenticedavid, 2018-10-11:
If "Software Scroll" works you should never see a readPixel() message.
You should see a yellow "SOFTWARE SCROLL" message move across the screen.

Are you saying that v2.9.8 Release works ok and that the v2.9.9 Beta has broken 0x1602 readPixel() ?

I am very grateful for your feedback.    It is the only way that I can fix a controller that I do not own.

David.

unclef1, 2018-10-11:
CrazyVito11 at IMG_0939.MOV see "readPixel() should be 24-bit" 
I see "readPixel() read 0x0"

unclef1, 2018-10-11:
At Adafruit_TFTLCD library I see readPixel() 0x00. Bat in this library no ID 1602 and no readPixel fo ILI9341

unclef1, 2018-10-11:
My code readPixel() also read 0x0

unclef1, 2018-10-11:
This code return ID 0x1602
uint16_t Adafruit_TFTLCD::readPixel(int16_t x, int16_t y) {
CS_ACTIVE;
uint8_t hi, lo;
setAddrWindow(x, y, _width-1, _height-1);
CS_ACTIVE;

CD_COMMAND;
write8(0xD3);
setReadDir(); // Set up LCD data port(s) for READ
CD_DATA;
setReadDir();
delayMicroseconds(50);setReadDir();
read8(hi); // First 2 bytes back are a dummy read
read8(hi);
read8(hi); // Bytes 3, 4 are actual pixel value
read8(lo);
setWriteDir(); // Restore LCD data port(s) to WRITE configuration
CS_IDLE;
return ((uint16_t)hi << 8) | lo;
}

This code return 0x000
uint16_t Adafruit_TFTLCD::readPixel(int16_t x, int16_t y) {
CS_ACTIVE;
uint8_t hi, lo;
setAddrWindow(x, y, _width-1, _height-1);
CS_ACTIVE;

CD_COMMAND;
write8(0x2E);
setReadDir(); // Set up LCD data port(s) for READ
CD_DATA;
setReadDir();
delayMicroseconds(50);
read8(hi); // First 2 bytes back are a dummy read
read8(hi);
read8(hi); // Bytes 3, 4 are actual pixel value
read8(lo);
setWriteDir(); // Restore LCD data port(s) to WRITE configuration
CS_IDLE;
return ((uint16_t)hi << 8) | lo;
}

unclef1, 2018-10-11:
case 0x1602:
        _lcd_capable = AUTO_READINC | MIPI_DCS_REV1 | MV_AXIS | INVERT_GS | READ_24BITS; //thanks Dumper
Please tell me, Dumper could see readPixel? 
May this processor have another reading command?

unclef1, 2018-10-11:
v2.9.6 v2.9.8 v2.9.9 not broken 0x1602 readPixel(). It is no readin. 
Maybe the processor does not support reading memory.

> If "Software Scroll" works you should never see a readPixel() message.
> You should see a yellow "SOFTWARE SCROLL" message move across the screen.
> Are you saying that v2.9.8 Release works ok and that [...]

prenticedavid, 2018-10-11:
> 
> 
> CrazyVito11 at IMG_0939.MOV see "readPixel() should be 24-bit"
> I see "readPixel() read 0x0"

Thanks for the information.    Please edit the readGRAM() method.

`		if (_lcd_ID == 0x1511 || _lcd_ID == 0x1602) READ_8(r);   //extra dummy for R61511 and 0x1602
`

I presume that LCD_ID_readreg sketch works ok for you.
I presume that graphictest_kbv shows ID=0x1602 on Serial and also in Adafruit_Tests page.

Please try readpixel_kbv example.    Experiment with extra dummy READ_8()

Your posts seem to quote Adafruit_TFTLCD when this library is called MCUFRIEND_kbv.
As far as I know,  Adafruit does NOT support 0x1602.  

David.

unclef1, 2018-10-12:
Adafruit does support 0x1602 as 0x9341 but Adafruit readPixel does NOT support ID 0x9341.
Thanks for the information. I edited the readGRAM() method.
if (_lcd_ID == 0x1511 || _lcd_ID == 0x1602) READ_8(r); //extra dummy for R61511 and 0x1602
No resalt. 0x0.

> Thanks for the information.    Please edit the readGRAM() method.
> if (_lcd_ID == 0x1511 || _lcd_ID == 0x1602) READ_8(r); //extra dummy for R61511 and 0x1602
> I presume that LCD_ID_readreg sketch [...]

unclef1, 2018-10-12:
in readpixel_kvb I changed
   tft.reset();
  //  g_identifier = tft.readID();
  g_identifier = 0x5310;//0x6814;//0x7789;//0x8357;//0xAC11//0x9302;//0x9338;//0x9329;//0x1511;
9 ID shows result. 0x0. ID 0x9341 and other does not work. My Adafruit_TFTLCD  made for ID 0x1602 and  
SPFD5408

prenticedavid, 2018-10-12:
> Please try readpixel_kbv example. 

If it just displays 00 everywhere,  this would prove that readGRAM() does not work.
If it does show something,   please either post a photo or simply type one complete line from the display.

Some controllers can only read ID and GRAM but no other registers e.g. SSD1289, SSD1297, ...
Other controllers can read ID but not read GRAM e.g. some 132x162 controllers from Himax
Some shields do not access the LCD_RD pin and can never read ID or registers.

Since your "0x1602" can read ID and read registers,   I would expect it to read GRAM too.
You can edit the LCD_ID_readreg example to read reg(0x2E) and reg(0x3E)

It is quite possible that it needs extra dummy reads or possibly no dummies at all.

I think that we have a language or translation problem.    You make references to Adafruit which is nothing to do with me.    I simply inherit the Adafruit_GFX class for graphics.

David.

unclef1, 2018-10-12:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)
reg(0x0000) 00 00 ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 38 80 00 Manufacturer ID
reg(0x0009) 00 00 61 00 00 Status Register
reg(0x000A) 08 08 Get Power Mode
reg(0x000C) 06 06 Get Pixel Format
reg(0x002E) 00 00 00 00 color pixel
reg(0x003E) 00 00 00 00 color pixel next
reg(0x0061) 00 00 RDID1 HX8347-G
reg(0x0062) 00 00 RDID2 HX8347-G
reg(0x0063) 00 00 RDID3 HX8347-G
reg(0x0064) 00 00 RDID1 HX8347-A
reg(0x0065) 00 00 RDID2 HX8347-A
reg(0x0066) 00 00 RDID3 HX8347-A
reg(0x0067) 00 00 RDID Himax HX8347-A
reg(0x0070) 00 A6 Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00 RD_DDB SSD1963
reg(0x00B0) 00 00 RGB Interface Signal Control
reg(0x00B4) 00 02 Inversion Control
reg(0x00B6) 00 24 02 00 00 Display Control
reg(0x00B7) 00 00 Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00 ILI9481, HX8357-B
reg(0x00C0) 0A 0A 0A 0A 0A 0A 0A 0A 0A Panel Control
reg(0x00C8) 40 40 40 40 40 40 40 40 40 40 40 40 40 GAMMA
reg(0x00CC) 00 00 Panel Control
reg(0x00D0) 00 00 00 Power Control
reg(0x00D2) 00 00 00 00 00 NVM Read
reg(0x00D3) 00 01 16 02 ILI9341, ILI9488
reg(0x00D4) 00 00 00 00 Novatek ID
reg(0x00DA) 38 38 RDID1
reg(0x00DB) 80 80 RDID2
reg(0x00DC) 00 00 RDID3
reg(0x00E0) 00 22 23 25 08 10 14 40 7B 50 0B 1B 22 20 2F 37 GAMMA-P
reg(0x00E1) 00 0C 14 23 0E 14 15 36 59 46 0B 1F 27 1F 20 22 GAMMA-N
reg(0x00EF) 00 00 00 00 00 00 ILI9327
reg(0x00F2) 00 24 04 04 1E 00 00 00 00 00 00 00 Adjust Control 2
reg(0x00F6) 00 00 00 00 Interface Control

> If it just displays 00 everywhere,  this would prove that readGRAM() does not work.
> If it does show something,   please either post a photo or simply type one complete line from the display.
> Some [...]

unclef1, 2018-10-12:
readReg(0x2C, 4, "wr color pixel ");
reg(0x002C) 00 00 00 00 wr color pixel
I went through all the commands 00...FE and read 4 bytes with black and white screen.
Сommand F0 gives all the time a different result.

unclef1, 2018-10-12:
Of course it would be easier for me to communicate in Russian. That google translated and write :)

unclef1, 2018-10-12:
No. Сommand F0 gives A3. 
In the loop, something was floating.

unclef1, 2018-10-12:
Can software reset?

unclef1, 2018-10-12:
Switched to your library. Programm size MCUFRIEND_kbv-26410 Adafruit_TFTLCD-27568. It is better.

prenticedavid, 2018-10-12:
> If it just displays 00 everywhere, this would prove that readGRAM() does not work.
> If it does show something, please either post a photo or simply type one complete line from the display.
>
> Some [...]

No,  I do not understand Russian.   Google Translate works well.    If you do not understand something that I have written,  just tell me e.g. quote my English sentence.

"readpixel_kbv.ino" example should show whether the 0x1602 controller can read pixels   i.e. readGRAM()

David.

prenticedavid, 2018-10-21:
"readpixel_kbv.ino" example should show whether the 0x1602 controller can read pixels i.e. readGRAM()

David.

Allanio, 2019-05-28:
I also happen to have this board.

I have edited the MCUFRIEND_kbv.cpp file edited the readGRAM() method.

When I compile readpixel_kbv I still see 0000 everywhere.
 
> Thanks for the information. Please edit the readGRAM() method.
>
> ` if (_lcd_ID == 0x1511 || _lcd_ID == 0x1602) READ_8(r); //extra dummy for R61511 and 0x1602 `
>
> I presume that LCD_ID_readreg sketch [...]

Allanio, 2019-05-28:
A video of graphictest_kbv

https://youtu.be/uU22TNofWX8

prenticedavid, 2019-05-28:
Yes,  I have a Red 0x1602 Shield.    You can read several registers but it does not appear to read GRAM.

I have no idea what manufacturer or any data sheet.
It does seem odd that it does "most things" for a MIPI-style but not read GRAM.   So I suspect that there is a register bit that might control the behaviour.    It is possible (but unlikely) that it uses a different readGRAM command to the regular MIPI 0x2E

However,  without a datasheet I am in the dark.
I can only suggest that you ask the Vendor for a datasheet or part number.    I would then try to fix it.

Incidentally,  CrazyVito11 started this issue with a 0x1580 (Renesas R61580).
unclef1 continued with 0x1602.   you have a 0x1602

David.

closed 2021-08-24 by prenticedavid