prenticedavid/MCUFRIEND_kbv #18: problem with ID = 0x1505

ksvdeluxe, 2017-09-22:
Hi ... after the update 2.9.7 does not work Touch_shield_kbv
on the beta version worked with editing Ravi_kanchan

prenticedavid, 2017-09-23:
Please be specific,
Does graphictest_kbv work ?
Does TouchScreen_Calibr_native work on the Uno ?
What does Touch_shield_kbv do?  What did you expect?

Note that any Release will reset any "special defines" back to a default state.
The 0x1505 should work on a Uno.
Your particular SPECIAL for your STM32 is not in the Release.
You will need to add it yourself.    If unsure,   I can always do it for you in a new "test_ravi" Branch

Does the ILI9340 work on both Uno and BluePill now?

David.

ksvdeluxe, 2017-09-23:
graphictest_kbv, TouchScreen_Calibr_native work on the Uno.
rolled back to 2.9.7 beta and in file MCUFRIEND_kbv.cpp uncommented -#endif

        goto common_9320;
    case 0x5408:
        _lcd_capable = 0 | REV_SCREEN | READ_BGR | INVERT_GS;
        goto common_9320;
    case 0x1505:                //R61505 thanks Ravi_kanchan2004. R61505V, R61505W different
    case 0x9320:
        _lcd_capable = 0 | REV_SCREEN | READ_BGR;
      common_9320:
and only works this

ksvdeluxe, 2017-09-23:
no... I am not Ravi_kanchan)))

prenticedavid, 2017-09-23:
Ah-ha.  So you are a different 0x1505 owner.

Your message makes no sense at all.   What #endif ? What Branch?   What date?

At one stage I included a specific initialisation sequence for the R61505U separate from the R61505V and R61505W cases.
Then Ravi said that the R61505U worked 100% with using the 0x9320 sequence.    So I just added as a ILI9320 clone.

If you are running GitHub properly,  you can see what changes have been made and with what comment.

I do not own any R61505 of any version.   I rely on owner feedback.

David.

ksvdeluxe, 2017-09-23:
2.9.7 beta MCUFRIEND_kbv.cpp - 
 _lcd_xor = 0;
    if (ID == 0)
        ID = 0x9341;
    switch (_lcd_ID = ID) {
2.9.7 beta MCUFRIEND_kbv.cpp - 
_lcd_xor = 0;
    switch (_lcd_ID = ID) {
in the absence of this line the reason

prenticedavid, 2017-09-23:
The (ID == 0) was a catch-all that I have never been happy with.

Please run the new "diagnose_TFT_support" sketch.
And paste the result from  "LCD_ID_readreg" sketch.

I would far prefer to identify the controller that you are using.
Which is what I did with my "ID=0" board.  i.e. it is now identified as ID=0x2053.
I still don't know what make it is.

As a kludge,  you can just restore the catch-all statement.

David.

ksvdeluxe, 2017-09-23:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 15 05	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2017-09-23:
Yes,  you have a Renesas R61505U.   It should identify with ID = 0x1505

tft.begin(0x1505) should work exactly like tft.begin(0x9320)  in the new Release.

There is absolutely no way that tft.begin(0x9341) would work with a 9320-style controller (like the R61505U) i.e. the catch-all would never work.

Please post an accurate report.    i.e. what Branch, what date, what sketch, what behaviour.
You have all the facilities on GitHub to specify any particular stage of a Beta version.
Life is clearer with a Release.   It is a fixed entity.

David.

Ankitrawat581, 2017-11-17:
hello david
   I just bought a mcufriend tft. After running the code mcufriend.kbv i got the id as 0x1505.
I am using it for my project to display some buttons.
Can you provide me with library for touch part.
     Thank you

closed 2018-05-16 by prenticedavid