prenticedavid/MCUFRIEND_kbv #19: Compiler error [-fpermissive]

RoelHermus, 2017-09-30:
I use MCUFRIEND library for my 3.5 inch touch display. I only use the screen, not the touch part. I use a simple example to show some  .bmp file on an Arduin Uno. Thats works perfect.

Now I try to change to an ESP32 and exact the same application can't compile and stops with an error.

In detail I get this error:
MCUFRIEND_kbv-master/UTFTGLUE.h:102:26: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]

settextcursor(st.c_str(), x, y); MCUFRIEND_kbv::print(st);}

                               ^
Can you fix this?

prenticedavid, 2017-10-01:
I have made some appropriate casts to keep the ESP32 compiler happy.
Note that the variants folder does not contain a board to match the "Wemos D1 R32" pin assignments.

I have a Protoshield that is hand-wired for the pinout in mcufriend_shield.h
This does not work as reliably as the "D1 R32" board when using the microSD.

Yes,  thanks for checking the UTFTGLUE class.   I do not expect people to write any new programs.
The class is useful for porting a legacy UFT sketch.

If you find any other problems,  please quote which example you are using.
Or post/attach a custom sketch that exhibits the problem.

David.

RoelHermus, 2017-10-02:
Hi David, Thanks so far. I'm happy with your work and it works fine. I've no other problems, but everything is hard to understand.  How to connect the display to the ESP32 is not clear. I found some info someware in your programs, thats all but it helps me. When there is a faults its hard to find out what kind of fault it is. Software or the hardware connetions. Maybe you can improve this point.

prenticedavid, 2017-10-02:
No,   I can't make ESP32 work out of the box.

Regular Arduinos have headers that receive standard Shields.  e.g. Uno, Zero, Mega, Due, ...
Other MCUs are available with standard headers.  e.g. Nucleo (STM32),  Teensy (Freescale) using Sparkfun adapter,  
Only one ESP32 board has Arduino headers.  i.e. the "D1 R32" that can be found on Ebay.

Although the D1 R32 has Analog headers and Digital headers,  the Analog headers are read-only.
Regular Arduino Analog pins can also be used for Digital.   

The only way to achieve Digital on the Analog header is by hard-wiring the adjacent unpopulated holes.
In other words.    You have to use a soldering iron before the shield can work.

If a proper ESP32 board ever appears on the market,   I would support it.

Note that the ESP32 and ESP8266 work very well with SPI displays.
The ESP32 is no better or faster than a Cortex M3 or even an Xmega in parallel mode
The ESP8266 does not have enough pins to ever work as parallel., 

David.

RoelHermus, 2017-10-02:
Hi, David, I do not have to get it out of the box. Only a wiring description, like this, a part of your script. It took me a lot of time to find it. Thanks so far, Roel

 LCD_RD  2  //LED
 LCD_WR  4
 LCD_RS 15  //hard-wired to A2 (GPIO35) 
 LCD_CS 33  //hard-wired to A3 (GPIO34)
 LCD_RST 32 //hard-wired to A4 (GPIO36)
 LCD_D0 12
 LCD_D1 13
 LCD_D2 26
 LCD_D3 25
 LCD_D4 17
 LCD_D5 16
 LCD_D6 27
 LCD_D7 14

SPI SS	  GPIO05
SPI SLK     GPIO18
SPI MOSI  GPIO23
SPI MISO  GPIO19

prenticedavid, 2017-10-02:
Some tips:
1.  search for "ESP32".   this will show where there is any conditional ESP32 code
2.  the pin defines are as intuitive as anyone could dream of.
3.  compare the ESP32 block to the Mega2560 block or Nucleo block.   I find this bit-twiddling complex

What hardware do you intend to use?  Have you got a reliable display yet?
I use soldered wires on a Protoshield or Protoboard.   I would not even think of a breadboard.

You can get away with bad layout on a 16MHz AVR.    A 240MHz processor is more fussy.

David.

artbody, 2018-01-12:
i have the same Problem with an STM32F103RB and MCUfriends 2.4" tft
`home/achim/Arduino/libraries/MCUFRIEND_kbv/UTFTGLUE.h:102:26: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
    settextcursor(st.c_str(), x, y); MCUFRIEND_kbv::print(st);}
                  ~~~~~~~~^~`
`/home/achim/Arduino/libraries/MCUFRIEND_kbv/UTFTGLUE.h:155:10: note:   initializing argument 1 of 'void UTFTGLUE::settextcursor(char*, int, int)'
     void settextcursor(char *st, int x, int y) {
          ^~~~~~~~~~~~~
/home/achim/Arduino/libraries/MCUFRIEND_kbv/UTFTGLUE.h: In member function 'void UTFTGLUE::printNumF(double, byte, int, int, char, int, char)':
/home/achim/Arduino/libraries/MCUFRIEND_kbv/UTFTGLUE.h:123:47: error: 'dtostrf' was not declared in this scope
    char buf[20]; dtostrf(num, length, dec, buf); settextcursor(buf, x, y); MCUFRIEND_kbv::print(buf);
                                               ^
/home/achim/mnt/Electronik/Pflamiku/arduino-micro-waage/micro-v1/pflamiku_2018_01_12_tft_stm/pflamiku_2018_01_12_tft_stm.ino: At global scope:
/home/achim/mnt/Electronik/Pflamiku/arduino-micro-waage/micro-v1/pflamiku_2018_01_12_tft_stm/pflamiku_2018_01_12_tft_stm.ino:37:38: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 UTFTGLUE myGLCD(0x0154,A2,A1,A3,A4,A0);
                                      ^
exit status 1
Fehler beim Kompilieren für das Board Nucleo-64.
`

artbody, 2018-01-12:
with an Arduino Mega2560
there is nearly the same Problem but there is only a warning;
`In file included from /home/achim/mnt/Electronik/Pflamiku/arduino-micro-waage/micro-v1/pflamiku_2018_01_12_tft_stm/pflamiku_2018_01_12_tft_stm.ino:36:0:
/home/achim/Arduino/libraries/MCUFRIEND_kbv/UTFTGLUE.h: In member function 'void UTFTGLUE::print(String, int, int, int)':
/home/achim/Arduino/libraries/MCUFRIEND_kbv/UTFTGLUE.h:104:27: warning: invalid conversion from 'const char*' to 'char*' [-fpermissive]
    settextcursor(st.c_str(), x, y); 
                           ^`
Looks like a bug

prenticedavid, 2018-01-12:
You appear to be using the ST core with probably a NUCLEO-F103

Add #include <avr/dtostrf.h> to UTFTGLUE.h for the ST core

Don't worry about the const char* warnings.   
Or add const as appropriate for anonymous strings

UTFTGLUE can "help" porting a legacy UTFT program
It is preferable to use native GFX methods in any new program.

You get all sorts of problems with UTFTGLUE and the Maple core.   Mostly due to the Maple core not understanding Analog pin numbers like A2

Did you realise that this issue had been closed?  I have re-opened it for you.

David.

artbody, 2018-01-14:
thx for your answer
I'll try to give it another chance at next weekend

artbody, 2018-01-19:
I added #include <avr/dtostrf.h> to UTFTGLUE.h 

but the same Error
`../Arduino/libraries/MCUFRIEND_kbv/UTFTGLUE.h:106:26: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
    settextcursor(st.c_str(), x, y);
                  ~~~~~~~~^~`

prenticedavid, 2018-01-19:
Look at the history of UTFTGLUE.h
There was an edit on 1 October which solves this error.

I built GLUE_Demo_320x240 for a Nucleo-F103 with both ST Core and the MapleCore
I just get warnings for
` UTFTGLUE myGLCD(0x0154,A2,A1,A3,A4,A0);
`
and
`#warning Uno Shield on NUCLEO
`
All arguments to the UTFTGLUE constructor are dummies.   If you replace the first arg with 123 it will stop the warning.

David.

artbody, 2018-01-19:
The 	 history of UTFTGLUE.h
 1 October gives me the following code
```
 void print(String st, int x, int y, int deg=0) {
		 //settextcursor(st.c_str(), x, y); 
		settextcursor((char*)(st.c_str()), x, y); 
		MCUFRIEND_kbv::print(st);}

```
And that results in a bunch of errors

```

/home/achim/Arduino/libraries/MCUFRIEND_kbv/examples/GLUE_Demo_320x240/GLUE_Demo_320x240.ino:26:38: warning: large integer implicitly truncated to unsigned type [-Woverflow]
 UTFTGLUE myGLCD(0x0154,A2,A1,A3,A4,A0);
                                      ^
In file included from /home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:31:0:
/home/achim/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:362:2: warning: #warning Uno Shield on NUCLEO [-Wcpp]
 #warning Uno Shield on NUCLEO
  ^~~~~~~
In file included from /home/achim/.arduino15/packages/STM32/hardware/stm32/2017.8.4/cores/arduino/pins_arduino.h:22:0,
                 from /home/achim/.arduino15/packages/STM32/hardware/stm32/2017.8.4/cores/arduino/Arduino.h:75,
                 from /home/achim/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:5,
                 from /home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.h:10,
                 from /home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:22:
/home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp: In member function 'virtual void MCUFRIEND_kbv::setRotation(uint8_t)':
/home/achim/.arduino15/packages/STM32/hardware/stm32/2017.8.4/variants/NUCLEO_F103RB/variant.h:77:33: error: expected unqualified-id before numeric constant
 #define BOARD_SPI_DEFAULT_SS    10
                                 ^
/home/achim/.arduino15/packages/STM32/hardware/stm32/2017.8.4/variants/NUCLEO_F103RB/variant.h:82:33: note: in expansion of macro 'BOARD_SPI_DEFAULT_SS'
 #define SS                      BOARD_SPI_DEFAULT_SS
                                 ^~~~~~~~~~~~~~~~~~~~
/home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:332:18: note: in expansion of macro 'SS'
     uint16_t GS, SS, ORG, REV = _lcd_rev;
                  ^~
/home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:360:44: error: lvalue required as left operand of assignment
             SS = (val & 0x40) ? (1 << 5) : 0;   //MX
                                            ^
/home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:404:44: error: lvalue required as left operand of assignment
             SS = (val & 0x40) ? (1 << 8) : 0;
                                            ^
/home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:422:44: error: lvalue required as left operand of assignment
             SS = (val & 0x40) ? (1 << 8) : 0;
                                            ^
/home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:442:44: error: lvalue required as left operand of assignment
             SS = (val & 0x40) ? (1 << 8) : 0;
                                            ^
/home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:445:13: error: 'ORG' was not declared in this scope
             ORG = (val & 0x20) ? (1 << 3) : 0;
             ^~~
exit status 1
Fehler beim Kompilieren für das Board Nucleo-64.

```
This error was not easy for me to interpret. However, the answer is actually in the 3 messages: MCUFRIEND_kbv.cpp:332:18: note: in expansion of macro 'SS'
So it is defined twice, with different meanings.

Solution:
Rename of all occurrences of the variable "SS"  with "SS_kbv"
"compiles well"

prenticedavid, 2018-01-20:
My apologies.   Yes,  there was a Commit to MCUFRIEND_kbv.cpp on 8 October.   Rename SS macro for ST core.

If you are a member of GitHub I would expect you to fork or clone the Repository.    Then you can try out a Beta but return to a Release whenever you want.

If you are just a regular Arduino User,   I would expect you to use the IDE Library Manager.    And ask questions on the Arduino Forum

As you have discovered,  when a project is built on many targets you can run into name conflicts  e.g. SS.
I should have known better.

David.

closed 2018-05-16 by prenticedavid