prenticedavid/MCUFRIEND_kbv #20: Issues with ID=0x5408

drnickyoung, 2017-10-11:
Hi,

Firstly thank you and well don with the library, an excellent piece of work.

I am using a TFT Shield on an Uno but getting a few minor issues. The first is that the text appears backwards. Simple fix that you might want to look at, or check to see if others are getting.  Simple solution by changing the following in **MCUFRIEND_kbv.cpp**:

```
case 0x5408:
        _lcd_capable = 0 | REV_SCREEN | READ_BGR | INVERT_GS;
        goto common_9320;
```

to

```
case 0x5408:
        _lcd_capable = 0 | REV_SCREEN | READ_BGR;
        goto common_9320;
```
As you can see my display has ID 0x5408. Second issue is when running **graphictest_kbv** the black appears to flicker, like scan lines. Everything else works fine. Not sure if it helps but the LCD_ID_ReadReg output is:

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 54 08	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) C4 C4 C4 C4	Manufacturer ID
reg(0x0009) C9 C9 C9 C9 C9	Status Register
reg(0x000A) CA CA	Get Power Mode
reg(0x000C) CC CC	Get Pixel Format
reg(0x0061) E1 E1	RDID1 HX8347-G
reg(0x0062) E2 E2	RDID2 HX8347-G
reg(0x0063) E3 E3	RDID3 HX8347-G
reg(0x0064) E4 E4	RDID1 HX8347-A
reg(0x0065) E5 E5	RDID2 HX8347-A
reg(0x0066) E6 E6	RDID3 HX8347-A
reg(0x0067) E7 E7	RDID Himax HX8347-A
reg(0x0070) F0 F0	Panel Himax HX8347-A
reg(0x00A1) E1 E1 E1 E1 E1	RD_DDB SSD1963
reg(0x00B0) F0 F0	RGB Interface Signal Control
reg(0x00B4) F4 F4	Inversion Control
reg(0x00B6) F6 F6 F6 F6 F6	Display Control
reg(0x00B7) F7 F7	Entry Mode Set
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C8) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0	Power Control
reg(0x00D2) D2 D2 D2 D2 D2	NVM Read
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D4 D4 D4	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) DC DC	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control
```

Would appreciate any pointers.
Regards,
Nick

drnickyoung, 2017-10-11:
So having had a play, I get the feeling the flickering is a hardware issue.

prenticedavid, 2017-10-11:
Do you have a Red 2.4" or Blue 2.8" Shield ?
I presume you are using v2.9.7 Release from the Library Manager.

Can you take a video or photo of your screen?
I do not have a SPFD5408.  I have never seen one.    I have to rely on feedback from owners.

Yes,  it is probably necessary to adjust the Power settings, Frame Rate, VCOM,

David.

drnickyoung, 2017-10-12:
Hi David,

Sorry I did a video and forgot to include the link:

[Youtube](https://youtu.be/gY2p1lPKrjo)

Yes using v2.9.7 from the library manager. It is a Red 2.4" shield. Happy to play with timings, just let me know where they are and will see what I can do.

Nick

prenticedavid, 2017-10-12:
The video seems to show the Blue Band in the wrong place at the start of Software Scroll.

Yes,  the display quality is appalling 
The  Power settings are wrong.    

If you are not "using" GitHub,   I will post a sketch on Arduino.cc Displays forum.
If you are using GitHub as a cloned project (as Nature intended),  I can create a test Branch.

David.

drnickyoung, 2017-10-12:
Dear David,

Github is fine, just let me know when it is ready and I will give it a shot and see what happens!

Regards,
Nick

drnickyoung, 2017-10-14:
Just wondering, I have tried a few other libraries and whilst none work at all, they all identify the panel as a "ILI9341". Have I missed something or is it eh same as 0x5408 ?

prenticedavid, 2017-10-14:
Your LCD_ID_readreg report said:

> reg(0x0000) 54 08	ID: ILI9320, ILI9325, ILI9335, ...

So you clearly have a genuine SPFD5408 which is completely different to an Ilitek ILI9341

Most of the 'other' libraries are hacked versions of Adafruit_TFTLCD.    Please post a link to the actual hacked library.    Some contain good code.    There is little point in me looking at the official Adafruit code if you are using a hacked version.

Sorry,  I have not got around to posting a diagnostic sketch.   I would appreciate an explanation for the "unexpected" behaviour on the video.

David.

drnickyoung, 2017-10-14:
Hi David,

Please don't be sorry, absolutly no rush. 

Just been digging around and learning the ins and outs of these little LCDs, registers and what not.  Wanted to test a few other libraries too to see if it would help identify the issue but cannot seem to get a single other one to do anything. All cleared out now and can confirm using original, non-hacked, version of Adafruit_TFTLCD, (https://github.com/adafruit/TFTLCD-Library and https://github.com/adafruit/Adafruit-GFX-Library ).

Nick

prenticedavid, 2017-10-14:
The SPFD5408 is similar to the ILI9320.

The Adafruit_TFTLCD library supports IDs: 0x9325, 0x9328, 0x9341, 0x8357, 0x7575
It does not support ILI9320 but the ILI9320 is "fairly similar" to the ILI9325.

I found a library here: https://github.com/JoaoLopesF/SPFD5408
this is calls itself SPFD5408 but the examples appear to use tft.begin(0x9341)

This and other GitHub libraries all seem to hacked from Adafruit_TFTLCD and use tft.begin(0x9341)

I am gobsmacked by the cavalier approach.  e.g. ignore readID() and just force 0x9341.
This would be acceptable if the controllers had the same style of registers.   However:

The SPFD5408 has 16-bit control and data registers.
The ILI9341 has 8-bit control registers and is MIPI compatible.

David.

drnickyoung, 2017-10-14:
Hi David,

Yep I saw that one. I have looked at several libraries and did note some hardwiring the ID. What I found strange was  JoaoLopesF even takes the time to read and identify the ID but then ignores it!

The last few I tested were the one you mentioned and:

- Adafruit_ILI9341 - Was told this was the controller on the LCD when I ordered it. What you get for going eBay!
- https://github.com/JoaoLopesF/SPFD5408 - white screen / uncommenting ReadID function returns 0x9341
- https://github.com/Smoke-And-Wires/TFT-Shield-Example-Code - works but same problem with line flicker and back to front writing. Reads ID as 0x5408
- https://github.com/telamon/utft - display is not supported

Likewise spotting most based on hacked versions of Adafruit.

I am reading through the data sheets for the display so things might become clearer!
Nick

drnickyoung, 2017-10-15:
Hi David,

I have downloaded/cloned the test_5408 branch today and tested it. Unfortunatly it makes it worse. See Youtube ( [https://youtu.be/03XncveScbw](https://youtu.be/03XncveScbw) ) for the result. I will go back through the datasheet for the LCD controller and see if anything springs to light.

Nick

prenticedavid, 2017-10-15:
Oops.   The Software Scroll is correct.   I had not used the 9320-style screens for a long time.  I have just dug out an old ILI9325 shield.    And it starts Software Scroll on the 64-row shifted screen.

I have added another two init sequences.   You just uncomment the appropriate init_table16(SPFD5408_xxx, ... statement.   Only ONE statement should be active.

Sorry,   I am being idle.   i.e. just hauling inits from Google instead of writing a diagnostic sketch.

Please let me know how each behave.
Incidentally another SPFD5408 owner on the Arduino Forum says his display looks ok. 

David.

drnickyoung, 2017-10-15:
Hi David,

I have tried all three:

```
//init_table16(SPFD5408_espruino, sizeof(SPFD5408_espruino));
//init_table16(SPFD5408_bjornfor, sizeof(SPFD5408_bjornfor));
init_table16(SPFD5408_mbed, sizeof(SPFD5408_mbed));
```

All have the flickering to different amounts.

I have been trawling the Arduino forums and noted one or two saying they have a SPFD5408 and 'xyz' library worked. So far none work properly for me. Just got me wondering if despite what it is reporting, is it a different version of SPFD5408 than has been seen before? Just find it strange others get it working fine but mine doesn't. Or other possibility, a faulty LCD?

Nick

prenticedavid, 2017-10-15:
A controller is a controller.   It should behave like any other silicon chip.  i.e. if it passes QC at manufacture,  it should work 100% for many years.

Yes,  you get silicon revisions.   Serious revisions have a new ID.   Small revisions might produce some noticeable difference in behaviour.

In practice,   these Ebay shields are assembled from surplus components.    That is why they are so cheap.   And also why you get a 10-year old silicon chip in a 2017 purchase from Ebay.     However most of these shields work 100% perfectly.
Also note that you have a silicon chip bonded to a TFT screen.    There are several TFT manufacturers.    The physical characteristics differ.    Hence the different init sequences.    

Regarding _saying they have a SPFD5408 and 'xyz' library worked_.   Do not put much faith in it.    Ebay vendors simply called the shields "SPFD5408" when they contained completely different controllers.   Likewise the "usable pins".    Ebay vendors lie.    Arduino owners _believe what they want to believe_.

David.

drnickyoung, 2017-10-15:
Hi David,

Thanks for the explination was just trying to rule out a sub version. I have not gone through the controller data sheet thoroughly yet so thought perhaps each ID has a version too. Now I know not. I cannot physically check the LCD to see if there is anything printed on the rear to help ID it as I am likely to break it.

I will have a play with the power and other setting later in the week and see if I can get it sorted.

Nick

prenticedavid, 2017-10-16:
Have you tried the Shield on a different Uno, Mega, Due, ... ?
Do you get the same behaviour?

Please can I post a link to your video(s) on the Arduino Forum.
It is more likely to reach other SPFD5408 Shield owners that way.

Where are you?   Perhaps we could mail shields to each other (if in Europe).
You could PM me on the Arduino Forum.

David.

drnickyoung, 2017-10-16:
Hi David,

Yes I have tried the LCD on a couple of other Uno's but same result. Sorry I should have mentioned that.  Afraid not got a Due or Mega. Have not looked at testing it with the Pi yet.

By all means share the youtube videos.

Nick

zoomx, 2017-10-16:
Maybe it is defective?

drnickyoung, 2017-10-16:
@zoomx I did think that and mentioned it earlier in the comments. However altering the power registers does seem to affect it and improve it. Not knowing the details of the LCD it is difficult to know exactly what settings should be used so assuming it is not defective it is a lot of guesswork.

Nick

prenticedavid, 2017-10-20:
I am pretty certain that you have a defective display.    A recent issue from BNorman implies that the 0x5408 displays that are currently on the Ebay market are ok.

I did offer to mail a "good shield" to you if you want to send me your 0x5408 shield.
If you are in the UK,   the shield fits in a 25mm cardboard mailer and can go as a "Large Letter" for £0.76 (or £0.98 1st class).      PM me with your postal address and I will send you a working 2.4" ST7781 shield as a free gift.

If you are outside of Europe,  it is not worth the time and postage.    Mind you,   Royal Mail provide an excellent service.    I sent a packet to Iran from UK which arrived in 6 days (5 working days)

David.

drnickyoung, 2017-10-20:
Dear David,

I will send details over to you tonight. I working display would be appreciated and yes I am more than happy to send you mine. I have not made much progress, some settings seem to improve the scan lines, but not having found much out about it comming to the same conclusion. Anyway, will send you details tonight.

Nick

drnickyoung, 2017-10-23:
Dear David,

See the private message feature seems to have gone. Can you send me your details via www.nickbits.co.uk/contact and I will get it in the post to you.

Nick

Kulbhushan-Chand, 2018-04-12:
If I may add, (not very much related to your scan lines problem), I too faced  hardware issue with display.
I have two identical arduino 2.4 inch TFT modules. One of them is not displaying BLACK color properly, and instead gives DARK BROWN. I have narrowed down the problem to be of hardware not software, by powering up the TFT display with 3.3V only and nothing connected to the any other pins (no even arduino) and facing the same problem.

closed 2018-12-04 by prenticedavid