prenticedavid/MCUFRIEND_kbv #21: Display flipped horizontally

BNNorman, 2017-10-18:
Ok, just received my mcufriend 2.4" 240x320. Plugged it into my MEGA2560. Downloaded the mcufriend library. Compiled the test card - displays on the TFT BUT everything is flipped left/right. I.e. it draws/prints from right to left. I see there's an option to rotate the display but these just, well, rotate the display.
Is there a switch option to correct this or have I bought a duffer? (It was very cheap from AliExpress)

BNNorman, 2017-10-18:
Nevermind - found the answer in one of your posts. My screen reports ID=0x5408 but using tft.begin(0x9320) the text now displays correctly.

prenticedavid, 2017-10-18:
The mirroring is fixed in the current v2.9.8-Beta.   It will appear in the next Release (as used by Library Manager)
There is a message on Arduino Forum with links to some videos on [https://forum.arduino.cc/index.php?topic=366304.msg3449953#msg3449953](url)
This refers to the current GitHub Issue from drnickyoung.

Do you get a clear display with 0x9320?   Or with 0x5408 when you remove the INVERT_GS attribute ?

David.

BNNorman, 2017-10-18:
Not sure, was just looking at text. Its late now,will check it out tomorrow

BNNorman, 2017-10-19:
The display looks the same and is clear with both 0x9320 (text correct) and with 0x5408 (text wrong). 
Defining INVERT_GS as 0 in the MCUFRIEND_kbv.cpp file and with ID=0x5408 the text is correct and the display still looks clear (the same in each case).

GLUE_demo_320x240 (no changes) also now displays with correct orientation with INVERT_GS=0

regards
Brian

prenticedavid, 2017-10-20:
You should not alter the INVERT_GS define.   Just look at the case 0x5408 in your "Release" and in the current "master" Branch.
The purpose of GiHub is to show exactly what changes have been made to a project.

Hint.   You do not need to run GitHub or even be a member.   You can view any file from any Branch from your Browser.    Then use your Browser to search for 0x5408.  Compare with the Release code on your PC.

Thanks for the feedback about "clearly drawn display".    I suspect that drnickyoung has got some bad hardware.

David.

BNNorman, 2017-10-20:
Ok, I don't code Arduino very often. Couldn't resist the low low price of
this TFT. I'll do as you suggest when i get a chance.

gazialankus, 2017-12-15:
I had the same situation as @BNNorman with this: https://www.aliexpress.com/item/Free-shipping-LCD-module-TFT-2-4-inch-TFT-LCD-screen-for-Arduino-UNO-R3-Board/32583491591.html
I can confirm that 

- tft.begin(0x9320) works with the version that comes from the Arduino IDE Library
- define INVERT_GS 0 without the above does not help
- Github latest version works without any changes

Thank you for your help

prenticedavid, 2017-12-15:
Thanks for your Pull request.
You have discovered an appalling typo.    It should have been Serial.begin(9600)

I have put your Calibration into the Beta.

I am intending to change the SwapXY approach.   And encourage people to use the XM, YP pins as diagnosed by my "native" sketch.

David.

closed 2018-05-16 by prenticedavid