prenticedavid/MCUFRIEND_kbv #22: wifiMulti.run() stops the touch screen

RoelHermus, 2017-10-22:
Hi David, If I add Wi-Fi support, the touch screen stops. In details, the command wifiMulti.run () is the root cause.
Details in the attached file with your own example. I hope you can solve this problem. I tryed with Wifi.h, but it gives the same problem. Roel
[TouchScreen_Calibr_native.txt](https://github.com/prenticedavid/MCUFRIEND_kbv/files/1404840/TouchScreen_Calibr_native.txt)

prenticedavid, 2017-10-22:
You can run the Mcufriend Uno shields on an ESP32.   It requires some minor hardware mods to the TTGO D1 R32 board.    This is the only ESP32 board that receives Arduino shields.    Anything else must be hand-wired.
Whatever you do,  you end up using most of the GPIO pins for the TFT.

I am not familiar with Wifi.h.   If it shares any pins,   you will need to restore the pinMode.   e.g. just like you have to with the TouchScreen.h functions.

Likewise,   when you have multiple SPI slaves that use different modes,   you should use SPI.beginTransaction() when you "change" devices.

David.

RoelHermus, 2017-10-22:
Hi David, I coated the TFT shield with a Nodemcu32S. Everything works perfectly. The tft screen, sd card and touchscreen. No problems so far. I used the example on your github. I have changed it and added Wi-Fi support. After that it still works, accept the touch screen. After a cup of coffee and some time I found out that wifiMulti.run () closed the touch screen. When I send it, the touch screen will work. It's now wifi or touch screen. But both is what I want.
I found some links on internet with discussions about that issue, but I have not enough ervaring and details to correct that in your software. I hope that you can help me. (My English is poor, excuse me for that point).

RoelHermus, 2017-10-22:
I reading more about wifi and adc.  Adc2 totaly used for WIFI when it switch on. Adc1 is useless for external input. It is imposible to use the touch screen with wifi on. Not realy clever to make chips with this shortcomming. So far so good, time to close this issue. I learned a bit today! Thanks for reading. Roel

prenticedavid, 2017-10-22:
The ESP32 has several ADC pins.    If the WiFi library has claimed two ADC pins,   you will need to choose alternative pins for TFT/Touch.    If shared,  you need to re-configure before each use. 

Note that TFT/Touch only needs two Analog pins but these pins must be capable of Digital too.

David.

RoelHermus, 2017-10-22:
Hi David, indeed, 2 adc pins for the touch screen, I try to change it, but without result.  Wifi claims all the  adc 2 channels. There is nothing left. I was confused about it, but it looks true. Bit stupid! Any idea to solve it? An XPT2046 chip?

RoelHermus, 2017-10-23:
Hi David, I made an solution. After reading again all the docs and with all your replies I found a solution!
Pin 32 and 33 are a part of ADC1 and I exchanged them with 4 and 15. After that I made changes in your software. That was all. Hereby a text file with my changes.  Roel

[touch screen changes.txt](https://github.com/prenticedavid/MCUFRIEND_kbv/files/1406819/touch.screen.changes.txt)

closed 2018-05-16 by prenticedavid

khvalera, 2024-04-07:
I encountered the same situation: when I connect WIFI, the touchscreen stops working! Why hasn't the author fixed the problem yet?

khvalera, 2024-04-09:
It looks like I burned the display with the above example :(