prenticedavid/MCUFRIEND_kbv #23: Display cropped, 3.97 inch

baettigp, 2017-10-27:
Hi, I have a (written on the back): "3.97" TFT LCF for arduino uno with spiflash for font ic".
It looks like a (about) 16:9 display but is detected as 
`
tft.readID() finds: ID = 0x9488

PORTRAIT is 320 x 480
`

Unfortunately, the vertical resolution seems to be less than 320, as everything seems cropped in the test programs. (I tried to add #define SUPPORT_9488_555 to the ino-files, but this seems unrelated to my problems). 
Pictures can be found here:
https://www.dropbox.com/sh/i60duov0t5ubmvn/AADFHS_o1blnf7aBdbpqId1Ga?dl=0&m=

Thank you very much for your help.

Best regards,

Pio

prenticedavid, 2017-10-27:
Your display panel looks like a 272x480.

The "diagnose_controller" sketch detects an ILI9488 (which is a 320x480 controller)
1.  draws a white rectangle 320x480 in PORTRAIT or 480x320 in LANDSCAPE.
2.  and a smaller rectangle 256x376 inside it.

You can see that the PORTRAIT width is smaller than 320.
If you are in PORTRAIT,   add 24 to the X coordinate.   i.e. HOME is 24,0
If you are in LANDSCAPE, add 24 to the Y coordinate. i.e. HOME is 0,24

I am 99.9% certain that the LCD_ID_readreg will report the default 320x480 reset values.
If the library could detect your "letterbox shape" panel,   I could apply the X correction for you.    Meanwhile,  you will need to do it yourself.    A bit annoying for text that "auto linefeeds"

272x480 panels are fairly unusual.
240x240 round panels have a a similar feature but in the GateScan direction.

David.

baettigp, 2017-10-27:
Hi David,

thank you very much for your help. Here is the output from LDC_ID_readreg:

`Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 02 02 3B 3B	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E	Panel Control
reg(0x00C8) 00 B0 B0 B0 B0 B0 B0 B0 B0 B0 B0 B0 B0	GAMMA
reg(0x00CC) 00 03	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 94 88	ILI9341, ILI9488
reg(0x00D4) 00 97 00 00	Novatek ID
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 00 00 07 0C 05 13 09 36 AA 46 09 10 0D 1A 1E 0F	GAMMA-P
reg(0x00E1) 00 00 20 23 04 10 06 37 56 49 04 0C 0A 33 37 0F	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 58 04 12 02 22 22 FF 0A 90 14 88	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
`
I am not entirely sure what to look for (or what the standard 320x480 values would be)?

Thank you very much.

Pio

prenticedavid, 2017-10-27:
That looks like normal ILI9488 reset values as described in the datasheet.

Can you live with the X adjustment?
You are the first person with a 272x480 panel with ILI9488.
Others have had a 272x480 panel with RM68140

If many owners pop up,  I could think about a library method.   It would add size and performance penalties.

David.

baettigp, 2017-10-28:
Thank you very much for your help. Is it possible to do this globally (in the library somewhere with a #define x_min=24 or something like that or does every instance of "x" need to be taken care of by hand?

Thank you very much for your help. 

Pio

prenticedavid, 2017-10-29:
I have added a "test_480x272" Branch on GitHub.

Please test it and report back.   If the "OFFSET_LETTERBOX" is defined,  it cripples all the good 320x480 displays.

David.

baettigp, 2017-10-29:
Just a quick report before I need to leave, graphicstest_kbv and Touch_shield_kbv of the new branch work and seem to display in the correct area.

(only the x-z orientation of the touch screen is rotated by 180 degrees (I need to set SwapXY = 1;, otherwise it is off by 90 or 270 degrees)

Thank you very much

Pio

closed 2018-05-16 by prenticedavid