prenticedavid/MCUFRIEND_kbv #24: LCD ID:04 04

LRagji, 2017-11-25:
I have a controller which has 40pin breakout and it reads the following : any idea what controller it is ?
reg(0x0000) 04 04	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 04 04 04 04	Manufacturer ID
reg(0x0009) 05 05 05 05 05	Status Register
reg(0x000A) 06 06	Get Power Mode
reg(0x000C) 04 04	Get Pixel Format
reg(0x0061) 65 65	RDID1 HX8347-G
reg(0x0062) 66 66	RDID2 HX8347-G
reg(0x0063) 67 67	RDID3 HX8347-G
reg(0x0064) 64 64	RDID1 HX8347-A
reg(0x0065) 65 65	RDID2 HX8347-A
reg(0x0066) 66 66	RDID3 HX8347-A
reg(0x0067) 67 67	RDID Himax HX8347-A
reg(0x0070) 74 74	Panel Himax HX8347-A
reg(0x00A1) 25 25 25 25 25	RD_DDB SSD1963
reg(0x00B0) 34 34	RGB Interface Signal Control
reg(0x00B4) 34 34	Inversion Control
reg(0x00B6) 36 36 36 36 36	Display Control
reg(0x00B7) 37 37	Entry Mode Set
reg(0x00BF) 37 37 37 37 37 37	ILI9481, HX8357-B
reg(0x00C0) 44 44 44 44 44 44 44 44 44	Panel Control
reg(0x00C8) 44 44 44 44 44 44 44 44 44 44 44 44 44	GAMMA
reg(0x00CC) 44 44	Panel Control
reg(0x00D0) 54 54 54	Power Control
reg(0x00D2) 56 56 56 56 56	NVM Read
reg(0x00D3) 57 57 57 57	ILI9341, ILI9488
reg(0x00D4) 54 54 54 54	Novatek ID
reg(0x00DA) 56 56	RDID1
reg(0x00DB) 57 57	RDID2
reg(0x00DC) 54 54	RDID3
reg(0x00E0) 64 64 64 64 64 64 64 64 64 64 64 64 64 64 64 64	GAMMA-P
reg(0x00E1) 65 65 65 65 65 65 65 65 65 65 65 65 65 65 65 65	GAMMA-N
reg(0x00EF) 67 67 67 67 67 67	ILI9327
reg(0x00F2) 76 76 76 76 76 76 76 76 76 76 76 76	Adjust Control 2
reg(0x00F6) 76 76 76 76	Interface Control

prenticedavid, 2017-11-25:
First off.   Post a link to your **actual** display.   e.g. the Ebay sale.

Either use the standard pin wiring for the Arduino Uno shield.
Or Edit the LCD_ID_readreg sketch to match your pin wiring.  (paste your defines)

If you are a GitHub member,  it is wise to clone a project and build specific Branches.
If you are a regular Arduino user,  you should install library with the Library Manager.   And ask questions on the Arduino.cc Displays Forum.

David.

LRagji, 2017-11-26:
Hi David, Got the controller from https://www.tanotis.com/products/tanotis-2pcs-3-2-inch-tft-lcd-module-display-touch-panel-sd-card-240x320-than-128x64-lcd?variant=20619224069
and shield for Mega from here https://www.ebay.in/itm/TFT-3-2-4-3-5-0-7-0-touch-LCD-Shield-for-Arduino-Due-Mega/112546865231?hash=item1a3450e04f:g:VG4AAOSwNnRYlYWP

prenticedavid, 2017-11-26:
Seriously,   your link is useless.   I can't view the photo of the pcb.
Please find an Ebay link or post a proper photo.

Either use UTFT with model: ILI9341_16
Or configure MCUFRIEND_kbv for USE_SPECIAL and USE_MEGA_16BIT_SHIELD

Note that you can't read the display ID with that Adapter Shield.
If you buy the ColdTears Adapter for Due,  you can make a trivial hardware mod.

David.

LRagji, 2017-11-29:
David does this help?
[image]

prenticedavid, 2017-11-29:
No,  it does not help very much.   Google does not find anything for H28003FPC

Surely you could reassemble the display.
Then take a photo of the pcb.  i.e. what any normal human would see.

David.

closed 2018-05-16 by prenticedavid