prenticedavid/MCUFRIEND_kbv #25: Inverted Color + glitch ILI9486 / ILI9481

psy460, 2017-12-10:
I'm using two 3,5" TFT screen with an arduino mega, both of them has inverted color, for exemple black is 0xFFFF, orange is 0x07FF, cyan is 0xF800.

[image]
[image]

after few seconds, some line, some characters or some pixels are displayed at the wrong place

[image]

The same sketch work perfectly on a third screen, with same pcb, i will check which driver is it.

psy460, 2017-12-10:
My working screen is 9481 too ... very strange, same PCB than other one.

[image]

**## ILI9486 not working** : 

reg(0x0000) 00 00 ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66 Manufacturer ID
reg(0x0009) 00 00 61 00 00 Status Register
reg(0x000A) 00 08 Get Powsr Mode
reg(0x000C) 00 66 Get Pixel Format
reg(0x0061) 00 00 RDID1 HX8347-G
reg(0x0062) 00 00 RDID2 HX8347-G
reg(0x0063) 00 00 RDID3 HX8347-G
reg(0x0064) 00 00 RDID1 HX8347-A
reg(0x0065) 00 00 RDID2 HX8347-A
reg(0x0066) 00 00 RDID3 HX8347-A
reg(0x0067) 00 00 RDID Himax HX8347-A
reg(0x0070) 00 00 Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30 RD_DDB SSD1963
reg(0x00B0) 00 00 RGB Interface Signal Control
reg(0x00B4) 00 00 Inversion Control
reg(0x00B6) 00 02 02 3B 3B Display Control
reg(0x00B7) 00 06 Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00 ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00 GAMMA
reg(0x00CC) 00 04 Panel Control
reg(0x00D0) 00 00 00 Power Control
reg(0x00D2) 00 00 00 00 00 NVM Read
reg(0x00D3) 00 00 94 86 ILI9341, ILI9488
reg(0x00DA) 00 54 RDID1
reg(0x00DB) 00 80 RDID2
reg(0x00DC) 00 66 RDID3
reg(0x00E0) 00 0F 21 1C 0B 0E 08 49 98 38 09 11 03 14 10 00 GAMMA-P
reg(0x00E1) 00 0F 2F 2B 0C 0E 06 47 76 37 07 11 04 23 1E 00 GAMMA-N
reg(0x00EF) 00 80 00 10 60 40 ILI9327
reg(0x00F2) 00 18 A3 12 02 B2 12 FF 10 00 00 00 Adjust Control 2
reg(0x00F6) 00 54 80 66 Interface Control

**## ILI9481 not working**
reg(0x0000) 00 00 ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00 Manufacturer ID
reg(0x0009) 00 00 61 00 00 Status Register
reg(0x000A) 08 08 Get Powsr Mode
reg(0x000C) 66 66 Get Pixel Format
reg(0x0061) 00 00 RDID1 HX8347-G
reg(0x0062) 00 00 RDID2 HX8347-G
reg(0x0063) 00 00 RDID3 HX8347-G
reg(0x0064) 00 00 RDID1 HX8347-A
reg(0x0065) 00 00 RDID2 HX8347-A
reg(0x0066) 00 00 RDID3 HX8347-A
reg(0x0067) 00 00 RDID Himax HX8347-A
reg(0x0070) 00 00 Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00 RD_DDB SSD1963
reg(0x00B0) 00 00 RGB Interface Signal Control
reg(0x00B4) 00 00 Inversion Control
reg(0x00B6) 00 00 00 00 00 Display Control
reg(0x00B7) 00 00 Entry Mode Set
reg(0x00BF) 00 02 04 94 81 FF ILI9481, HX8357-B
reg(0x00C0) 00 10 3B 00 02 11 00 00 00 Panel Control
reg(0x00C8) 00 00 44 06 11 44 55 0A 08 17 33 77 11 GAMMA
reg(0x00CC) 00 00 Panel Control
reg(0x00D0) 00 00 43 Power Control
reg(0x00D2) 00 01 22 00 00 NVM Read
reg(0x00D3) 00 01 22 00 ILI9341, ILI9488
reg(0x00DA) 00 00 RDID1
reg(0x00DB) 00 00 RDID2
reg(0x00DC) 00 00 RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 GAMMA-N
reg(0x00EF) 00 00 00 00 00 00 ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00 Adjust Control 2
reg(0x00F6) 00 00 00 00 Interface Control

**## ILI9481 Working**
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 08	Get Powsr Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 02 04 94 81 FF	ILI9481, HX8357-B
reg(0x00C0) 00 10 3B 00 02 11 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 43	Power Control
reg(0x00D2) 00 01 22 00 00	NVM Read
reg(0x00D3) 00 01 22 00	ILI9341, ILI9488
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 33 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 80 80 80	Interface Control

prenticedavid, 2017-12-10:
Note that the physical TFT panels differ by TFT manufacturer.
The Silicon chip controller is made by a Semiconductor company.   It can be configured to suit the different panels.

My ILI9486 has correct colours.   Other people have reported inverted colours on their panels.

Most Blue 3.5" Uno shields have an ILI9481.   My Blue 3.5" one has correct colours.
These ILI9481 modules are prone to overheat.   

You can either change the REV_SCREEN attribute in the case 0x9481 block.
Or just call tft.invertDisplay() in your sketch. 

I never cease to be amazed by people using Uno shields with a Mega2560.
Mega shields are pretty cheap.   Unfortunately the microSD is not reliable on the red Mcufriend Mega2560 Shields.

Both of your photos have bent digital#2 pins.

David.

psy460, 2017-12-11:
i will try to invert colors and report you. 
My sketch is too big for arduino uno, i have no choice to use mega2560. 

#2 pin are bend because i also have canbushield between TFT and mega, and both shield use pin 2.
Mega's D2 pin is linked to TFT #2 pin.  

Do you know why my not working ILI9481 as much more vivid color and much better view  angle than my two other one ? 

[image]

prenticedavid, 2017-12-11:
Panels vary between manufacturers.   Some have better viewing angle.   Some have inverted colours.   Some have different Colour characteristics.

That is why you have different Manufacturer registers and Gamma settings.

First off.   Unpeel the plastic film.   It destroys the clarity of any colours.

You can try different initialisations and gammas by commenting/uncommenting different arrays in  the table8_ads assignment. 

David.

psy460, 2017-12-11:
thx al lot for the hint, will try all that :)

psy460, 2017-12-11:
Ok, so invertColor work fine, i need to do more test for glitchs i find better gamma calibration.

I notice that pixel mode (0x3A) is 55 = 565 color = 65k, couldn't we work with 272k colors ? it will need uint32_t for color right ?

prenticedavid, 2017-12-11:
No.   65k colours is what you want.

psy460, 2017-12-12:
i'm playing with screen configuration to improve color on my ili9486, 

do you have an idea why i can't set 0x54 to 0x2C with 0x53 command ? 

To enable brightness,  backlight and dynamic dimming function ? 
[image]

this code : 
[image]
Give me : "reg : 0"

prenticedavid, 2017-12-12:
Both ILI9486 and ILI9481 are MIPI controllers.
You should use pushCommand() rather than WriteCmdData().
e.g. uint8_t val = 0x2C; tft.pushCommand(0x53, &val, 1);

Neither of the displays in your photos have access to the backlight.   It is permanently on.
There is little point in playing with reg(0x53) unless you have control of the backlight.

Just accept that you always lose ~200mA to a backlight whether the TFT is displaying Black or White.
If you are battery powered,  you will just have to go to the shops every few hours.

David.

psy460, 2017-12-12:
Here is my ILI9486, the worst of my screen after some improvement : 

[image]
[image]

            0xB1, 2, 0xf0, 0x1f, //Frequency Full color                                       [B0 11]
            0xB4, 1, 0x00,      //Inversion Control [00]
            0xB6, 3, 0x02, 0x02, 0x3B,  // Display Function Control [02 02 3B]
	    0xB7, 1, 0x47, 	//Entry Mode Set                                                                         
			
	    0xC0, 2, 0x1F, 0x1F,        //Power Control 1 [0E 0E] 
            0xC1, 2, 0x20, 0x00,        //Power Control 2 [43 00]
            0xC2, 1, 0x43,      //Power Control 3 [33]                                               
            0xC5, 4, 0x00, 0x00, 0x00, 0x00,    //VCOM  Control 1 [00 40 00 40]

prenticedavid, 2017-12-12:
I would need to study the meaning of your changes.   Some of them look a little strange.

Your first photo has obviously been taken from a different angle.    TFTs prefer a 90 degree view.
The second photo is showing a good contrast.

It is best to compare Gammas with known photos in a bitmap.
Power and VCOM are best with sharp lines and graphics.

I would expect one or another of the initialisation tables would suit your panel.   After all,  they come from Ilitek  and the panel manufacturers.   You obviously understand how to adjust directions and colour inversion.    Get those correct first.  Then compare the picture(s).

Ah-ha.   I have given you multiple Gammas but only one choice of Power setting.     Look in the unused folder.   Note that each table entry needs the argument count inserted.    I assume that you can see how my tables work.    

David.

psy460, 2017-12-13:
Take a look on ILI9486 description sheet when you have some time
Yes first photo was take from the best view angle, second from the same angle than picture above.

I tried the four or five gamma setting, but for the working ones, i can't see any big difference, i think because i'm displaying simple monochrome UI.

color inversion is fixed now, all is fine. i just need to test if i get glitch while driving (it's live car's data monitoring)

Oh i didn't see ili9486 init.h file, i willl try them and repport you ;)

bpoag, 2018-01-15:
Have you tried tft.invertDisplay(true); ?

closed 2018-05-16 by prenticedavid

surathshan, 2018-08-08:
hello my ILI9486 tft display didn't work for MCUFRIEND_kbv version: 2.9.9 i use all library examples but it didnt work

surathshan, 2018-08-08:
please help me