prenticedavid/MCUFRIEND_kbv #26: ESP32 (Generic dev modules)

cdegrassi, 2017-12-11:
Hi, Thank you for the great development effort!

I have just purchased a couple of ESP32 modules and I was wondering if there is anybody that could give me some hints on how to make it work with the mcufriends_kbv library. 

As usual, the 3.5" TFT works great on the UNO, but I am not sure where I need to change the GPIOs to have it work with the ESP32.
I have noticed that there is an ESP32 reference in the mcufriend_shield.h. I have tried changing the pins there, to match the wired pattern but with no result.
I would really appreciate some help.

I did try the LCD_ID test example but even that does not seem to give any result (0x0 unidentified id).

Thank you
Chris

prenticedavid, 2017-12-11:
I have "D1 R32" board with Uno headers from Ebay. 
And a regular ESP32 module i.e. ESP-WROOM-32 mounted on a DIL pcb.

I mounted the ESP32 module on a Uno ProtoShield with wiring as described in mcufriend_shield.h i.e. soldered trailing wires to the header sockets.

The D1_R32 board already has headers but the Analog header contained "analog-only" pins.   Which is why I did the hardware mod to wire "GPIO" pins to the Analog header.    The TFT Shield needs "digital" on the control pins but "analog" when using the TouchScreen.

MCUFRIEND_kbv works on both hardware boards.   But it is not 100% stable on the ProtoShield board.   Let's face it.   You have fast digital signals on loose wires.   It is not going to be as reliable as a well routed soldered multi-layer pcb.

I would NEVER consider connecting a parallel TFT to a breadboard with jumper cables.
It works ok with a 16MHz AVR but the ESP32 goes a lot faster.

I strongly recommend buying a Uno-style ESP32 board if you want to use Uno-style Shields.
ESP32 and ESP8266 are far better suited to SPI displays.   Less wires.   Good SPI hardware.

David.

cdegrassi, 2017-12-11:
Hello David,
Thank you for the reply. I totally agree with you in regard to using a proper PCB. I am currently prototyping the hardware and just wanted to make sure it was working.

So I went back to the _sheild.h file and matched your pinout except I am not sure what the meanings of the comment are here:
#define LCD_RS 15  //hard-wired to A2 (GPIO35) 
#define LCD_CS 33  //hard-wired to A3 (GPIO34)
#define LCD_RST 32 //hard-wired to A4 (GPIO36)

Are these pins physically connected to the (GPIOXX) in the comments or to the ones in the declaration (i.e LCD_RS is connected to actual GPIO 15)?

I used the 15, 33, and 32 and I did get the screen to work except that I get reversed colors.
Any suggestions?

Thank you again.
Cheers
Chris

[image]
[image]

prenticedavid, 2017-12-11:
As I explained,   The D1 R32 has analog-only port pins connected to Analog Header e.g. A2=GPIO35.
I hard-wired GPIO15 to A2 from the adjacent empty holes.  (and similarly with the other comments)

I suggest that you Google for "D1 R32" and look at my board.
Obviously you have not done too badly.   Your panel obviously requires changing the REV_SCREEN attribute,    Is it a Red 3.5" ILI9486 Shield?

David.

cdegrassi, 2017-12-11:
Ok, got it. I did test tft.invertDisplay(true) and it works. 
So as far as I am concerned that is enough for me.

I have did ordered a couple of D1 R32 this morning, thank you.

Performance is very good considering the poor wiring and now I can start working on the custom pcb for my project.

Thank you again. 
I am no expert in c/c++ hardware drivers development but let me know if you need any help.
Cheers
Chris