prenticedavid/MCUFRIEND_kbv #30: Judgment of RM68140

nopnop2002, 2018-01-22:
Judgment of RM68140 is not right.
RM68140's Display ID Register is 0x04h.
And Register value is 32bit [xx 0x54 0x80 0x66]

e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 54 54 80 66	Manufacturer ID -------------> This is RM68140
reg(0x0009) 00 00 61 00 00	Status Register

prenticedavid, 2018-01-22:
I read reg(0x00BF) e.g.
`
reg(0x0004) 54 54 80 66   Manufacturer ID
reg(0x00BF) FF FF 68 14 00 FF   ILI9481, HX8357-B
`

reg(0xBF) is not documented in my Raydium datasheet.    As it happens,  I do not own a readable RM68140.   So I have to rely on user reports e.g. as quote.
Note that the RM68042 datasheet does document its reg(0xBF) :  `xx 01 D0 68 04 FF`
Of course my assumption about RM68140 may be wrong.

If you have a genuine RM68140 you can do some diagnosis for yourself.   e.g. compare Manufacturer registers with "unusual" bit fields.    Write to them and read back.

You really do not want to trust reg(4).    Many different controllers will return `xx 54 80 66`

David.

nopnop2002, 2018-01-22:
Hello Davis.
Thank you for useful library.

I have a genuine RM68140.
Does this provide with your useful information?

e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 54 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 66 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 02 02 02 3B 00	Display Control
reg(0x00B7) 06 06	Entry Mode Set
reg(0x00BF) FF FF 68 14 00 FF	ILI9481, HX8357-B
reg(0x00C0) 0E 0E 0E 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 94 86	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 54 54	RDID1
reg(0x00DB) 80 80	RDID2
reg(0x00DC) 66 66	RDID3
reg(0x00E0) 00 00 54 07 44 05 08 00 54 07 44 05 08 44 44 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2018-01-22:
Yes,  that is exactly the same report as afarina2 quoted.

Did you notice that reg(0xD3) looks like an ILI9486.   Ilitek ILI9486 and Raydium RM68140 have subtle differences.

David.

nopnop2002, 2018-01-22:
Thank you for your explanation.
Please close this isue.

closed 2018-01-22 by prenticedavid