prenticedavid/MCUFRIEND_kbv #31: LCD TFT  On / Off

artbody, 2018-01-24:
After a lot of research, i didn't find any method to switch off and on the display
my display is a spfd5408 mcufriends version
what i found is https://github.com/PaxInstruments/STM32CubeF3/blob/master/Drivers/BSP/Components/spfd5408/spfd5408.h
for stm32.
so i think this should work in one or another way

the .h file
```
void     spfd5408_DisplayOn(void);
void     spfd5408_DisplayOff(void);
```
https://github.com/PaxInstruments/STM32CubeF3/blob/master/Drivers/BSP/Components/spfd5408/spfd5408.c
the c file
```
void spfd5408_DisplayOn(void)
{
  /* Power On sequence ---------------------------------------------------------*/
  spfd5408_WriteReg(LCD_REG_24, 0x36); /* Display frame rate = 70Hz RADJ = '0110' */
  spfd5408_WriteReg(LCD_REG_25, 0x01); /* OSC_EN = 1 */
  spfd5408_WriteReg(LCD_REG_28, 0x06); /* AP[2:0] = 111 */
  spfd5408_WriteReg(LCD_REG_31,0x90); /* GAS=1, VOMG=00, PON=1, DK=0, XDK=0, DVDH_TRI=0, STB=0*/
  LCD_Delay(10);
  /* 262k/65k color selection */
  spfd5408_WriteReg(LCD_REG_23, 0x05); /* default 0x06 262k color,  0x05 65k color */
  /* SET PANEL */
  spfd5408_WriteReg(LCD_REG_54, 0x09); /* SS_PANEL = 1, GS_PANEL = 0,REV_PANEL = 0, BGR_PANEL = 1 */
 
  /* Display On */
  spfd5408_WriteReg(LCD_REG_40, 0x38);
  LCD_Delay(60);
  spfd5408_WriteReg(LCD_REG_40, 0x3C);
}

/**
  * @brief  Disables the Display.
  * @param  None
  * @retval None
  */
void spfd5408_DisplayOff(void)
{
  /* Power Off sequence ---------------------------------------------------------*/
  spfd5408_WriteReg(LCD_REG_23, 0x0000); /* default 0x06 262k color,  0x05 65k color */
  spfd5408_WriteReg(LCD_REG_24, 0x0000); /* Display frame rate = 70Hz RADJ = '0110' */
  spfd5408_WriteReg(LCD_REG_25, 0x0000); /* OSC_EN = 1 */
  spfd5408_WriteReg(LCD_REG_28, 0x0000); /* AP[2:0] = 111 */
  spfd5408_WriteReg(LCD_REG_31, 0x0000); /* GAS=1, VOMG=00, PON=1, DK=0, XDK=0, DVDH_TRI=0, STB=0*/
  spfd5408_WriteReg(LCD_REG_54, 0x0000); /* SS_PANEL = 1, GS_PANEL = 0,REV_PANEL = 0, BGR_PANEL = 1 */

  /* Display Off */
  spfd5408_WriteReg(LCD_REG_40, 0x38);
  LCD_Delay(60);
  spfd5408_WriteReg(LCD_REG_40, 0x04);
}
```

prenticedavid, 2018-01-24:
And your point is?

You can send any display controller to sleep,  power on, off, wake, ...
The controller will take less current when asleep or off.  e.g. 10mA to 20mA.

Backlight LEDs draw 50mA to 400mA.   The Mcufriend Shields do not have access to the backlight.
Saving 10mA is not very significant.

No,   I have not looked up the decimal address values of your mystery registers.    But they don't look quite right for an ILI9320-style controller.   Most datasheets quote hex values.

David.

artbody, 2018-01-26:
ok thx, then i'll switch off the complete display with a little Mosfet

verbessern, 2019-03-28:
I think its generally a good idea to have on/off functionality, in case you run the display on a battery. That may be a backup battery for the main power supply and can prolong the run-time if the controller is able to detect the disconnected main power supply and turn the display off to save power. Not to say that turning the display off in case the device is not used, may increase the life of the display.

prenticedavid, 2019-03-28:
I explained on Jan 24.   There are not many pins on a Uno Shield.   The Mcufriend Uno Shield has no means of accessing the backlight LED.

The current Mcufriend pcb is **WRONG**.   It power the backlight from the Uno/Mega 3.3V pin.
The original Mcufriend pcb had an onboard AMS1117-3.3 regulator powered from the 5V pin.
Yes,  you could snip the 5V header pin and switch 5V with an external transistor on those pcb.

SPI displays use less pins.   You can have TFT,  SD, Flash, XPT2046 all on the same SPI bus.
Most have an LED pin.   So you can use an external transistor switch.
Some Ebay SPI displays have onboard MOSFET / transistor.

David.

verbessern, 2019-03-28:
What about sending sleep commands?

prenticedavid, 2019-03-28:
> You can send any display controller to sleep, power on, off, wake, ...
> The controller will take less current when asleep or off. e.g. 10mA to 20mA.
> 
> Backlight LEDs draw 50mA to 400mA. The Mcufriend Shields do not have access to the backlight.
> Saving 10mA is not very significant.
> 
You can send a Sleep command.   It will reduce the current  by 10mA.
It only msakes sense if you can turn the backlight off too.  

Look at the pcb.    Does it have an AMS1117-3.3 regulator?   Does it have an empty SOIC-8 footprint?
If unsure,  post a photo of the pcb.

David.

verbessern, 2019-03-29:
[image]
I would be happy and for 10mA. Is it possible to attach the 3.3v display module Vcc to some digital pin and just turn it on/off writing to the pin? In page 140 of the LGDP4532 specs there is display off and power off sequences related to fig 75, will this sequences to the job?

prenticedavid, 2019-03-29:
I am gobsmacked by trailing wires to a shield.    God invented Shields so that male pins could mate with correspondong female sockets.

Writing software commands is easy.   But your pcb is the crap Mcufriend design without LDO regulator.
You need a transistor or MOSFET to switch backlight current.

Simply switching the 3.3V power pin means that you get voltages leaking from the signal lines.
You must **NEVER** allow signal voltages with an "unpowered" board. 

My advice is to tell Mcufriend that they have a crap pcb.   Either chuck the display in the trash or buy more batteries.

David.

verbessern, 2019-03-29:
I understood that for the backlight, but I'm not good in hardware, so I prefer to write some code for the moment. I don't understand 'voltages leaking' thing but I will look at it. I want to work with my display however, first that I already have it, second that I will collect some skills in the processes. When I understand the current display better, I will be able to choose more wisely in the future. I will try some commands to make the display sleep and power off some day soon, and probably. 

Do you think that interrupting the 3.3V VCC line between the controller and the display will be enough, or I **must** cut the trace to the backlight itself? When I was connecting the display I have not given it the 3.3V correctly (as you can see in the photo I used one pin down by mistake), then the display works normally, just its very dark, but did not broke happily. Do you think that is _safe_ way to do it? Cutting a trace sounds to me unpleasant at least.

prenticedavid, 2019-03-29:
Measure the current taken from the 3.3V pin.   i.e. just connect GND.   And connect 3.3V pin via a DMM.

Your LGPD4532 probably takes 50mA.

You could cut the backlight trace.   And route it to A5 via a 47R resistor.
This will mean you could switch the backlight on and off via the A5 pin.   (if you keep the current less than 40mA)

And you could reduce the LGDP4532 current with Sleep or Display Off.
Arduinos are not ideal "low current" devices even if you send everything to sleep.

David.

verbessern, 2019-03-29:
I use Wemos Lolin32. The current of the 3.3V is 75mA. I understand the situation now, and will work on that soon. Thank you very much for you help David.

verbessern, 2019-03-29:
Btw, how can i find this backlight trace?

prenticedavid, 2019-03-29:
By following the 3.3V trace.   See where it goes.

As far as I know ESP32 has max 12mA per pin.   You will have to check for yourself.

Since you have a good 3.3V power source,   you would probably use 0R or 22R for the backlight series resistor.

Note that the ESP32 takes a massive current when the Wireless is operating.

David.

verbessern, 2019-03-29:
The 3.3V goes directly under the display, and its not visible what happens under it :|

prenticedavid, 2019-03-29:
My rude comments about the Mcufriend pcb concern Uno, Mega, ...
The Uno, Mega have 5V GPIO and a wimpy 3.3V regulator.

I would be interested in your current consumption.    The 2.4 inch displays will probably be reasonable.

I could dismantle "crap Mcufriend pcb" and see where the traces go.    There is little point in wasting my time if you are not prepared to hack your pcb.

The obvious solution is to use an ILI9341 SPI display like https://www.ebay.com/itm/240x320-2-4-SPI-TFT-LCD-Touch-Panel-Serial-Port-Module-with-PCB-ILI9341-3-3V/142671834108?epid=2201986457&hash=item2137e76bfc:g:EzkAAOSwOhFbNfPo

Note that it has yellow pin header and Q1 transistor as well as U1 regulator.
The black pin header boards do not have Q1.   So you need an external transistor.
XPT2046 Touch controller,  SD cage,  switchable LED.

Bodmer's TFT_eSPI library works nicely with ESP32.

David.

verbessern, 2019-03-29:
I can dismantle mine, and would like to know how this can be done (if at all). What should i do to disconnect the display from the pdb? Just pull maybe with a thin sharp cutter?

verbessern, 2019-03-29:
I liked the display so much, because its the first display I managed to connect and start. I have ordered two more even, they are on the way: https://www.aliexpress.com/item/2-4-inch-TFT-LCD-Touch-Screen-Shield-for-UNO-R3-Mega2560-LCD-Module-Display-Board/32827149292.html?spm=a2g0s.9042311.0.0.27424c4dZIGwdw and https://www.aliexpress.com/item/Free-shipping-LCD-module-TFT-2-4-inch-TFT-LCD-screen-for-Arduino-UNO-R3-Board/32583491591.html?spm=a2g0s.9042311.0.0.27424c4dZIGwdw
One have this element that i think is converting 5V to 3.3V this with the 3 small legs and 1 big.

prenticedavid, 2019-03-29:
Both AliExpress links come up in Russian.    Do you come from Russia?
AliExpress always comes up for me in a non-English language.

Do not trust the photos.   I suspect that your first link will have a non-regulator pcb when it arrives through your door.

You dismantle by either cutting the foam pads with an X-Acto or Razor blade.
Or apparently a hairdryer will soften the sticky.

You should be able to identify the LED-A and LED-K pins on the ribbon footprint.
It is easier to switch LED-K with a NPN transistor.   Or you switch LED-A with a PNP transistor.

David.

verbessern, 2019-03-29:
No i'm not, it seems AliExpress recognizes you wrongly. Here is a photo:
[image]
The 3.3V is up right second pin. Whats next?

verbessern, 2019-03-29:
Btw. will this command sequences I was talking about early not actually turn the display off? Have you tested?

prenticedavid, 2019-03-29:
What country are you in?

The backlight and the controller are two separate things.
You can sleep the controller but this makes no difference to the backlight.
I have a similar pcb on a dead Mcufriend but the ribbon traces are completely different.

Life would be much easier with SPI ILI9341.

David.

verbessern, 2019-03-29:
I like privacy. We maybe better stop with the task, I could not find any information about this TFT8K... display. I guess I will have to tolerate the backlight, and think twice next time I buy that. Thank you.

verbessern, 2019-03-30:
Btw. I think that is what is all about! After the turn off sequence, the display must be turn off by shutting the power. That is what is in Figure 75 of the specs (LGDP4532 v0.12), "Turn the power supply off (Vcc, Vci, IOVcc)" after the power off sequence. So the display is perfectly fine, just needs additional hardware care to turn it off. That is balanced by lower price of the module compared to the others.

closed 2021-08-24 by prenticedavid