prenticedavid/MCUFRIEND_kbv #33: stm32f411RE didn't compile

artbody, 2018-01-26:
my issue is my new board
this because my measurement system needs a lot of ram  
therfor i've changed from a stm32f103rb nucleo 64 to
stm32f411RE nucleo 64
but my code didn't compile with mcufriends_kbv

errormessage
Board is not supported
so i started to edit 
mcufriends_shield.h

```

//####################################### STM32 ############################
#elif defined(__STM32F4__) || defined (__STM32F1__) || defined(STM32F103xB) || defined(STM32F411xE) || defined(STM32L476xx) // MAPLECORE or STM32CORE 

#if 0
#elif defined(STM32L476xx)
#define WRITE_DELAY { WR_ACTIVE; WR_ACTIVE; }
#define READ_DELAY  { RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; }
#define REGS(x) x
#define PIN_MODE2(reg, pin, mode) reg=(reg&~(0x3<<((pin)<<1)))|(mode<<((pin)<<1))
#define GPIO_INIT()   { RCC->AHB2ENR |= RCC_AHB2ENR_GPIOAEN | RCC_AHB2ENR_GPIOBEN | RCC_AHB2ENR_GPIOCEN; }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1)
#elif defined(ARDUINO_NUCLEO_F103C8) || defined(ARDUINO_NUCLEO_F103RB)  || defined(ARDUINO_NUCLEO_F411RE)//regular CMSIS libraries
#define REGS(x) x
#define GPIO_INIT()   { RCC->APB2ENR |= RCC_APB2ENR_IOPAEN | RCC_APB2ENR_IOPBEN | RCC_APB2ENR_IOPCEN | RCC_APB2ENR_IOPDEN | RCC_APB2ENR_AFIOEN; \
        AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1;}
#else                                                                  //weird Maple libraries
#define REGS(x) regs->x
#endif
#if defined(__STM32F4__) || defined(__STM32F1__) || defined(STM32F103xB) || defined(STM32F411xE) || defined(STM32F411RE)
#define WRITE_DELAY { }
#define READ_DELAY  { RD_ACTIVE; }
#define GROUP_MODE(port, reg, mask, val)  {port->REGS(reg) = (port->REGS(reg) & ~(mask)) | ((mask)&(val)); }
#define GP_OUT(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x33333333)
#define GP_INP(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x44444444)
#define PIN_OUTPUT(port, pin) {\
        if (pin < 8) {GP_OUT(port, CRL, 0xF<<((pin)<<2));} \
        else {GP_OUT(port, CRH, 0xF<<((pin&7)<<2));} \
    }
#define PIN_INPUT(port, pin) { \
        if (pin < 8) { GP_INP(port, CRL, 0xF<<((pin)<<2)); } \
        else { GP_INP(port, CRH, 0xF<<((pin&7)<<2)); } \
    }
#endif
#define PIN_HIGH(port, pin)   (port)-> REGS(BSRR) = (1<<(pin))
#define PIN_LOW(port, pin)    (port)-> REGS(BSRR) = (1<<((pin)+16))

#if 0
#elif defined(ARDUINO_GENERIC_STM32F103C) || defined(ARDUINO_NUCLEO_F103C8)
#warning Uno Shield on BLUEPILL
#define RD_PORT GPIOB
#define RD_PIN  5
#define WR_PORT GPIOB
#define WR_PIN  6
#define CD_PORT GPIOB
#define CD_PIN  7
#define CS_PORT GPIOB
#define CS_PIN  8
#define RESET_PORT GPIOB
#define RESET_PIN  9

// configure macros for the data pins
#define write_8(d)    { GPIOA->REGS(BSRR) = 0x00FF << 16; GPIOA->REGS(BSRR) = (d) & 0xFF; }
#define read_8()      (GPIOA->REGS(IDR) & 0xFF)
//                                         PA7 ..PA0
#define setWriteDir() {GP_OUT(GPIOA, CRL, 0xFFFFFFFF); }
#define setReadDir()  {GP_INP(GPIOA, CRL, 0xFFFFFFFF); }

[... 192 lines cut ...]
     setWriteDir();
     ^~~~~~~~~~~
/home/achim/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:406:92: error: 'struct GPIO_TypeDef' has no member named 'CRL'
 #define setWriteDir() {GP_OUT(GPIOA, CRH, 0xFFF); GP_OUT(GPIOB, CRH, 0xF00); GP_OUT(GPIOB, CRL, 0xFFF000); GP_OUT(GPIOC, CRL, 0xF0000000); }
                                                                                            ^
/home/achim/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:316:17: note: in definition of macro 'REGS'
 #define REGS(x) x
                 ^
/home/achim/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:326:43: note: in expansion of macro 'GROUP_MODE'
 #define GP_OUT(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x33333333)
                                           ^~~~~~~~~~
/home/achim/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:406:78: note: in expansion of macro 'GP_OUT'
 #define setWriteDir() {GP_OUT(GPIOA, CRH, 0xFFF); GP_OUT(GPIOB, CRH, 0xF00); GP_OUT(GPIOB, CRL, 0xFFF000); GP_OUT(GPIOC, CRL, 0xF0000000); }
                                                                              ^~~~~~
/home/achim/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:78:5: note: in expansion of macro 'setWriteDir'
     setWriteDir();
     ^~~~~~~~~~~
/home/achim/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:406:92: error: 'struct GPIO_TypeDef' has no member named 'CRL'
 #define setWriteDir() {GP_OUT(GPIOA, CRH, 0xFFF); GP_OUT(GPIOB, CRH, 0xF00); GP_OUT(GPIOB, CRL, 0xFFF000); GP_OUT(GPIOC, CRL, 0xF0000000); }
                                                                                            ^
/home/achim/Arduino/libraries/MCUFRIEND_kbv/utility/mcufriend_shield.h:316:17: note: in definition of macro 'REGS'
 #define REGS(x) x
                 ^
```
Have you some idea how we can solve this problem

prenticedavid, 2018-01-26:
I have added support for NUCLEO-F401RE and NUCLEO-F411RE to the Master Branch (beta).
No one had ever asked for these before.
Likewise,  no one has asked for Nucleo-F030, F091, F303, L053 or any Nucleo-144 board
If they ever do,   I will have to simplify the conditionals in mcufriend_shield.h because it is getting messy.

In an ideal world I could drop the Maple Core.   But ST don't support BluePill and MapleMini.    And MapleCore has some useful features e.g. DMA

David.

prenticedavid, 2018-01-27:
I have put support for all Nucleo-64 boards in the "test_nucleo" Branch.

I do not own any Nucleo-F030, F091, F303, L053 
So I would appreciate if an owner can test this code.

Does anyone own or use the Nucleo-144 boards?

David.

artbody, 2018-01-28:
ok it compiles without error
but 
diagnose_TFT_support
```
Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt

tft.readID() finds: ID = 0x5408

PORTRAIT is 240 x 320

Run the examples/graphictest_kbv sketch
All colours, text, directions, rotations, scrolls
should work.  If there is a problem,  make notes on paper
Post accurate description of problem to Forum
Or post a link to a video (or photos)

I rely on good information from remote users

```
https://plus.google.com/photos/photo/105577366774187711417/6516217885365147522
blank white screen ... after a few seconds a 10mm from right side darker gray rectangle appears. in this rectangle there are 3 dots with around 20px diameter. thy disappear after a few seconds

TouchScreen_Calibr_new
```
TouchScreen.h Calibration
Only AVR can diagnose Touch Pins
Other targets use default Pins
Touch Pin Wiring XP=7 XM=A1 YP=A2 YM=6

```
that's it 
white screen no touch function available

prenticedavid, 2018-01-28:
The diagnose_TFT_controller sketch recognises your SPFD5408 chip.
I just tried a (faulty) SPFD5408 board on a Nucleo-F401.

I get a white screen too.   The F401 is too fast for the ILI9320-style  controllers like SPFD5408.
I just tried it with a ILI9320.   It is too fast for reading the ID and readPixel().   But if I force the ID with tft.begin(0x9320) the graphictest_kbv sketch works.

I will have a look tomorrow.
I presume that your SPFD5408 runs ok on a regular Uno.
You will need to calibrate the Touch on the Uno to find the Touch pins.
The Calibration sketch just uses a "default" wiring on an ARM.

David.

prenticedavid, 2018-01-29:
I realised that you have F411.  I have changed the F401 to a Nucleo-F411.
I get the same White screen for the SPFD5408 but the ILI9320 runs ok (for write)

The SPFD5408 runs on a Nucleo-F103 @ 72MHz if I force the ID.   It does not read ID or readPixel correctly
The SPFD5408 runs ok on a Teensy3.2 @ 96 MHz

I put the SPFD5408 on an Xmega which is easy to change F_CPU.
The SPFD5408 does the Adafruit Tests in 0.83 sec with F_CPU @ 62MHz
The ILI9320 does the Adafruit tests in 1.72 sec with F_CPU @ 30MHz

So I would not expect a speed problem with the F411.
Slowing down the write sequence on the F411 still gets a White Screen.
It appears to be a Cortex-M4 GPIO feature of the F4xx that does not like SPFD5408

I will have to investigate further.

David.

prenticedavid, 2018-01-29:
Ah-ha.   The SPFD5408 has different timing to ILI9320 e.g.
```
tCYCW = 125ns  100ns
tCYCR = 450ns  300ns
PWLW  =  45ns   50ns
PWHW  =  70ns   50ns
tDSW  =  25ns   10ns
tH    =  10ns   15ns
tDDR  = 150ns  100ns

```
Change the PIN_HIGH and PIN_LOW macros for the STM32 e.g.
```
#if 0
#define PIN_HIGH(port, pin)   (port)-> REGS(BSRR) = (1<<(pin))
#define PIN_LOW(port, pin)    (port)-> REGS(BSRR) = (1<<((pin)+16))
#else
#define PIN_HIGH(port, pin)   (port)-> REGS(ODR) |= (1<<(pin))
#define PIN_LOW(port, pin)    (port)-> REGS(ODR) &= ~(1<<((pin)))
#endif

```
It looks as if tDSW is the problem.    I doubt if many people would use an obsolete SPFD5408 with a STM32F4xx.   So I suggest that you just use the slower macros for yourself.

David.

artbody, 2018-01-29:
Hy David, 
cool it compiles AND graphic is showing up
https://plus.google.com/photos/photo/105577366774187711417/6516546420846072130
https://plus.google.com/photos/photo/105577366774187711417/6516546594597976578
there are some shadow stripes in the image , 
but this is the case with arduino mega2560 , stm32f103 and my new board stm32f411re
so i think this is related to the mcufriends display
which i bought on ebay for less then 4€
2.4" TFT LCD Display Shield Touch Panel ILI9341 240X320 for Arduino UNO MEGA
but the ILI9341 is a SPFD5408

the story behind the scene:
 i started to develop a application with the  SPFD5408 and an arduino mega with a one channel measurement device
the mega2560 was to slow for the measurement device
so we decided to do it with an stmf103 nucleo board 
my customer now wants 16 channels and the F103RB runs out of RAM 
that's why i decided to switch to the F411RE Nucleo 
this board is fast enough

prenticedavid, 2018-01-29:
If you are worried about the health of the display post a photo of a screen from **graphictest_kbv** running on Uno or Mega.

I have never owned a SPFD5408.    The one on my desk is a damaged one from DrNick.
Ebay adverts stated "SPFD5408" on these Shields for many years.    No Shields ever contained an SPFD5408 until Summer 2017.   Obviously a batch of obsolete screens has appeared on the surplus market.     They get mounted on Mcufriend pcb and shifted through Ebay.

Most people seem to be very happy with them.    Your one looks damaged (to me).

David.

artbody, 2018-01-29:
the wave pattern in extreme looks like a google artefact from converting the images.
have a look at this pictures 
i have uploaded 3 on my homepage
http://artbody.de/node/28
i mean these shadows 
there are no shadows if background has a color

prenticedavid, 2018-01-29:
I would like to see a photo of the Shield on Mega or Uno.   There should not be shadows.

I suspect that your SPFD5408 is faulty.    Easiest solution is to buy a new Shield.   Most of them contain IL9341 but Ebay Vendors are a lottery.

MIPI controllers like ILI9341 have 66ns Write Cycle.   Compare with tCYCW for SPFD5408 and ILI9320.
You still have to slow down a Cortex-M4 controller.

David.

artbody, 2018-01-29:
the shadows are on arduino mega2560 the same .

artbody, 2018-02-16:
So i'm back again with new TFT's 3,5" ID 9488 ( i bought 3 peace's) 
one on a Mega 2560 = looks good 
see:
https://www.youtube.com/watch?v=h4EFEVUo_EA

one on a STM32F103RB = looks good
see:
https://www.youtube.com/edit?o=U&video_id=bT5uEMazLek

BUT
one on the STM32F411RE Nucleo 64 doesn't look good
all three TFT's show up with the same behaviour
 see 👎 
https://www.youtube.com/watch?v=IadD5COlvRc

all three TFT's show up on the other boards quite well
so i could be fore sure that the TFT#s are healthy
UTFT_Demo_480x320

prenticedavid, 2018-02-16:
First off.   Please use graphictest_kbv for any tests.

Yes,  the F411 is too fast for my ILI9488 Shield.   I had to add an extra WR_ACTIVE and an extra RD_ACTIVE
```
#elif defined(STM32F401xE) || defined(STM32F411xE)
#define WRITE_DELAY { WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; }
#define READ_DELAY  { RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; }
...

```

This is getting a little unwieldy.    Originally this library was written for AVRs and Cortex-M0.
Those MCUs were never too fast.

With the improvements in Xmega, M3, M4 GPIO drivers and faster clock speeds it became necessary to add some delays.    Ideally I would calculate the delay on clock speed but I also need to know the Family.

No,  I have not put these extra delays onto GitHub yet.   Please try them for yourself.

David.

artbody, 2018-02-17:
Ok i've now edited a lot of XX_ACTIVE in steps of 1 in mcufriends_shield.h

```
#elif defined(STM32F401xE) || defined(STM32F411xE)
#define WRITE_DELAY { WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; WR_ACTIVE; }
#define READ_DELAY  { RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; RD_ACTIVE; }
#define GPIO_INIT()   { RCC->AHB1ENR |= RCC_AHB1ENR_GPIOAEN | RCC_AHB1ENR_GPIOBEN | RCC_AHB1ENR_GPIOCEN; }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1)
```
BUT
no effect
watch:
 https://www.youtube.com/watch?v=3i3L9hFSQFA

prenticedavid, 2018-02-17:
You appear to read the ID correctly.  i.e. 0x9488
What do you see on the Serial Terminal?

The screen looks narrow and tall.  A 320x480 screen should look 3:4
The Adafruit Tests are really odd.
The Report text is unreadable.   
Apart from a brief moment when ID=0x9488 is in focus and F_CPU says 96MHz.
My F411 has F_CPU=100MHz  

Please run the LCD_ID_readreg sketch.   Perhaps your controller is not made by Ilitech.

David.