prenticedavid/MCUFRIEND_kbv #37: Experimental HX8352B Landscape modes broken

mink99, 2018-02-06:
on the HX8352B both landscape (1 & 3) modes do not work. even in the grapricstest. Shield is a open-smart 240*400

prenticedavid, 2018-02-06:
Yes,   I know.    But I don't have a HX8352-B on my desk.    And Arduino.cc member dunozf only has a Due.
If you have a Uno or Mega please could you tell me how the OPEN-SMART library works.    You can download from [https://forum.arduino.cc/index.php?topic=519044.msg3573296#msg3573296](url)

Please you can try it and tell me whether everything in graphictest_kbv works.
You will need to delete the current MCUFRIEND_kbv directory because the OPEN-SMART library is a hacked (obsolete) MCUFRIEND_kbv

Once I get some feedback,   I will make an effort to study the HX8352-B datasheet.

David.

prenticedavid, 2018-04-08:
I have just received an OPEN-SMART HX8357-B shield.
It should run on the test_8357B Branch.

I have updated the master Branch to support it but you have to configure the macros e.g.
SUPPORT_8357B, USE_SPECIAL, USE_OPENSMART_SHIELD_PINOUT.

Note that the Open Smart board shares LCD_D3 with SPI_MOSI on a Uno.
This means you must disable SPI when accessing the TFT.   e.g. beginTransaction(), endTransaction()
It is not a problem with Mega2560 or Due.

David.

closed 2018-05-16 by prenticedavid