prenticedavid/MCUFRIEND_kbv #38: Where is ADA_GFX_kbv.h in this project collection?

jsweet13, 2018-02-08:
I noticed ADA_GFX_kbv.h was mentioned in the mcufriend_kbv.h file as an #include, but can't seem to find it. Can we use the original Adafruit_GFX.h as they wrote it instead? I'm having to transcribe new files since I currently don't have a way to download. My graphics ID is 5408. Thanks.

prenticedavid, 2018-02-08:
ADA_GFX_kbv.h only gets used by non-Arduino projects.  e.g. MBED or Keil

It just uses sprintf() to implement most Print.h methods.
Once you start inheriting the Arduino Print.h class it gets very hairy on non-Arduino targets.

What target are you building for?

David.

jsweet13, 2018-02-08:
I'm currently trying to build for Arduino Uno R3, and the untouched Adafruit_GFX.h from their library returns ID: 5408.

However, it returns C0C0 when the display is plugged into a USB Host Shield I'd like to use in the middle after I figure out which pins the boards are trying to share so I can reroute the signals.

I was defining the pins individually like you have shown in the lcd_id example for reading the display registers hoping to initialize the display manually with something similar to your writecommand() writedata() functions which I think could work.

Do you know if the displays can handle one bit per pixel, where a byte handles 8 pixels?

Another board I have is an Atmel UC3A3256 Xplained the company has a ili9341 display for, and an Atmel ATmega328PB Mini which is roughly an ATmega328p with a few extra ports. These two boards would require wire hookups through their pin headers.

I have had luck before with an authentic Adafruit ili9341 and an Electronic Assembly SPI display initializing them manually one byte at a time per their user manuals, but I can't find anything on this Tftlcd for Arduino Uno (SPI flash) unit. 

I might solder on a 64-mbit SPI flash or FRAM to the board, or maybe SRAM. Haven't decided yet. The chip pads share the Micro-SD pads.

Thanks for your help.

prenticedavid, 2018-02-08:
The Mcufriend display shields have the pins clearly marked.
If you use Serial, TFT, Touch and microSD you have one spare pin left on a Uno : Analog#5
You can use A5 for Analog or Digital.

XMINI-328P and XMINI-328PB have some extra Analog pins: A6, A7

Sit down with a nice cup of tea and a clean sheet of paper.    Design your pin budget.    Choose a MCU with more pins if you don't have enough.

In practical terms,   you can use a SPI display instead of an 8-bit parallel.   If you have lots of pins you can use 16-bit parallel.

MCUFRIEND_kbv is only intended for parallel displays.    God invented female Arduino header sockets for the male pins on your Shield.

David.

jsweet13, 2018-02-08:
Correct, I have pin A5 available which I might use for the USB Host Shield chip select pin.

I might try something here before coding from scratch, will try to initialize the display with ID: 9325 to see if it does anything. The USB shield seems to work without a chipselect.

Will keep trying to get it to work, it would be nice to get working since it already fits the Arduino board.

I am fearing the display might have been damaged during transit when I ordered it, but the controller is responding. Maybe there's still a chance.

I might order a Display Xplained kit from Atmel and just code up with Atmel Studio. Their modules plug directly into their boards but not sure if they have boards supporting USB peripherals.

An SPI display would be better I believe for the Arduino.

jsweet13, 2018-02-08:
Ok, here's my findings...

The display does initialize properly when forcing the native graphics example with unchanged Adafruit_GFX.h + Adafruit_TFTLCD.h.

I commented out the if(identifier ==) code and just forced a tft.begin(0x9325);

So... if you have a Tftlcd for Arduino Uno (spiflash) board, just force it to use 0x9325.

Next, I'll look into swapping the screen writing order to reverse it's current interpretation. The words are backwards, but since I'll be plotting pixels manually from an array, this can be done by flipping the bits in the bytes or going for a if(data && 128) {"set bit 0 high"} and so forth.

Coolness.

prenticedavid, 2018-02-08:
From memory,   the current v2.9.7 Release supports the SPFD5408.    The working Beta  (Master Branch) certainly does.

The SPFD5408 is similar to the ILI9320 not the ILI9325.
if display is upside down in Portrait,  alter the INVERT_GS attribute.
If text is mirrored in Portrait,   alter the INVERT_SS. attribute. 

If unsure,  ask.    Preferably with a photo that shows BLUE Band and Penguin and the flexi-ribbon or USB cable. 

David.

jsweet13, 2018-02-09:
Doing a color test, I found a controlled write could be performed by the following. Though it stopped working right when I changed something I can't figure out in the code.

Void loop(void) {
    tft.setRotation(1);
    FillScreen();
}

Void FillScreen(void) {
    uint32_t color = 0x00000000
    int x, y;
    for (y = 0; y < tft.height(); y++){
        for (x = tft.width(); x >= 0; x--){
            tft.drawline(x , y, x, y, color >> 16);
            color++;
        }
    }
}

prenticedavid, 2018-02-09:
Please copy-paste real code into a CODE window.   This applies to GitHub messages or Forum messages.
Most Browsers / HTML  programs will mangle source code e.g. Void instead of `void`

Please continue this topic on the Arduino.cc Displays Forum
I will answer it there.

David.

jsweet13, 2018-02-09:
Down to an iPhone for the minute but developed a font in a byte array I'll be printing with to the display writing to the x coordinates in reverse, subtracting from tft.width(). It appears mirrored when using the tft.print() so will have to mess around with the invert_ss option after figuring out which file allows access to it. Going down to binary bits with this, at 8 bytes per char for the pixel artwork.

arnanilmak, 2018-05-05:
Hello.
I'm working on my project and I have the same problem.
ADA_GFX_kbv.h and "Adafruit_GFX.h" are not available and I do not know what to do.
Can anyone help me?

prenticedavid, 2018-05-05:
Run the Arduino IDE.    Install an STM32 core e.g. from ST Microelectronics with the Boards Manager
Install Adafruit_GFX library with the Library Manager
Install MCUFRIEND_kbv library with the Library Manager

David.

arnanilmak, 2018-05-06:
You mean I can use these libraries in my stm32f217 MCU without any changes in the libs?
I'm using Keil for coding. When I use Arduino IDE am I able to continue my project with Keil IDE?

prenticedavid, 2018-05-06:
No,  I do not support the F217.   If you have a NUCLEO-144 with an Arduino Core from STM,    I will add support. 

Yes,  you can use Keil but I would expect you to know how to clone projects from GitHub.    And know how to use Keil.

The library is designed for Arduino users.     Plug a shield in,   install a library,   run examples.

It needs intelligence if you want to do something different.

David.

arnanilmak, 2018-05-08:
Thanks.
I installed the Arduino IDE and found the ada fruit libs. For using these libs in my stm32f217 I have to port all the libraries in all of the folders(extra folder and utility folder )?
Would you tell me which are the must and which are not?

arnanilmak, 2018-05-08:
I cant find this ADA_GFX_kbv.h that is included in MCUFRIEND_kbv.h either. But Arduino itself doesn't make any error that this lib does not exist.

prenticedavid, 2018-05-08:
The STM Core does not support the F217.    It does support the Nucleo-144 with F207ZG.

If you want to use standard Arduino facilities,  you need to have a supported target.

When I build in Keil or Rowley,    I just write trivial hardware support files.     And link with "Ada_GFX_kbv.cpp" to provide the simple graphics methods.    It uses sprintf() to implement the Arduino Print.h class.

Yes,  you would have to write / port a whole series of Arduino files if you want to inherit the Print class in the normal way.

I do provide "mcufriend_keil.h".    I have no intention of providing the low level files.    I can email Ada_GFX_kbv.cpp to you.

What do you actually want to do?

David.

arnanilmak, 2018-05-08:
I will be thankful if you Email the Ada_GFX_kbv.cpp to me. arman_ilmak@yahoo.com this is my email address.

I'm working on my university project and I needed a TFT to display some data on. I searched in the bazaar but the TFTs in which had libs that supported stm32f217 were too small for me. So i bought this TFT and i didn't think it will be too hard for me to port the libs.

closed 2018-12-04 by prenticedavid

moahrs, 2020-05-28:
And where can I download ADA_GFX_kbv.h and ADA_GFX_kbv.cpp? I needed for keil and STM32