prenticedavid/MCUFRIEND_kbv #40: iLI9488 colors are off.

MohammadMubin, 2018-03-23:
Hi,I i have used your library for the ILi9488 MCUFRIEND Display  made for Ardunio UNO and i confirmed it with the test sketch and it works but there were lines and boxes and it looked terrible and the colors were completely off!
I changed the following settings in the .cpp file from `0x3A, 1, 0x55,      //Interlace Pixel Format [XX]` to 
`0x3A, 1, 0x36,      //Interlace Pixel Format [XX]` and it made improvement but the colors were not 'punchy' like red,green,blue were bit off. Any Solutions?

prenticedavid, 2018-03-23:
I have answered your question on the Arduino.cc Forum.  [here](https://forum.arduino.cc/index.php?topic=537035.msg3660213#msg3660213)

Your photo is very interesting.   The "61503" might mean that it is a Renesas R61503U
However the R61503 is a 176x220 controller.   Your screen behaves like 9488_555 hardware problem. 

David.

p.s. unrelated questions   
Does anyone have a FocalTech FT1509 datasheet?
Does anyone recognise a controller with ID=0x9401 and have a datasheet?

MohammadMubin, 2018-03-23:
Hmm i wasn't aware that manufactures can 'spoof'(or if that is what i can call it) there 'ids' interesting...  anyways i need to know why to changes from 0x55 to 0x36 worked i mean in the datasheet there was nothing about this... i found this on a another forum Thanks :D

prenticedavid, 2018-03-23:
As a general rule,  different manufacturers have their own IDs and part numbers.

Sometimes "popular" controllers are made by different makes.  e.g. Ilitek ILI9325
Himax HX8347-B has ID=0x9325
Raydium RM68050 has ID=0x9325

The Raydium RM680140 has ID=0x9486 and an undocumented register that tells you it is 6814

Likewise,  the popular 132x162 controllers like ST7735S have MANY competitors that will return the same ID=0x7C89F0 or the ILI9163C with ID=0x548066.   In fact reg(0x04)=0x548066uL is returned by many controllers of many geometries.

It is handy that the MIPI spec enables User programs to behave the same.    Most of the screens appearing on Ebay at the moment are probably obsolete Phone replacement screens.    These 10-15 year old controllers often have "features".    (which is why they are obsolete)

MohammadMubin, 2018-03-23:
Well this sums it up i never have been this confused like right now but thanks for clearing this up :D also another off topic question (sorry) the controller i got does it have SPI?  since the datasheet does suggest that but since you said that the 'rule' is the 'different manufacturers have their own IDs and part numbers.' does it imply to this display? and how  can i be certain who produced this display  without poking around the display it self. ( gee wiz i am really sorry for the inconvenience)

prenticedavid, 2018-03-23:
Your Uno shield has got a microSD that works with SPI.    As shown by your Tiger photo on the Arduino forum.

The TFT has an 8080-8 8-bit parallel interface.    You can NOT use SPI.    Anyway,  your screen is badly cracked.

Please continue these conversations on the Arduino Forum.    GitHub issues are more suited to source code programming problems.

David.

closed 2018-03-23 by MohammadMubin