prenticedavid/MCUFRIEND_kbv #42: Support for Adafruit 2050 TFT (HX8357D)

pilant, 2018-03-27:
I'm using an Adafruit 2050 TFT/touch display with the TFT-VFO-ubitx2.9 (GitHub) project and for some reason the colors used for the display appear to be inverted.  If I use the Adafruit TFTLCD-Library (with the ubitx code), the colors appear to be mostly normal.  The only hardware difference is using the ATmega2560 pins 22-29 for data (TFTLCD-Library) instead of the UNO 2-9pins for data (MCUFriend).

I looked through the 8357 code to see if I could spot anything, not there did not appear to be anything obvious that would account for the color inversion.

I did try the invertDisplay() method, but it did not appear to have any effect.

Any ideas?

- Mark

pilant, 2018-03-27:
BTW, I did try the graphictest_kbv code, and the colors are inverted from what appears in the Adafruit graphicstest code.  (The code was pulled March 24, 2018.)

More than happy to try any modifications you might suggest.

- Mark

prenticedavid, 2018-03-27:
Please provide links to your Adafruit hardware and your GitHub project.

Then ask a specific question.    Either on GitHub or on the Arduino Displays Forum.

David.

pilant, 2018-03-27:
GitHub project: https://github.com/sprakashb/TFT_TouchScreen_for_uBitx
Display: https://www.adafruit.com/product/2050
ATmega2560: https://www.banggood.com/ATmega2560-Funduino-Mega-2560-Arduino-Compatible-Improved-Version-p-928760.html?cur_warehouse=CN

Specific question: Why are the default colors for the MCUFriend 8357 code inverted from the Adafruit 8357 code?  (I can "fix" this by calling invertDisplay(), but would like to understand if this is expected or a possible bug.

- Mark

prenticedavid, 2018-03-27:
Your GitHub sketch seems to be configurable for different displays e.g. 
`#define         VE0x7783    // Ventor Tech display`

The MCUFRIEND_kbv library normally reads the ID.   Then calls the appropriate `tft.begin(ID)`;

If you are using that particular Adafruit display,   I suggest that you select the test_8357X Branch of the library.   Run the graphictest_kbv example and report back with ID, colours, directions, ...

My car is Blue-Grey colour.   My next-door neighbour has a red sports car,  red pickup and two silver cars.   Why is that?

David.

pilant, 2018-03-28:
I'm actually running with a local copy of the GitHub sketch in an attempt to get the display working.  I did add the code "identifier = 0x9090;" which is the id returned by the readID() method in the MCUFriend code.  So "tft.begin() is called with the value 0x9090.

The graphictest_kbv test returns the same id.   As I mentioned "above" graphictest_kbv runs fine except the colors are inverted.  (At the end of the run, where it does the color inversion test, the colors match what is seen by the Adafruit tests.)

- Mark

prenticedavid, 2018-03-28:
As I explained earlier,   my car is a different colour to Louise's car.
Your Adafruit TFT panel is different to the Mcufriend TFT panel.

You can either change the REV_SCREEN attribute in the `case 0x9090:` block
Or call `tft.invertDisplay(); `

However,   I would appreciate it if you installed the test_8357_99 Branch.
It is probably easier to continue this discussion on the Arduino Forum.   I have even posted a special LCD_ID_readreg sketch there.

David.

pilant, 2018-03-28:
Sorry, my bad.  I didn't understand the "My car" statement.

I'm going to test the _99 code shortly...  If I got the new library installed correctly (I deleted the original from the Arduino\libraries folder and there is now a MCUFRIEND_kbv-test_8357_99 folder) the results of the graphictest are the same.

I guess my root questions is: assuming the Adafruit and MCUFriend initialize / reset the display to the same state, why would one display a "red" color as red (Adafruit) while the other display displays the inverted color.  The only thing I can  think of is the initialization isn't really the same.  (I couldn't find anything obvious in the code for either which issued the inversion command to the display controller.)

I can take this to another forum.  Do you have a pointer?

- Mark

pilant, 2018-03-28:
Just thought of a clearer simpler explanation of the issue.

Yes the panels may be different.  But using the same display (and HX8357D controller), if I write text to the display using the color code 0x0000, the text is black with the Adafruit code but white with the MCRFriend code.  I would expect 0x0000 to be black regardless of the code base.  It only appears as black with the MCRFriend code if I call invertDisplay().

I hope this doesn't confuse things even further :-)

- Mark

prenticedavid, 2018-03-28:
Different panels have different characteristics.  Just like my car is a different colour to my neighbour's.

The modified sketch is attached to 
[this message on Arduino forum](https://forum.arduino.cc/index.php?topic=366304.msg3634582#msg3634582)

Seriously.   There is no way that I can detect whether you have an Adafruit or an Mcufriend shield.
My 0x9090 initialisation is trivial.    It just uses the power-up defaults.

And your "project" is designed for a 240x320 screen.    Your HX8357-D is 320x480.

It is more appropriate to discuss Arduino sketches on an Arduino Forum.
And discuss technical C++ library problems on GitHub.

David.

pilant, 2018-03-28:
OK.  Let's reset to eliminate non-issues.

First, I'm using a display with an HX8357-D display controller.  Second, the software running is the MCRFriend graphictest_kbv.ino sketch on an ATmega2560 board.  (Ignore anything about the original sketch.)  So I believe we can ignore any Arduino or C++ concerns, as well as the display geometry.

The results show whenever the graphictest_kbv.ino sketch writes anything WHITE ("0xFFFF") it appears on the display as black; anything written BLUE (0x001F) appears as a YELLO (0xFF1E0), etc.

Here are the results from running the register sketch.  Note Display Mode is 0x00 (which should be inversion off) and the ID is 0x99:

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x000A) 00 08	Get Power Mode
reg(0x000B) 00 00	Get Address Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x000D) 00 00	Get Display Mode
reg(0x000E) 00 00	Get Signal Mode
reg(0x000F) 00 00	Get Diagnostic Result
reg(0x00A1) 00 00 00 00 00 FF	RD_DDB Start
reg(0x00B0) 00 66 00	SETOSC
reg(0x00B1) 00 00 11 1B 1B 83 77 2F	SETPOWER
reg(0x00B2) 00 00 83 3B 00 20	SETDISPLAY
reg(0x00B3) 00 00 00 06 06	SETRGB
reg(0x00B4) 00 02 40 00 2A 2A 0D 96	SETCYC
reg(0x00B5) 00 0B 0B 66	SETBGP
reg(0x00B6) 00 4B 00	SETCOM
reg(0x00B7) 00 00 00 00 00 00	SETOTP
reg(0x00B9) 00 FF 83 57	SETEXTC
reg(0x00C0) 00 73 50 00 3C C4 08	SETSTBA
reg(0x00C1) 00 00 00 08 10 18 20 28 30 38 40	SETDGC
reg(0x00C3) 00 00 80 00 00	SETID
reg(0x00C4) 00 00 00 00 00	SETDDB
reg(0x00C9) 00 0F 82	SETCABC
reg(0x00CC) 00 00	SETPANEL
reg(0x00D0) 00 99 00 00	GETICID
reg(0x00E0) 00 0B 0C 11 1D 25 37 43 4B 4E 47 41 39 35 31 2E 21 1C 1D 1D 26 31 44 4E 56 44 3F 39 33 31 2E 28 1D E0 01	SETGAMMA
reg(0x00E9) 00 00	SETIMAGEI
reg(0x00EA) 00 02	SETMESSI

So why the test appears to have inversion on is a mystery.

- Mark

prenticedavid, 2018-03-29:
Yes,  you have the same controller as Phaedra and one or two other people.
reg(0xD0) is not documented in the HX8357-C datasheet.
I have only seen 0x90 in reg(0xD0) from my display.   My controller is probably HX8357-C

Note that none of the readreg reports match the power-on default values of each register.
It is not unusual for datasheets to have "features".    The current library Release accepts 0x90 and 0x99 and will identify as 0x9090 and call it HX8357-D

I suggest that you add or remove the REV_SCREEN attribute from the `case 0x0099:` block.
This will invert the display.  Otherwise you need to call `tft.invertDisplay(1);` in every sketch.

David.

pilant, 2018-03-29:
David, thank you very much.

Using the _99 code, setting the ID to 0x0099, and removing the REV_SCREEN from the MCUFriend code now has the display as expected.

Please feel free to close or keep open this issue as you need.

- Mark

closed 2018-05-16 by prenticedavid