prenticedavid/MCUFRIEND_kbv #43: id 0x1602.....how to solve LCD_ID_readreg.ino Output is ...please

vilasrao, 2018-03-30:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 38 80 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 06 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 A6	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 24 02 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 0A 0A 0A 0A 0A 0A 0A 0A 0A	Panel Control
reg(0x00C8) 40 40 40 40 40 40 40 40 40 40 40 40 40	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 01 16 02	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 38 38	RDID1
reg(0x00DB) 80 80	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 22 23 25 08 10 14 40 7B 50 0B 1B 22 20 2F 37	GAMMA-P
reg(0x00E1) 00 0C 14 23 0E 14 15 36 59 46 0B 1F 27 1F 20 22	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 24 04 04 1E 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2018-03-30:
`reg(0x00D3) 00 01 16 02 ILI9341, ILI9488
`

You have an ID=0x1602 controller.   I do not own this 0x1602.   I do not have a datasheet.   I do not even know the Manufacturer.

Seriously.   You should ask your question on the Arduino Forum.    After actually "composing" a question.
Please quote library version,  which "Example" sketch you are running and what is going wrong.

David.

vilasrao, 2018-04-01:
Thanks for quick reply! 
I contacted the seller but he don't know the driver ic..

prenticedavid, 2018-04-01:
Go on.   Please ask a question.   Please quote library version, which "Example" sketch you are running and what is going wrong.

As far as I know,   the 0x1602 works.    But I can not fix a problem if I do not know what problem you have.

Likewise,   your seller would probably help if you were asking for your money back e.g. if the display does NOT work.    If the seller does not know the driver i.c. she can ask her supplier.
It costs nothing for a seller to email / phone her supplier.    Far better for business than refunding unhappy customers.

David.

vilasrao, 2018-04-01:
Thanks a lot! Your library working for 0x1602! Actually the problem was loose connection! After several attempt of uploading graphictest_kbv sketch three times  the tft started working perfectly.. Again fpc connection lost
Now I know the problem is in Fpc strip connection!

vilasrao, 2018-04-01:
Sorry for my poor english!

closed 2018-04-08 by prenticedavid