prenticedavid/MCUFRIEND_kbv #45: ESP32 not compiling

laurentopia, 2018-04-28:
GLUE_Demo_480x320.ino:11: In file included from
Error compiling project sources
Build failed for project 'GLUE_Demo_480x320'
UTFTGLUE.h: In member function void UTFTGLUE::print(String, int, int, int)
 
UTFTGLUE.h: 102:26: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
    settextcursor(st.c_str(), x, y); MCUFRIEND_kbv*: print(st);}
UTFTGLUE.h:155: note    initializing argument 1 of void UTFTGLUE  settextcursor(char*, int, int)
   void settextcursor(char *st, int x, int y) {

prenticedavid, 2018-04-28:
ARM and AVR targets use the GCC compiler
ESP32 and ESP8266 targets use the Expressif compiler.

Yes,  there are slight nuances between the strictness of each compiler e.g. const char* versus char*.
My Arduino does not get upset by the const
It does get upset by undefined A1, A2 symbols and by looking for dtostrf.h

Thanks for your feedback.    But I need more information.

Which MCUFRIEND_kbv Release or which Branch are you using?
The Library Manager reports the version.    The Manager always installs a "Release"
Downloading an installing from ZIP or Forking a Beta branch from GitHub requires clarification.

Which version of Arduino IDE?
Which version of ESP32 tool package?

Are you using a proper shield plugged into a TTGO R32 board?
Using a bare module even with soldered wires is a bit flaky as a hardware target.

David.

laurentopia, 2018-04-28:
Thanks for the comprehensive response, I wasn't aware the compilo was so different.

mcu v 2.9.7
ide 1.8.4
esp32 hardware lib               "version": "1.22.0-80-g6c4433a-5.2.0"
I'm using a shield connected via wires as the board isn't shield comatible with an arduino shield

prenticedavid, 2018-04-28:
Looking at the tools/dist folder,  I find xtensa-esp32-elf-win32-1.22.0-75-gbaf03c2-5.2.0.zip
So my tools are slightly different.    My IDE is 1.8.1.    I am using a Beta version of the library.

You can solve the "error" by casting.    I will update my ESP32 installation one day.
UTFTGLUE is a handy way to run a legacy UTFT project.    Not something for a new project.

David.

laurentopia, 2018-05-01:
I didn't try the native GFX because it doesn't have a parallel 8 support, or I haven't seen one.
I tried TFT_eSPI and it doesn't work either and Boldmer gaveme this register thingy, don't know what it does, we'll see.

prenticedavid, 2018-05-01:
Most hardware libraries inherit the Adafruit_GFX class (or implement conforming methods)
UTFT is a unique class that does not conform to the Adafruit_GFX class.

I provide the UTFTGLUE class as an easy way to run a legacy UTFT program.    **You should not use it in a new project.**
Bodmer provides a UTFT_demo that has every UTFT call hand-edited to use the equivalent GFX method. 
Hand-editing for beginners is not practical.

Most examples and sketches in the Arduino world use the Adafruit_GFX style of Graphics methods.    It is easy to re-build a TFT program for SPI ST7735 or a big parallel SSD1963.    They all use the same Adafruit_GFX and Print calls.

**I would expect you to try recommended library examples first.**
It is important to resolve any problems with the native library before looking at UTFTGLUE.

It would be more appropriate to continue this discussion on Arduino.cc forum.    You can provide links to your conversations with Bodmer.     Messages are more suited to "user" experience.    e.g. "thingy"
GitHub is better for collaborative programming questions.

David.

closed 2018-05-16 by prenticedavid