prenticedavid/MCUFRIEND_kbv #47: Esp32 3.5" ili9488 blank screen

ddjikic, 2018-04-29:
Hi can someone help me I am getting a white screen and 
the ID varies  0x0 , 0xd3d3, 0x200 
what am I doing wrong

prenticedavid, 2018-04-30:
It is better to post this sort of question on the Arduino.cc Displays Forum.

Please post a link to the screen e.g. the Ebay sale
Please post a link to the ESP32 Wemos/TTGO R32 board e.g. the Ebay sale
Please post a photo of your hardware modification to the R32 board. 

Run the LCD_ID_readreg sketch with the correct defines for your ESP32

David.

ddjikic, 2018-04-30:
Hi sorry 
the screen is working with Arduino mega but its too slow

screen 
https://www.ebay.com/itm/3-5-inch-TFT-LCD-Display-Arduino-Touch-Screen-Module-UNO-R3-Board-Plug-and-Play/253562964975?hash=item3b0987cbef:g:-vwAAOSwIMxa0wWe

esp 32 
https://www.ebay.com/itm/ESP-32S-ESP32-NodeMCU-Development-Board-2-4GHz-WIFI-Bluetooth-Dual-Mode/272656534488?hash=item3f7b988bd8:g:oPIAAOSw7U5Y-Hh0

I wired on a breadboard like  the pins in mcufriend_shield.h
#elif defined(ESP32)       //regular UNO shield on TTGO D1 R32 (ESP32)

thank you
Best regards

prenticedavid, 2018-04-30:
You have a regular Uno shield.
It is better to buy a R32 (ESP32) board and make the hardware modifications.

I would never use a parallel display with a breadboard.
Test the wiring with the LCD_ID_readreg sketch.

David.

ddjikic, 2018-05-02:
Until it arrives I hand soldered all the pins 
LCD_RD  to IO2  
LCD_WR to IO 4
LCD_RS to IO15
LCD_CS to IO33 
LCD_RST to IO32

LCD_ID_readreg gives only one response

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 04 04 04 04	Manufacturer ID
reg(0x0009) 09 09 01 01 01	Status Register
reg(0x000A) 0A 0A	Get Power Mode
reg(0x000C) 0C 0C	Get Pixel Format
reg(0x0061) 61 61	RDID1 HX8347-G
reg(0x0062) 62 62	RDID2 HX8347-G
reg(0x0063) 63 63	RDID3 HX8347-G
reg(0x0064) 64 64	RDID1 HX8347-A
reg(0x0065) 65 65	RDID2 HX8347-A
reg(0x0066) 66 66	RDID3 HX8347-A
reg(0x0067) 67 67	RDID Himax HX8347-A
reg(0x0070) 70 70	Panel Himax HX8347-A
reg(0x00A1) A1 A1 A1 A1 A1	RD_DDB SSD1963
reg(0x00B0) B0 B0	RGB Interface Signal Control
reg(0x00B4) B4 B4	Inversion Control
reg(0x00B6) B6 B6 A6 A6 A6	Display Control
reg(0x00B7) B7 B7	Entry Mode Set
reg(0x00BF) BF BF AF A7 A7 A7	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 80	Panel Control
reg(0x00C8) C8 C8 C8 C0 C0 C0 C0 C0 C0 80 80 80 80	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 C0	Power Control
reg(0x00D2) D2 D2 C2 C2 C2	NVM Read
reg(0x00D3) D3 D3 C3 C3	ILI9341, ILI9488
reg(0x00D4) D4 D4 C4 C4	Novatek ID
reg(0x00DA) DA DA	RDID1
reg(0x00DB) DB DB	RDID2
reg(0x00DC) DC DC	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 A0 20 A0 A0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 A0 20 20 E0 E1 E1 A0 20	GAMMA-N
reg(0x00EF) EF EF EF E7 E7 A6	ILI9327
reg(0x00F2) F2 F2 EA EA E2 A2 A2 E2 EA EA AA 22	Adjust Control 2
reg(0x00F6) F6 F6 E6 E6	Interface Control

prenticedavid, 2018-05-02:
The readreg report is clearly BAD.

The hardware mod states that you should:
```
#define LCD_RS 15  //hard-wired to A2 (GPIO35) 
#define LCD_CS 33  //hard-wired to A3 (GPIO34)
#define LCD_RST 32 //hard-wired to A4 (GPIO36)

```
and the data bus should be wired:
```
#define LCD_D0 12
#define LCD_D1 13
#define LCD_D2 26
#define LCD_D3 25
#define LCD_D4 17
#define LCD_D5 16
#define LCD_D6 27
#define LCD_D7 14

```
When your wiring is correct,   LCD_ID_readreg should give the same output on any target.

It is a lot easier to just plug a shield into a regular Uno, Due, Nucleo, Zero
It works on a Mega or Leonardo but will be SLOWER than a Uno.

David.

ddjikic, 2018-05-03:
Thank you for your help and sorry for bothering.
 
I am confused with the comment should it be pin 15 or 35 ?
#define LCD_RS 15  //hard-wired to A2 (GPIO35)

prenticedavid, 2018-05-03:
GPIO15 is a digital pin.   GPIO35 is an analog-only pin

For the TFT,  you must connect GPIO15.     GPIO35 is the default pin on the TTGO Analog header.

Have you ordered a TTGO board from Europe or from China?    I got mine from the UK.

David.

ddjikic, 2018-05-04:
I ordered mine from China and I noticed that there is something wrong with pin 4, until the R32 arrives I will try with another esp32 chip

ddjikic, 2018-05-04:
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0b20  PS      : 0x00060730  A0      : 0x800d0ca2  A1      : 0x3ffca1e0  
A2      : 0x00000000  A3      : 0x00000002  A4      : 0x3f400fa7  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d0bd7  A9      : 0x3ffca1c0  
A10     : 0x000000b0  A11     : 0x00000001  A12     : 0x3ffc1028  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d0b20:0x3ffca1e0 0x400d0c9f:0x3ffca200 0x400e3133:0x3ffca220

Rebooting...

prenticedavid, 2018-05-04:
I edited the LCD_ID_readreg sketch like this:
```
#if defined(ESP32)       //regular UNO shield on TTGO D1 R32 (ESP32)
#define LCD_RD  2  //LED
#define LCD_WR  4
#define LCD_RS 15  //hard-wired to A2 (GPIO35) 
#define LCD_CS 33  //hard-wired to A3 (GPIO34)
#define LCD_RST 32 //hard-wired to A4 (GPIO36)

#define LCD_D0 12
#define LCD_D1 13
#define LCD_D2 26
#define LCD_D3 25
#define LCD_D4 17
#define LCD_D5 16
#define LCD_D6 27
#define LCD_D7 14
#else
#define LCD_RST A4
#define LCD_CS A3
#define LCD_RS A2
#define LCD_WR A1
#define LCD_RD A0

#define LCD_D0 8
#define LCD_D1 9
#define LCD_D2 2
#define LCD_D3 3
#define LCD_D4 4
#define LCD_D5 5
#define LCD_D6 6
#define LCD_D7 7
#endif

```
I ran the sketch with an ILI9341 screen:
```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 0A 82 27 04	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 71	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 03 03	NVM Read
reg(0x00D3) 00 00 93 41	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 0F 16 14 0A 0D 06 43 75 33 06 0E 00 0C 09 08	GAMMA-P
reg(0x00E1) 00 08 2B 2D 04 10 04 3E 24 4E 04 0F 0E 35 38 0F	GAMMA-N
reg(0x00EF) 00 03 80 02 02 02	ILI9327
reg(0x00F2) 00 02 02 02 02 02 02 02 02 02 02 02	Adjust Control 2
reg(0x00F6) 00 01 00 00	Interface Control

```
I don't know which controller you have.   But the readreg sketch should report the same on any Arduino.

David.

ddjikic, 2018-05-05:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 02 04 94 81 FF	ILI9481, HX8357-B
reg(0x00C0) 00 10 3B 00 02 11 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 43	Power Control
reg(0x00D2) 00 01 22 00 00	NVM Read
reg(0x00D3) 00 01 22 00	ILI9341, ILI9488
reg(0x00D4) 00 01 22 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 33 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 80 80 80	Interface Control

ddjikic, 2018-05-05:
Thank you, 
I reinstalled everything, changed the USB cable and other stuff now it works

prenticedavid, 2018-05-05:
You have an ILI9481.   
`reg(0x00BF) 00 02 04 94 81 FF ILI9481, HX8357-B`

Obviously your wires are "good enough" for the digitalWrite() style of sketch.
They must be correct for the library.

Have you tried any library examples yet?  e.g. graphictest_kbv.ino

David.

ddjikic, 2018-05-05:
I have my code that worked on the arduino mega I put that and its working fast and fine I would like some more speed but its already smooth enough

closed 2018-05-16 by prenticedavid