prenticedavid/MCUFRIEND_kbv #49: MCUfriend Due example using parallel 8bit and DMA?

laurentopia, 2018-05-06:
I bought a3.5" mcufriend shield.
All mcufriend examples are extremely slow so I think they use SPI and none of the DUE unique features.
Is there a fork of this library that make use of parallel and co?

prenticedavid, 2018-05-16:
The library uses parallel 8080-8 interface with Mcufriend Uno Shields.
The Shield plugs into the header sockets of any regular Arduino.    The wiring matches PORTs quite well on a Uno.    On other Arduinos the wiring involves random PORT bits which makes it less efficient than a Uno.

You can use use custom wiring of the 8080-8 interface as a SPECIAL e.g. routeing the data bus to PORTA on a MEGA2560 or contiguous PORT bits on a Due.   It works very FAST with sensible routeing.

If your display has an 8080-16 parallel interface e.g. Mcufriend for mega2560 it works fast with the appropriate SPECIAL.

A 40-pin display with USE_DUE_16BIT_SHIELD Adapter on a Due is dramatic.   e.g. 800x480 SSD1963

Cortex-M3, Cortex-M4 are much faster than the TFT.    DMA is no good for random PORT pin mapping.

David.

laurentopia, 2018-05-17:
thanks, so the examples on a due and the mcu friend are very fast?

prenticedavid, 2018-05-17:
The standard Adafruit Tests with 8080-8 on a 240x320 screen have a limit of about 0.75 seconds.
The standard Adafruit Tests with 8080-16 on a 240x320 screen are a bit faster.

Uno Shields routed to random PORT pins slow any target down.   i.e. you are limited by CPU
Route to contiguous PORT bits and even a 16MHz AVR works fast.

Cortex-M3 and M4 can drive faster than the TFT controller.   i.e. you are limited by TFT

Seriously.   You either make an Adapter to re-route a Shield or you accept the standard Shield wiring.
Or you make a custom pcb to receive a bare 2.8V/3.3V TFT panel.   Dupont cables are a BAD idea.

David.

laurentopia, 2018-05-18:
Ok I removed all Dupont cables and connected the mcufriend 480x320 shield on the DUE.
What's the filename for the " _standard Adafruit Tests with 8080-8_"?

prenticedavid, 2018-05-18:
I am confused.   You appear to have several GitHub projects of your own.    You must know how to Fork or clone a project.     Or you simply let the Arduino IDE do everything for you:

I would assume that you know how to:
1.  install libraries with the Library Manager
2.  plug a shield into the Due.
3.  upload a library/example from the File menu e.g. graphictest_kbv

Arduino user level questions are best on the Arduino.cc Displays forum
Technical C++ or GitHub questions are best as project Issues i.e. like you have done.  

It is difficult to know what level of knowledge or experience to respond to.
Oh,  and if you post on Arduino.cc **and** GitHub,  please post links.   Then other readers can follow.

David.

closed 2018-12-04 by prenticedavid