prenticedavid/MCUFRIEND_kbv #54: Open-Smart TFT shield not working after updating several library

JazekerXX, 2018-07-19:
hi David,

thx for all the great work! 
i was tinkering with a TFT shield with HX8352B chip when i got an popup in the arduino IDE that there where some library updates. after updating the TFT no longer works (white screen). i've tried to downgrade several library but i don't remember al the ones i've updated. do you know wich one could cause the problem. or is there a log somewhere so that i can undo the right updates? 
i hope you can point me in the right direction...
thanks anyway!

greetings,
Maikel

prenticedavid, 2018-07-19:
You must edit MCUFRIEND_kbv.cpp to enable support for the HX8352B

Since it is Open-Smart you must:
utility/mcufriend_shield.h:  #define USE_SPECIAL
utility/mcufriend_special.h:  #define USE_OPENSMART_SHIELD_PINOUT
MCUFRIEND_kbv.cpp:  #define SUPPORT_8352B

None of these edits require skill.   You just remove the two forward slash characters.
It is explained in the extras/mcufriend_how_to.txt file

David.

JazekerXX, 2018-07-20:
Hi David,

thx a lot! that did the trick. though i tried older versions of several libraries but didn't work. also i found 2 sets of libraries on my computer ( 1 under my documents and 1 under program files x86. ) wich maybe also give some trouble. copied it all to the program files folder. and changed the documents wich you specified in the C:\Program Files (x86)\Arduino\libraries\ .
thx again for the fast reply!

greetings,
Maikel

closed 2018-07-20 by JazekerXX

TJWi, 2018-12-18:
Hi david,

Sorry that i have to ask you on a closed topic but this is relevant. I brought a 2.2" TFT LCD screen, well you get the point if you look at the picture below:
[image]

This is display uses ILI9225 in MCU 8bit configuration. It doesn't work when I edited the following line:
MCUFRIEND_kbv.cpp: #define SUPPORT_9225

It only works after i edited the following lines:
MCUFRIEND_kbv.cpp: #define SUPPORT_9225
MCUFRIEND_kbv.cpp: #define SUPPORT_8352B

why?

prenticedavid, 2018-12-18:
I have no idea what your rats' nest of wires are doing.   It looks like a 5V Uno.   Your ILI9225 panel is 3.3V. 

I would expect the display to work with #define SUPPORT_9225

If you tell me what your rats are up to,   I will plug a 3.3V ILI9225 module into a STM32 or other 3.3V board.

The MCU at the bottom right of of your photo looks like a regular STM32 board with 3.3V GPIO and USB comms.    Why not use that board?

David.

prenticedavid, 2018-12-18:
My ILI9225 display is running graphictest_kbv with SUPPORT_9225

TJWi, 2018-12-18:
Yep, 5v to 3.3v logic level converter. (And yes, what a pain)

I believe the display should work fine with just #define SUPPORT_ILI9225, but in my case here, I have to also add #define SUPPORT_8352B, which I kinda like to know why it only works after I added that. Do you want the readreg()?

That black board is an Atmega2560, so it runs on a 5V GPIO. Didn’t use it because I don’t know why the display isn’t working before.

And this literally is like an Open-Smart tft lcd excluded the shitty pin configuration.

TJWi, 2018-12-18:
Hmmm... I think I’ll do more test, let me know if you need any more info. I’m just happy that it works after tinkering with it for over 5 days.

prenticedavid, 2018-12-18:
My ILI9225 uses the standard Mcufriend wiring.    It is a 3V board that happens to be made by Open-Smart but is NOT the 5V Open-Smart Shield.

I have no intention of changing the wiring on my Protoshield Adapter.   But if you want to use the USE_OPENSMART_SHIELD_PINOUT_UNO style,  you need to activate the SPECIAL.

Yes,   as always.   Run the LCD_ID_readreg sketch.   It should report `reg(0) : 92 25`

David.

TJWi, 2018-12-19:
reg(0x0000) 92 25	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 02 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 01 00 01 00 01 00 01 00 01	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 18 AA 18 AA 18 AA	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

Thanks for the assistance again.

prenticedavid, 2018-12-19:
Your wiring is clearly ok.   LCD_ID_readreg verifies it.   My display gives exactly the same output.
Your level shifters must be working correctly.   i.e. bidirectional data bus.

You have never posted the  level shifter schematic.
You have never said which library version.   The Arduino Library Manager would install v2.9.8 Release.
I used the current Beta that is on my PC.    But I would expect any Release since v2.9.7 to work.

Looking at v2.9.7 Release,   the SUPPORT_9225 is defined.
Looking at v2.9.8 Release,   SUPPORT_9225 is not defined by default.

I suggest that you delete your current directory.   Then re-install via the Library Manager.   Enable the define. 
There is no point in forking or using a Branch.    Unless you have a special need.     But I am intrigued to know how 8352B would interfere with 9225.

David.

TJWi, 2018-12-19:
I followed Adafruit for the level shifter schematic. (74LVC245)
[image]

reinstalled your library and did some test again, and this is what I've found. 

There is nothing wrong with or without 8352B. Tbh, it might have to do with the connection with those rats nest wires that gives me inconsistent LCD_ID_readreg values before I manage to get it working. I guess this is a false alarm on my part, I apologies. I'm sorry.

prenticedavid, 2018-12-19:
The Adafruit schematic looks fine to me.
Note that many Arduino hardware shield designs are plain WRONG.   e.g. Mcufriend,  Arduino ST7735.

I can never understand why anyone chooses 5V GPIO.    It is painful.
If I was Microchip,   I would specify AVR controllers at 3.3V.    Just say that they are 5V tolerant.     I would **not** specify a higher F_CPU for different voltage.    Only derate for 1.8V.

In practice,  AVRs work fine @3.3V and 16MHz (at room temperature)
I would not advise for a commercial product.    Hey-ho,   God invented ARM processors. 

David.

prenticedavid, 2018-12-19:
Yes,   I like AVR too.   But I tend to run on Uno clones with switchable 3V GPIO.

The beauty of Arduino is that you can run the same sketch on Uno, Zero, Due,  STM32, ESP32, ... 

Yes, LVC245 buffers work well.    And I have no problem with a proper pcb.    But quite honestly,   how can a rats' nest of unsoldered wires ever work reliably? 

David.

TJWi, 2018-12-19:
well, since i know i'm worst then any actual electronic student who knows what works and what not by reading the document, my method for prototyping probably isn't recommended, but i want to make sure everything works as intended before i draw the schematic for the product i'm designing and PCB the circuit then designing a cover for it.

As for the Rat nest wiring, also isn't recommended, I should be buying the those single wire measure them to size, cut and attach them like so to prevent any pins crossing over to the other ports.
is either this:
[image]

or good luck trying to figure out what is wrong with your circuit:
[image]