prenticedavid/MCUFRIEND_kbv #58: tft.readID() finds: ID = 0x4040 This ID is not supported

doleron, 2018-07-29:
Hello, 

I tried to run the library with an mcufriend LCD TFT 2.4 with no good results. This is the output on graphictest_kbv.ino: https://www.youtube.com/watch?v=Pyezm6JvmiY

This is the output from LCD_ID_readreg.ino:

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 26	Get Pixel Format
reg(0x0061) 40 40	RDID1 HX8347-G
reg(0x0062) 40 40	RDID2 HX8347-G
reg(0x0063) 40 40	RDID3 HX8347-G
reg(0x0064) 40 40	RDID1 HX8347-A
reg(0x0065) 40 40	RDID2 HX8347-A
reg(0x0066) 40 40	RDID3 HX8347-A
reg(0x0067) 40 40	RDID Himax HX8347-A
reg(0x0070) 40 40	Panel Himax HX8347-A
reg(0x00A1) 40 FF FF 40 40	RD_DDB SSD1963
reg(0x00B0) 40 40	RGB Interface Signal Control
reg(0x00B4) 40 40	Inversion Control
reg(0x00B6) 40 40 40 40 40	Display Control
reg(0x00B7) 40 40	Entry Mode Set
reg(0x00BF) 40 41 62 55 60 42	ILI9481, HX8357-B
reg(0x00C0) 40 40 40 40 40 40 40 40 40	Panel Control
reg(0x00C8) 40 40 40 40 40 40 40 40 40 40 40 40 40	GAMMA
reg(0x00CC) 40 40	Panel Control
reg(0x00D0) 40 40 40	Power Control
reg(0x00D2) 40 40 40 40 40	NVM Read
reg(0x00D3) 40 40 40 40	ILI9341, ILI9488
reg(0x00D4) 40 40 40 40	Novatek ID
reg(0x00DA) 40 40	RDID1
reg(0x00DB) 40 40	RDID2
reg(0x00DC) 40 40	RDID3
reg(0x00E0) 40 40 40 40 40 40 40 40 40 40 40 40 40 40 40 40	GAMMA-P
reg(0x00E1) 40 FF FF 40 40 40 40 40 40 40 40 40 40 40 40 40	GAMMA-N
reg(0x00EF) 40 40 40 40 40 40	ILI9327
reg(0x00F2) 40 40 40 40 40 40 40 40 40 40 40 40	Adjust Control 2
reg(0x00F6) 40 40 40 40	Interface Control

Kind regards,
Luiz

prenticedavid, 2018-07-29:
It is a regular lookng mcufriend shield.
You have plugged it into the Uno correctly.
I do not recognise that controller or have ANY idea of the manufacturer, but it is probably MIPI-compliant.

You can try to force tft.begin(0x9329) and any 240x320 geometry IDs
Please let me know if you get a response.

What country are you in?

David.

doleron, 2018-07-30:
Hi, David. Thank you for reply. Yes, I guess I have plugged it right on my UNO.
I just tried to force adding tft.begin(0x9329) in the ending of setup(). Now I have a random output as response: https://www.youtube.com/watch?v=pJdvoIsquZA
If I remove out the line tft.begin(0x9329) I go back to prior state and the screen blink one time white and then gets gray.
I'm from Brazil.
Kind regards, Luiz.

prenticedavid, 2018-07-30:
Ok,  so it looks like we can get this going.    I will study the video later.

prenticedavid, 2018-07-31:
The video is really weird.   There are brief flashes of visible text at 0:26, 0:58, 1:24 corresponding to 0x9320, 0x1505, 0x5408 which implies that it is an ILI9320 style controller.
I can not pause the video to examine the picture.

However you get output with 0x9329 which uses completely different registers because it is MIPI-style.

Have you tried forcing tft.begin(0x9320) in graphictest_kbv sketch?

My advice is:  ask the Ebay vendor to identify the controller.   (she just has to ask her suppier)
If the Ebay vendor will not help,  ask for a refund.

If you lived in the UK,   it is practical to mail the shield to me.    Not sensible from Brasil.

David.

doleron, 2018-07-31:
Hi David,
I have manually checked graphictest_kbv with every single ID found in the code (9320 too) with no success. I have several megas and other devices (nodemcu) here but just only one UNO, the one I have used in the videos. I'm planning to try another UNO board and test again but since I'm in travel right now it should spend some days.
I have no control where this device was actually bought. I will reply here as soon I have the results with another UNO board.

prenticedavid, 2018-07-31:
There is no point in changing Uno for Mega or anything else.   They should all work exactly the same.

If you were in the UK,   a £0.70 postage stamp and I could try it for myself.

Your 0x9329 video makes it look like a MIPI-style.   Your readReg(0x000A) looks like a MIPI-style.
Your diagnose_D3D3 looks like a ILI9320-style.

Just put it down to bad luck and ask for a refund.
It would be nice if the vendor asked her supplier but pigs will probably fly before this ever happens.

David.

doleron, 2018-08-01:
My email is doleron@gmail.com. Please could you send me the address details in order I can send the board for you?

closed 2018-08-24 by doleron