prenticedavid/MCUFRIEND_kbv #59: BOBUINO & ili7783 (& ili9481)

kluv58, 2018-08-04:
I welcome you David. 
I try to connect to Atmega 1284P the display on the ili7783 controller. everything is collected according to:

//################################### UNO SHIELD on BOBUINO ##############################
#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__) //UNO shield on BOBUINO
#warning regular UNO shield on BOBUINO
#define RD_PORT PORTA
#define RD_PIN  7
#define WR_PORT PORTA
#define WR_PIN  6
#define CD_PORT PORTA
#define CD_PIN  5
#define CS_PORT PORTA
#define CS_PIN   4
#define RESET_PORT PORTA
#define RESET_PIN   3

#define BMASK         0x0F              //
#define DMASK         0x6C              //
#define write_8(x)    { PORTB = (PORTB & ~BMASK) | ((x) >> 4); \
        PORTD = (PORTD & ~DMASK) | ((x) & 0x0C) | (((x) & 0x03) << 5); }
#define read_8()      ( (PINB << 4) | (PIND & 0x0C) | ((PIND & 0x60) >> 5) )
#define setWriteDir() { DDRB |=  BMASK; DDRD |=  DMASK; }
#define setReadDir()  { DDRB &= ~BMASK; DDRD &= ~DMASK; }
#define write8(x)     { write_8(x); WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

#define PIN_LOW(p, b)        (p) &= ~(1<<(b))
#define PIN_HIGH(p, b)       (p) |= (1<<(b))
#define PIN_OUTPUT(p, b)     *(&p-1) |= (1<<(b))

the controller is loaded, but on the screen only blinking in a step of program execution is observed. prompt where and what needs to be corrected for normal work?

prenticedavid, 2018-08-04:
There is no Ilitek ILI7783 controller.
The Sitronics ST7781 controller has ID = 0x7783 (240x320) 
The Sitronics ST7793 controller has ID = 0x7793 (240x400) #define SUPPORT_B509_7793

If your "Bobuino" is wired correctly,   the Serial Terminal will report the ID when you run most of my examples.

Please run the LCD_ID_readreg or LCD_ID_readnew sketch to identify what your controller is.
Or at least post a link or photo of the Shield that you have bought.

It is probably better to report a problem on the Arduino.cc Displays forum.

David.

kluv58, 2018-08-04:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 77 83	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Powsr Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0030) 00 00 00 00 00	PTLAR
reg(0x0033) 00 00 00 00 00 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 0E 0C	RGB Interface Signal Control
reg(0x00B3) 05 54 05 54 05	Frame Memory
reg(0x00B4) 00 00	Frame Mode
reg(0x00B6) 00 44 00 44 00	Display Control
reg(0x00B7) 36 0D	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C1) 00 00 00 00	Display Timing
reg(0x00C5) 00 00	Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00 00	Power Control
reg(0x00D1) 00 0C 00 0C	VCOM Control
reg(0x00D2) 00 01 00	Power Normal
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) B0 00 B0 00 B0 00 B0 00 B0 00 B0 00	Adjust Control 2
reg(0x00F6) 21 11 21 11	Interface Control

kluv58, 2018-08-04:
and the connection is classic for UNO, MEGA
[image]

kluv58, 2018-08-04:
The fact is that the problem is technical, not hardware. After all, displays work fine with UNO, MEGA, and with 1284 do not want.

prenticedavid, 2018-08-04:
You have a regular Sitronics ST7781.   The Registers are being read correctly.

My mcufriend_shield.h has:
```
//################################### UNO SHIELD on BOBUINO ##############################
#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__) //UNO shield on BOBUINO
#warning regular UNO shield on BOBUINO
#define RD_PORT PORTA
#define RD_PIN  7
#define WR_PORT PORTA
#define WR_PIN  6
#define CD_PORT PORTA
#define CD_PIN  5
#define CS_PORT PORTA
#define CS_PIN  4
#define RESET_PORT PORTA
#define RESET_PIN  3

#define BMASK         0x0F              //
#define DMASK         0x6C              //
#define write_8(x)    { PORTB = (PORTB & ~BMASK) | ((x) >> 4); \
        PORTD = (PORTD & ~DMASK) | ((x) & 0x0C) | (((x) & 0x03) << 5); }
#define read_8()      ( (PINB << 4) | (PIND & 0x0C) | ((PIND & 0x60) >> 5) )
#define setWriteDir() { DDRB |=  BMASK; DDRD |=  DMASK; }
#define setReadDir()  { DDRB &= ~BMASK; DDRD &= ~DMASK; }
#define write8(x)     { write_8(x); WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

#define PIN_LOW(p, b)        (p) &= ~(1<<(b))
#define PIN_HIGH(p, b)       (p) |= (1<<(b))
#define PIN_OUTPUT(p, b)     *(&p-1) |= (1<<(b))

```
Note that it is looking for the` __AVR_ATmega1284P__ `macro.    Your quote has removed the double underscores.

Please run ANY sketch on your Bobuino.   It should report an ID on the Serial Terminal.
And if you have turned on verbose build in Preferences,  you should see:
**warning regular UNO shield on BOBUINO**

I have never seen a Bobuino.    I believe that it has proper Arduino header sockets.    You should plug the shield into the female sockets.  Every male pin should be intimately connected to its corresponding socket.
 
David.

kluv58, 2018-08-04:
[image]
This is a homemade board "copy a la BOBUINO" with Atmega1284P on board, quartz 20 MHz and minimal strapping. It is programmed via USBasp.

kluv58, 2018-08-04:
[image]
connections are all checked and called. but does not want to work.

prenticedavid, 2018-08-04:
Your pcb looks nicely made.   It does not have a schematic.   And I certainly can't see any AMS1117 or similar 3.3V regulator.
Your TFT will die if you do not provide 3.3V.
The TFT should be happy if you power with 3,3V from a Chinese USBASP.

Traditional Mcufriend shields power the backlight via an on-shield AMS1117.     These require 5V to supply the AMS1117 and 3.3V as reference for GPIO.    Your shield should work with 3.3V powering GPIO and backlight (which will be dim).

Seriously,    if you want to make a Bobuino,   you should provide all the components including 3.3V regulator and Arduino headers.

Before you go any further,   check that your ST7781 still works on a Uno.
Then sort out your hardware properly.    Connect to Serial Terminal.     Verify your wiring with LCD_ID_read??? sketch running on the Bobuino.

David.

kluv58, 2018-08-04:
connection:
#define LCD_CS 4                 //PA4 
#define LCD_CD 5                //PA5  
#define LCD_WR 6               //PA6
#define LCD_RD 7               //PA7 
#define LCD_RESET 3          //PA3

D0 connects to PB0
D1 connects to PB1
D2 connects to PC0
D3 connects to PC1
D4 connects to PC2
D5 connects to PC3
D6 connects to PC4
D7 connects to PC5

kluv58, 2018-08-04:
This board was made only for testing the capabilities of the 1284P, including working with TFT displays and running programs. after all the experiments, the board will go to waste. so that complicate the scheme did not make sense. the display works perfectly UNO.

prenticedavid, 2018-08-04:
Looking at [https://www.flickr.com/photos/28521811@N04/8451021610](url)
LCD_RD is on analog#0 // PA7
LCD_D0 is on digital#8 // PD5

There is no point in you inventing your own non-Bobuino wiring and expecting the Bobuino shield to work.

If you provide a proper schematic and verify your non-standard wiring with LCD_ID_readreg,    I will write a SPECIAL for you.

Currently,   I am just getting irritated.    You have spelling mistakes in your title and seem to be very uncooperative.

God invented Arduino.   It was good.   The pins were specified so that the whole world knew e.g. in variants/pins_arduino.h

Why have you invented digital#7 for PA7 when it s clearly an analog pin?

David.

kluv58, 2018-08-04:
Well, I will take your comments into account and will continue to be more attentive to the writing of texts.

kluv58, 2018-08-05:
David, good morning!
Now I will determine what I have done.
- I have a board with ATmega1284P with a minimum strapping, assembled according to the scheme:
[image]

with a few changes, removed UART. D1 is on PB7, the RST button is added with a suspender of + 5V through the 10K resistor.
For work, the ARDUINO IDE has a MightyCore core installed.
https://github.com/MCUdude/MightyCore#mightycore
In the kernel settings:
- board: ATmega 1284
- variant: 1284P
- pinout: standart
[image]
Connection:
#define LCD_CS    A4 //  Chip Select goes to Analog 4
#define LCD_CD    A5 // Command/Data goes to Analog 5
#define LCD_WR    A6 // LCD Write goes to Analog 6
#define LCD_RD    A7 // LCD Read goes to Analog 7
#define LCD_RESET A3 // Can alternately just connect to Arduino's reset pin
LCD_D0 - PB0
LCD_D1 - PB1
LCD_D2 - PD7
LCD_D3 - PD6
LCD_D4 - PD5
LCD_D5 - PD4
LCD_D6 - PD3
LCD_D7 - PD2
So the name "BOBUINO" can no longer be remembered. Now we are talking about ATmega1284 with a standard pinout.
David, do you like the Russian? And then with English I have a hard time, I have to use an interpreter.
It is even possible to change the name of the topic so as not to mislead people.

prenticedavid, 2018-08-05:
It looks as if you are using https://github.com/MCUdude/MightyCore/blob/master/avr/variants/standard/pins_arduino.h

No,  I do not understand Russian.    Google Translate does a pretty good job.    It is the best way to communicate.
However,  you must check the spelling of technical terms.   e.g. do NOT say Bobuino when you mean MightyCore/Standard.
do not say ili7783 when you mean Sitronix ST7781

I would like you to verify your wiring by running LCD_ID_readreg sketch.
This involves #defining each "Arduino pin" e.g.
```

#define LCD_RST A3 //A4
#define LCD_CS A4 //A3
#define LCD_RS A5 //A2
#define LCD_WR A6 //A1
#define LCD_RD A7 //A0

#define LCD_D0 0 //8
#define LCD_D1 1 //9
#define LCD_D2 15 //2
#define LCD_D3 14 //3
#define LCD_D4 13 //4
#define LCD_D5 12 //5
#define LCD_D6 11 //6
#define LCD_D7 10 //7

```

If you verify that the wiring is correct,   I will write the SPECIAL for you.
Please note that the performance will be very SLOW.

It does not matter which pins are used for the control pins.
But the data bus will work faster if you put it on one PORT or use consecutive bits on mixed ports.

Seriously.   Life is simpler if you use the Bobuino design.

David.

kluv58, 2018-08-05:
Unfortunately I can not open the port monitor as the download goes through USBasp. Through the USB-UART program produces an error: avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
Apparently something with the loader

prenticedavid, 2018-08-05:
I really think that this subject would be better on the Arduino.cc forum.

You have built a "non-standard" Arduino board.    The first thing to do is test a Blinky.ino sketch.
You have either got a bootloader or you upload via JTAG or ISP i.e. with an external programmer. 

After you have tested blinking an LED with digitalWrite(),   you test a sketch that writes to the Serial Terminal e.g. with Serial.begin() and Serial.print().

You obviously know how to use a Uno.    Your MightyCore/Standard should be able to run exactly the same sketches. 

There will probably be several Russian speakers on Arduino.cc forums that can help you get your MightyCore/Standard to work.    Just write a message (in English) to the Displays section and explain that you are a Russian speaker.

I still think that you would be better with Bobuino.   The Bobuino pins correspond to regular Arduino pins e.g. digital#0 is RX for Serial.

David.

kluv58, 2018-08-05:
The controller ATmega1284P is programmed perfectly, there are no questions. The question of vizualization of recorded programs, i.e. about connecting the controller to the display through your program. Is it possible? If yes, how?

kluv58, 2018-08-05:
> I am quite happy to write a SPECIAL for you. But only when you have proved that your Serial port is working.

 well, I will work on this

kluv58, 2018-08-06:
Dear David. Good afternoon.
I was able to read the data from the port monitor

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 01 01	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 05 05 05 07	Manufacturer ID
reg(0x0009) 09 09 09 0B 0B	Status Register
reg(0x000A) 09 09	Get Power Mode
reg(0x000C) 0D 0D	Get Pixel Format
reg(0x0061) 61 61	RDID1 HX8347-G
reg(0x0062) 61 63	RDID2 HX8347-G
reg(0x0063) 61 63	RDID3 HX8347-G
reg(0x0064) 65 67	RDID1 HX8347-A
reg(0x0065) 65 65	RDID2 HX8347-A
reg(0x0066) 65 67	RDID3 HX8347-A
reg(0x0067) 65 67	RDID Himax HX8347-A
reg(0x0070) 71 73	Panel Himax HX8347-A
reg(0x00A1) 21 23 21 21 21	RD_DDB SSD1963
reg(0x00B0) 31 31	RGB Interface Signal Control
reg(0x00B4) 35 37	Inversion Control
reg(0x00B6) 35 37 35 35 35	Display Control
reg(0x00B7) 35 37	Entry Mode Set
reg(0x00BF) 3D 3D 3D 3D 3D 3F	ILI9481, HX8357-B
reg(0x00C0) 41 43 41 41 41 43 41 41 41	Panel Control
reg(0x00C8) 49 4B 49 49 49 4B 49 49 49 49 49 49 4B	GAMMA
reg(0x00CC) 4D 4F	Panel Control
reg(0x00D0) 51 53 51	Power Control
reg(0x00D2) 51 51 51 51 51	NVM Read
reg(0x00D3) 51 53 51 51	ILI9341, ILI9488
reg(0x00D4) 55 55 57 55	Novatek ID
reg(0x00DA) 59 5B	RDID1
reg(0x00DB) 59 5B	RDID2
reg(0x00DC) 5D 5D	RDID3
reg(0x00E0) 61 61 61 61 61 63 61 61 61 61 61 63 63 61 63 61	GAMMA-P
reg(0x00E1) 61 61 61 61 61 63 61 61 61 61 61 63 63 63 61 61	GAMMA-N
reg(0x00EF) 6D 6D 6F 6D 6D 6F	ILI9327
reg(0x00F2) 71 71 71 71 71 73 73 71 71 71 71 73	Adjust Control 2
reg(0x00F6) 75 75 77 75	Interface Control

Now the decision is yours

kluv58, 2018-08-06:
no, the UNO report at the beginning

kluv58, 2018-08-06:
just at UNO I get: reg (0x0000) 77 83
but on the Atmega1284P: reg (0x0000) 01 01

prenticedavid, 2018-08-06:
Your wiring for  MightyCore/Standard is WRONG.   WROING.  WRONG.

Life would be much easier if you wire the screen for Bobuino.
And select Bobuino as your board.

You would not even need to edit the LCD_ID_readreg defines.   Because Bobuino has the same digital# and analog# pins as Uno, Mega, Due, Zero, ...

David.

kluv58, 2018-08-06:
Хорошо, попробую

kluv58, 2018-08-06:
well, I'll try

kluv58, 2018-08-06:
in the last test I connected the screen for Bobuino

#define LCD_RST A4
#define LCD_CS A3
#define LCD_RS A2
#define LCD_WR A1
#define LCD_RD A0

sorry, but all the same: reg (0x0000) 01 01

kluv58, 2018-08-06:
strange, but some of the answers are:
reg(0x0000) 37 30

kluv58, 2018-08-10:
dear David! How are things progressing on my question?

prenticedavid, 2018-08-10:
I have given up.

I advised you to use a proper 3.3V reference voltage for the level shifters.
I would write a SPECIAL for your MightyCore/Standard **if and when** you verify the wiring.

I advised you to use a proper 3.3V reference voltage for the level shifters.
I advised that it would be much easier for you to use MightyCore/Bobuino pinout.  The library will work without modification.

You have made no effort to provide a 3.3V reference voltage.
You have made no effort to wire correctly for either MightyCore/Bobuino or MightyCore/Standard.

Your shield will plug directly into a Uno, Mega, Zero, Due, Nucleo, ...

David.

kluv58, 2018-08-10:
Mr. David. This display works fine with UNO, MEGA without any drop in the voltage level up to 3.3V. There are two logical level converters SUM74HC245 on the board. So 3,3V does not have to do with it. The wiring has been checked repeatedly: from the controller output to the soldering on the display board. Connection is made according to:
//################################### UNO SHIELD on BOBUINO ##############################

#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644P__) //UNO shield on BOBUINO
#warning regular UNO shield on BOBUINO
#define RD_PORT PORTA 
#define RD_PIN  7
#define WR_PORT PORTA
#define WR_PIN  6
#define CD_PORT PORTA
#define CD_PIN  5
#define CS_PORT PORTA
#define CS_PIN   4
#define RESET_PORT PORTA
#define RESET_PIN   3
your library.
In the settings of the card is selected: BOBUINO
And I do not need to make a SPECIAL offer. You better check your library so that it WORKS with all the claimed displays. And then in your reasoning, the keyword MUST WORK. You can close this question: this topic became not interesting to me. But correct the library!

kluv58, 2018-08-10:
[image]
This is according to your what? And why is it here? You are looking at the documents on the controller, and I'll see the board assembly with the controller and the display. How can I prove that my wiring is OK?

kluv58, 2018-11-27:
Good day!
After 4 months I decided to finish this topic with a fresh head. The solution took 2 hours!

Pinout connection
Pinout taken https://github.com/MCUdude/MightyCore/blob/master/avr/variants/bobuino/pins_arduino.h
#define LCD_RST A4 //PA3
#define LCD_CS A3  //PA4
#define LCD_RS A2  //PA5
#define LCD_WR A1//PA6
#define LCD_RD A0//PA7

#define LCD_D0  D8 //PD5
#define LCD_D1  D9 //PD6
#define LCD_D2  D2 //PD2
#define LCD_D3  D3 //PD3
#define LCD_D4  D4 //PB0
#define LCD_D5  D5 //PB1
#define LCD_D6  D6 //PB2
#define LCD_D7  D7 //PB3
The choice in the Arduino IDE: Standart, Bobuino or Sanguino does not matter.

for visualization of the connection cite the scheme for 'bare' ATmega1284P
[image]
The display was based on the ILI9487 controller, but I think that other displays will work.
Dear DAVID, thank you for your understanding and patience!

prenticedavid, 2018-11-27:
It "looks" like the official Bobuino pin wiring.    I hope you have proper headers.

It will not matter what controller you use.   You can just plug in and go.     The ILI9487 is supported by default.    Not sure what the difference is between ILI9487 and ILI9488.    There is no datasheet.    Can you post a LCD_ID_readreg report?

kluv58, 2018-11-27:
By all means!
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 02 02 3B 3B	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 03	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 09	NVM Read
reg(0x00D3) 00 00 94 87	ILI9341, ILI9488
reg(0x00D4) 00 97 00 01	Novatek ID
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 00 0F 1F 1C 0B 0E 09 48 99 38 0A 14 06 11 09 00	GAMMA-P
reg(0x00E1) 00 0F 36 2E 09 0A 04 46 66 37 06 10 04 24 20 00	GAMMA-N
reg(0x00EF) 00 80 00 10 60 40	ILI9327
reg(0x00F2) 00 58 10 12 02 92 92 FF 12 90 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2018-11-27:
Oops.   I did have a 9487 report.    Your 9487 is similar to ILI9486 but not quite.    e.g. Pixel Format, Gamma, ...

I presume that it is 320x480 and everything works 100% in graphictest_kbv.     Or is it a 480x320 landscape controller? 

David.

kluv58, 2018-11-27:
everything works 100% in graphictest_kbv.

closed 2018-12-04 by kluv58