prenticedavid/MCUFRIEND_kbv #60: Moving A4 pin (reset) to free up i2c SDA pin

PaksuPepe, 2018-08-09:
Hi and sorry to bother.
I am designing a project that would require freeing up the i2c bus from the display pins. I wish to move the screen reset pin from pin A4 to A6  or preferably to digital pin 10. No, this would not mean "mutilating" the beautiful shield, because I have chosen to use an arduino nano (atmega328p-AU) which means using jumper cables anyways. I have tried redefining the pins in the sketch, but I have been unsuccessful at that. I have also tried modifying the mcufriend_special.h file (fearing that i would break something) and that didn't wok out either.
The display I'm using is this one: [ebay link](https://www.ebay.com/itm/3-5-inch-TFT-LCD-Touch-Screen-Display-Module-480X320-fr-Arduino-UNO-Mega-2560-JS/222772682548?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649) and it uses 2 SUM74HC245T chips. The datasheet claims them to be ILI9341 controller chips. Also i am not using the SD pins, so i think the software should not be able to use pins D10-D13.
Help would be appreciated. Thank you!

-Pepe.

prenticedavid, 2018-08-10:
I have answered this question many times on the Arduino.cc forum.
I bet that Google will find many of these replies.

The real mystery is "why use a Nano with a shield?"   A regular Uno or Mega will take up less space and not involve soldering custom wiring  and complex mechanical mounting.

If you still have a problem,  please ask on the Arduino.cc Forum (Displays topic)

David.

closed 2018-08-10 by PaksuPepe

pacaj2am, 2021-02-03:
nice, I asked myself exactly the same question and went here, to get the answer... :-)