prenticedavid/MCUFRIEND_kbv #61: TFT.readID() Returns 0x0, white screen on display.

DanielDiamond7987, 2018-08-10:
Hi, 
I bought two MCUfriend displays the other day, one works fine with your library, but the other only shows a white screen. As mentioned in the title the ID is supposedly 0x0. I had a replacement arrive today from the seller and it is having the exact same problem. The PCB of the issue LCDs is different from the working LCD and both issue LCDs had a sticker reading "8340" on the back. I have tried forcing every ID I have come across (Don't have a list, so would be happy to try any recommended ones.)  [According to the seller the driver chip is a "7781"](http://irishelectronics.ie/epages/950018241.sf/en_IE/?ObjectPath=/Shops/950018241/Products/TFT-LCD). 

I have tried several libraries and solutions found on forums to no avail. I am not really sure what to try next. I would appreciate if you could take a look at the register read below, and maybe point me in the right direction. 

The Following is my result from running the readreg sketch. It is identical for both modules except for reg(0x0067) which reads 92 92 on the other issue module. 

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) DB DB DB DB DB	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 06 06	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) D9 D9	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 49 49	RDID3 HX8347-A
reg(0x0067) 93 93	RDID Himax HX8347-A
reg(0x0070) 28 28	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

[image]
[image]

Thanks, 
  Daniel.

prenticedavid, 2018-08-10:
It is clearly not a ST7781 controller.
The "8340" on the paper sticker suggests that it could be HX8340-A or HX8340-B but these are 176x220 controllers.    And your readreg report does not correspond to either chip.    (which I do not currently support anyway)

I do not recognise your controller.
I suggest that you ask your vendor for your money back.     Please ask her which controller is mounted.    It only takes one email to her supplier.

There is one thing you can try. in LCD_ID_readnew sketch  look for unlock_8357 references and add:

```
uint8_t unlock_8340[]     = { (0xC1), 3, 0xFF, 0x83, 0x40};
...
        unlock = unlock_8340;
        ...

```
See if this reveals more registers.

The other thing to force tft.begin(0x9320) or tft.begin(0x9329) in graphictest_kbv sketch.

But the best strategy is to ask the vendor.    Please let me know how you get on.

David.

DanielDiamond7987, 2018-08-10:
Thanks for the quick reply, unfortunately no new registers were revealed and neither of those IDs cause a difference in operation. I have sent an email to the seller and will hopefully have a reply on Monday.

closed 2018-12-04 by prenticedavid