prenticedavid/MCUFRIEND_kbv #63: SevenSegNumFontPlusPlus font

Fabien69002, 2018-09-02:
Hello,
I would like to use "SevenSegNumFontPlusPlus" font with the MCUFRIEND_kbv library but I don't know how to do this.
I included the SevenSegNumFontPlusPlus.c file in my project folder and I tried to use it as a normal freefont. But it doesn't work. 
What are the different steps to use this font ?
Thanks in advance for your help.
Best regards
Fabien

prenticedavid, 2018-09-02:
It is always better to ask this sort of question on the Arduino Forum.    I bet that I have already posted a file there.    FreeSevenSegNumFontPlusPlus.h

```
// generate FreeFont from UTFT font
const uint8_t FreeSevenSegNumFontPlusPlusBitmaps[] PROGMEM = {
    // 7x7 [0x2E '.']
    0x38, 0xFB, 0xFF, 0xFF, 0xEF, 0x8E, 0x00,
    // 0x0 [0x2F '/']

    // 27x46 [0x30 '0']
    0x03, 0xFF, 0xF8, 0x00, 0xFF, 0xFF, 0x80, 0x3F, 0xFF, 0xF8, 0x03, 0xFF, 0xFE, 0xC3, 0x3F, 0xFF,
    0xBC, 0xF0, 0x00, 0x0F, 0xFF, 0x00, 0x01, 0xFF, 0xE0, 0x00, 0x3F, 0xFC, 0x00, 0x07, 0xFF, 0x80,
    0x00, 0xFF, 0xF0, 0x00, 0x1F, 0xFE, 0x00, 0x03, 0xFF, 0xC0, 0x00, 0x7F, 0xF8, 0x00, 0x0F, 0xFF,
    0x00, 0x01, 0xFF, 0xE0, 0x00, 0x3F, 0xFC, 0x00, 0x07, 0xFF, 0x80, 0x00, 0xFF, 0xF0, 0x00, 0x1F,
    0xFC, 0x00, 0x00, 0xFE, 0x00, 0x00, 0x07, 0x00, 0x00, 0x00, 0x40, 0x00, 0x00, 0x04, 0x00, 0x00,
    0x00, 0xE0, 0x00, 0x00, 0x7F, 0x00, 0x00, 0x3F, 0xF0, 0x00, 0x1F, 0xFE, 0x00, 0x03, 0xFF, 0xC0,
    0x00, 0x7F, 0xF8, 0x00, 0x0F, 0xFF, 0x00, 0x01, 0xFF, 0xE0, 0x00, 0x3F, 0xFC, 0x00, 0x07, 0xFF,
    0x80, 0x00, 0xFF, 0xF0, 0x00, 0x1F, 0xFE, 0x00, 0x03, 0xFF, 0xC0, 0x00, 0x7F, 0xF8, 0x00, 0x0F,
    0xFF, 0x00, 0x01, 0xFF, 0xE0, 0x00, 0x3F, 0x78, 0x00, 0x03, 0xC6, 0x7F, 0xFF, 0x30, 0x1F, 0xFF,
    0xF0, 0x07, 0xFF, 0xFF, 0x00, 0x7F, 0xFF, 0xC0, 0x07, 0xFF, 0xF0, 0x00,
    // 6x39 [0x31 '1']
    0x31, 0xEF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x3C, 0x30, 0x40, 0x00,
    0x33, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFD, 0xE3, 0x00,
    // 27x46 [0x32 '2']
    0x03, 0xFF, 0xF8, 0x00, 0xFF, 0xFF, 0x80, 0x3F, 0xFF, 0xF8, 0x03, 0xFF, 0xFE, 0xC0, 0x3F, 0xFF,
    0xBC, 0x00, 0x00, 0x0F, 0xC0, 0x00, 0x01, 0xF8, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x07, 0xE0, 0x00,
    0x00, 0xFC, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x03, 0xF0, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x0F, 0xC0,
    0x00, 0x01, 0xF8, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x07, 0xE0, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x1F,
    0x80, 0x00, 0x00, 0xF0, 0x7F, 0xFF, 0x86, 0x1F, 0xFF, 0xFC, 0x4F, 0xFF, 0xFF, 0xE4, 0xFF, 0xFF,
    0xF8, 0xE7, 0xFF, 0xFC, 0x1F, 0x00, 0x00, 0x03, 0xF0, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x0F, 0xC0,
    0x00, 0x01, 0xF8, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x07, 0xE0, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x1F,
    0x80, 0x00, 0x03, 0xF0, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x0F, 0xC0, 0x00, 0x01, 0xF8, 0x00, 0x00,
    0x3F, 0x00, 0x00, 0x07, 0xE0, 0x00, 0x00, 0x78, 0x00, 0x00, 0x06, 0x7F, 0xFF, 0x00, 0x1F, 0xFF,
    0xF0, 0x07, 0xFF, 0xFF, 0x00, 0x7F, 0xFF, 0xC0, 0x07, 0xFF, 0xF0, 0x00,
    // 25x46 [0x33 '3']
    0x0F, 0xFF, 0xE0, 0x0F, 0xFF, 0xF8, 0x0F, 0xFF, 0xFE, 0x03, 0xFF, 0xFE, 0xC0, 0xFF, 0xFE, 0xF0,
    0x00, 0x00, 0xFC, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x0F, 0xC0,
    0x00, 0x07, 0xE0, 0x00, 0x03, 0xF0, 0x00, 0x01, 0xF8, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x7E, 0x00,
    0x00, 0x3F, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x0F, 0xC0, 0x00, 0x07, 0xE0, 0x00, 0x00, 0xF1, 0xFF,
    0xFE, 0x19, 0xFF, 0xFF, 0xC7, 0xFF, 0xFF, 0xF8, 0xFF, 0xFF, 0xF8, 0x1F, 0xFF, 0xF1, 0x80, 0x00,
    0x03, 0xC0, 0x00, 0x07, 0xE0, 0x00, 0x03, 0xF0, 0x00, 0x01, 0xF8, 0x00, 0x00, 0xFC, 0x00, 0x00,
    0x7E, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x0F, 0xC0, 0x00, 0x07, 0xE0, 0x00, 0x03,
    0xF0, 0x00, 0x01, 0xF8, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x0F,
    0x07, 0xFF, 0xF3, 0x07, 0xFF, 0xFC, 0x07, 0xFF, 0xFF, 0x01, 0xFF, 0xFF, 0x00, 0x7F, 0xFF, 0x00,
    // 27x39 [0x34 '4']
    0x00, 0x00, 0x01, 0x86, 0x00, 0x00, 0x79, 0xE0, 0x00, 0x1F, 0xFE, 0x00, 0x03, 0xFF, 0xC0, 0x00,
    0x7F, 0xF8, 0x00, 0x0F, 0xFF, 0x00, 0x01, 0xFF, 0xE0, 0x00, 0x3F, 0xFC, 0x00, 0x07, 0xFF, 0x80,
    0x00, 0xFF, 0xF0, 0x00, 0x1F, 0xFE, 0x00, 0x03, 0xFF, 0xC0, 0x00, 0x7F, 0xF8, 0x00, 0x0F, 0xFF,
    0x00, 0x01, 0xFF, 0xE0, 0x00, 0x3F, 0xF8, 0x00, 0x01, 0xFC, 0xFF, 0xFF, 0x0E, 0x3F, 0xFF, 0xF8,
    0x9F, 0xFF, 0xFF, 0xC1, 0xFF, 0xFF, 0xF0, 0x0F, 0xFF, 0xF8, 0xC0, 0x00, 0x00, 0x78, 0x00, 0x00,
    0x3F, 0x00, 0x00, 0x07, 0xE0, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x03, 0xF0, 0x00,
    0x00, 0x7E, 0x00, 0x00, 0x0F, 0xC0, 0x00, 0x01, 0xF8, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x07, 0xE0,
    0x00, 0x00, 0xFC, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x03, 0xF0, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x07,
    0x80, 0x00, 0x00, 0x60,
    // 27x46 [0x35 '5']
    0x03, 0xFF, 0xF8, 0x00, 0xFF, 0xFF, 0x80, 0x3F, 0xFF, 0xF8, 0x03, 0xFF, 0xFE, 0x03, 0x3F, 0xFF,
    0x80, 0xF0, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x07, 0xE0, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x1F, 0x80,
    0x00, 0x03, 0xF0, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x0F, 0xC0, 0x00, 0x01, 0xF8, 0x00, 0x00, 0x3F,
    0x00, 0x00, 0x07, 0xE0, 0x00, 0x00, 0xFC, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x03, 0xF0, 0x00, 0x00,
    0x7C, 0x00, 0x00, 0x0E, 0x7F, 0xFF, 0x81, 0x1F, 0xFF, 0xFC, 0x0F, 0xFF, 0xFF, 0xE0, 0xFF, 0xFF,
    0xF8, 0x07, 0xFF, 0xFC, 0x60, 0x00, 0x00, 0x3C, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x03, 0xF0, 0x00,
    0x00, 0x7E, 0x00, 0x00, 0x0F, 0xC0, 0x00, 0x01, 0xF8, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x07, 0xE0,
    0x00, 0x00, 0xFC, 0x00, 0x00, 0x1F, 0x80, 0x00, 0x03, 0xF0, 0x00, 0x00, 0x7E, 0x00, 0x00, 0x0F,
    0xC0, 0x00, 0x01, 0xF8, 0x00, 0x00, 0x3F, 0x00, 0x00, 0x03, 0xC0, 0x7F, 0xFF, 0x30, 0x1F, 0xFF,
    0xF0, 0x07, 0xFF, 0xFF, 0x00, 0x7F, 0xFF, 0xC0, 0x07, 0xFF, 0xF0, 0x00,
    // 27x46 [0x36 '6']
    0x03, 0xFF, 0xF8, 0x00, 0xFF, 0xFF, 0x80, 0x3F, 0xFF, 0xF8, 0x03, 0xFF, 0xFE, 0x03, 0x3F, 0xFF,
[... cut ...]
    { 625, 27, 46, 32,  2, -48 }, // 0x35 '5'
    { 781, 27, 46, 32,  2, -48 }, // 0x36 '6'
    { 937, 23, 42, 32,  6, -48 }, // 0x37 '7'
    {1058, 27, 46, 32,  2, -48 }, // 0x38 '8'
    {1214, 27, 46, 32,  2, -48 }, // 0x39 '9'
    {1370,  7, 28, 32, 14, -38 }, // 0x3A ':'
};
const GFXfont FreeSevenSegNumFontPlusPlus PROGMEM = {
    (uint8_t  *)FreeSevenSegNumFontPlusPlusBitmaps,
    (GFXglyph *)FreeSevenSegNumFontPlusPlusGlyphs,
    46, 59, 50
};
// FreeSevenSegNumFontPlusPlus bitmap size = 1395 [1525] [2617]

```

Fabien69002, 2018-09-03:
Hello,
I apologize for my mistake. I will post on the Arduino forum next time.
Thank you very much for your help, it works fine !
Best regards
Fabien

prenticedavid, 2018-09-03:
No problem.     I have an Arduino sketch that can convert UTFT, Codevision, MikroElektronika fonts to FreeFont format.

closed 2018-09-03 by prenticedavid