prenticedavid/MCUFRIEND_kbv #65: ESP32 DevKit + ili9341 8-bit parallel

biloge, 2018-09-26:
Hi , my name is Fabio,
I tried your library with Arduino UNO and ili9341 8-bit parallel Shield and it works fine.
Now , I would try it on ESP32 DevKit, but I can't, and I don't understand if is possibile.
This is my Dev Board
[https://4.bp.blogspot.com/-au-q5R7T_Gk/WIJp1zzZikI/AAAAAAAAGJQ/IZ2_MwIiYLAB7sL8TZgbe9jTCLrx6chdACLcB/s1600/IMG_20170117_072209%257E2.jpg](url)

And with is the shield
[http://www.microsolution.com.pk/wp-content/uploads/2017/05/2.4-INCH-TFT-LCD-in-lahore-pakistan.jpg](url)

Thank you , have a nice day
Fabio

prenticedavid, 2018-09-26:
If your shield works with the Uno it should work with the ESP32 too.

Just connect according to the wiring in mcufriend_shield.h

I do not ever recommend using hand-wired modules.   Nature prefers a shield with male pins mating with female sockets on an Arduino-shape board.
It is better to use the ready-made TTGO board (with minor hardware mod)

Bodmer's TFT_eSPI library should work with your shield.    His website has photos of the hardware mod.

David.

biloge, 2018-10-11:
Hi, I tried many times and finally it works :D 
I used with configuration in mcufriend_shield.h

#define LCD_RD  2  //LED
#define LCD_WR  4  
#define LCD_RS 15  
#define LCD_CS 33  
#define LCD_RST 32 

#define LCD_D0 12
#define LCD_D1 13
#define LCD_D2 26
#define LCD_D3 25
#define LCD_D4 17
#define LCD_D5 16
#define LCD_D6 27
#define LCD_D7 14

a-x-, 2019-06-03:
@biloge, can you describe the values?
How pins can be connected?
I want to connect Arduino TFT Display Shield 2.4" to WeMos D1 R2

prenticedavid, 2019-06-04:
Count the LCD_xx pins on the display shield.
Count the header sockets on D1 R2.   Note that several signals are repeated.
Count your fingers and toes.

If you want to use an ESP8266,  choose a TFT with SPI interface.  Use Bodmer's TFT_eSPI library.

David.

fubax, 2020-07-22:
> Count the LCD_xx pins on the display shield.
> Count the header sockets on D1 R2. Note that several signals are repeated.
> Count your fingers and toes.
> 
> If you want to use an ESP8266, choose a TFT with SPI interface. Use Bodmer's TFT_eSPI library.
> 
> David.

Is it really needed to count her fingers and toes?
Absolutely NOT!
Just use esp12e ,connect RD to VCC & CS to GND
And Bingo, Currently you have even unused toes to play with !

prenticedavid, 2020-07-22:
The Shield Touch functions will not work on a ESP8266 which has a single Analog pin.

The TFT library functions will not work properly with CS permanently low.
Obviously Read functions can never work with RD not under program control.

I **strongly** advise people to use SPI displays with ESP8266 e.g. with Bodmer's TFT_eSPI library.
Although 8080-8 parallel displays will work with ESP32 it uses lots of pins.
And ESP32 is excellent for driving SPI displays.

David.

11Naweed11, 2020-12-01:
Hello I need help.
your library does not compile for ESP32.
my display is 2.4 inch, LG DP4532, 8bit.
it works fine will all kinds of arduino boards uno, mega, due, but it gives error with ESP32.

**exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.**

prenticedavid, 2020-12-01:
What Error ?
Please paste the Error message.   
i.e.  4 or 5 lines.   e.g. the actual "g++ command line" and the specific "Error" line(s)
I don't want 1000 lines.

It should compile for the ESP32.
It will only "work" if you `#define SUPPORT_4532`

David.

11Naweed11, 2020-12-01:
Yes I have defined support 4532 otherwise it wouldn't have worked with Arduino boards as well.

**cc1plus.exe: some warnings being treated as errors
Multiple libraries were found for "Adafruit_GFX.h"
 Used: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\Adafruit_GFX_Library
 Not used: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\arduino_510291
Multiple libraries were found for "MCUFRIEND_kbv.h"
 Used: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv
 Not used: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv-master
Multiple libraries were found for "Wire.h"
 Used: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware\espressif\esp32\libraries\Wire
 Not used: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\Wire
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.
**

11Naweed11, 2020-12-01:
by the way other sketches and libraries work fine with esp32, in case if think that the esp32 drivers are not installed.

11Naweed11, 2020-12-01:
the TFT eSPI library does compile and upload but you know it doesn't support 4532.

prenticedavid, 2020-12-01:
I suggest that you leave the IDE.
Delete C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\Adafruit_GFX_Library
Delete C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\arduino_510291
Delete  C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv
Delete  C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv-master
Delete  C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\Wire

Start IDE.   Run Library Manager.
Install Adafruit_GFX
Install MCUFRIEND_kbv
Edit: SUPPORT_4532

File->Preferences: verbose compile

Copy-paste the "Build sketch" command line
Copy-paste the Error line(s)

David.

11Naweed11, 2020-12-01:
I did all of that unfortunately the same results.
below are the messages. do you want me to copy and past all of it?

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'uint16_t MCUFRIEND_kbv::readReg(uint16_t, int8_t)':

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:162:13: warning: unused variable 'lo' [-Wunused-variable]

     uint8_t lo;

             ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'uint16_t MCUFRIEND_kbv::readID()':

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:194:19: warning: unused variable 'ret2' [-Wunused-variable]

     uint16_t ret, ret2;

                   ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'int16_t MCUFRIEND_kbv::readGRAM(int16_t, int16_t, uint16_t*, int16_t, int16_t)':

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:294:22: warning: unused variable 'tmp' [-Wunused-variable]

     uint8_t r, g, b, tmp;

                      ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'virtual void MCUFRIEND_kbv::setRotation(uint8_t)':

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:355:29: warning: unused variable 'REV' [-Wunused-variable]

     uint16_t GS, SS_v, ORG, REV = _lcd_rev;

                             ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:473:11: error: label 'common_ORG' defined but not used [-Werror=unused-label]

           common_ORG:

           ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'void MCUFRIEND_kbv::vertScroll(int16_t, int16_t, int16_t)':

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:719:13: warning: variable 'sea' set but not used [-Wunused-but-set-variable]

     int16_t sea = top;

             ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'void MCUFRIEND_kbv::begin(uint16_t)':

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:1684:30: warning: unused variable 'ST7789_regValues_arcain6' [-Wunused-variable]

         static const uint8_t ST7789_regValues_arcain6[] PROGMEM = {

                              ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:2595:30: warning: unused variable 'ILI9341_regValues_ada' [-Wunused-variable]

         static const uint8_t ILI9341_regValues_ada[] PROGMEM = {        // Adafruit_TFTLCD only works with EXTC=0

                              ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:2675:30: warning: unused variable 'ILI9481_CPT29_regValues' [-Wunused-variable]

         static const uint8_t ILI9481_CPT29_regValues[] PROGMEM = {    // 320x430

                              ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:2687:30: warning: unused variable 'ILI9481_PVI35_regValues' [-Wunused-variable]
[... 39 lines cut ...]
 static uint16_t color565_to_555(uint16_t color) {

                 ^

C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:68:17: warning: 'uint16_t color555_to_565(uint16_t)' defined but not used [-Wunused-function]

 static uint16_t color555_to_565(uint16_t color) {

                 ^

cc1plus.exe: some warnings being treated as errors

Using library SPI at version 1.0 in folder: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware\espressif\esp32\libraries\SPI 

Using library Adafruit_GFX_Library at version 1.10.3 in folder: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\Adafruit_GFX_Library 

Using library MCUFRIEND_kbv at version 2.9.9-Release in folder: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\MCUFRIEND_kbv 

Using library Adafruit_BusIO at version 1.4.1 in folder: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries\Adafruit_BusIO 

Using library Wire at version 1.0.1 in folder: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware\espressif\esp32\libraries\Wire 

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

11Naweed11, 2020-12-01:
here are the messages of a working sketch of (GetChipID) and it does give the chip id in the serial monitor.

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Ahmad Naweed Rustami\AppData\Local\Arduino15\packages -hardware C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Ahmad Naweed Rustami\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=warn -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\AHMADN~1\AppData\Local\Temp\arduino_build_578325 -warnings=all -build-cache C:\Users\AHMADN~1\AppData\Local\Temp\arduino_cache_477071 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32\examples\ChipID\GetChipID\GetChipID.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Ahmad Naweed Rustami\AppData\Local\Arduino15\packages -hardware C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Ahmad Naweed Rustami\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Ahmad Naweed Rustami\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=115200,DebugLevel=warn -vid-pid=10C4_EA60 -ide-version=10813 -build-path C:\Users\AHMADN~1\AppData\Local\Temp\arduino_build_578325 -warnings=all -build-cache C:\Users\AHMADN~1\AppData\Local\Temp\arduino_cache_477071 -prefs=build.warn_data_percentage=75 -verbose C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware\espressif\esp32\libraries\ESP32\examples\ChipID\GetChipID\GetChipID.ino
Using board 'esp32doit-devkit-v1' from platform in folder: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\Ahmad Naweed Rustami\Documents\Arduino\hardware\espressif\esp32
cmd /c if exist "C:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32\\libraries\\ESP32\\examples\\ChipID\\GetChipID\\partitions.csv" copy /y "C:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32\\libraries\\ESP32\\examples\\ChipID\\GetChipID\\partitions.csv" "C:\\Users\\AHMADN~1\\AppData\\Local\\Temp\\arduino_build_578325\\partitions.csv"
cmd /c if not exist "C:\\Users\\AHMADN~1\\AppData\\Local\\Temp\\arduino_build_578325\\partitions.csv" copy "C:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32\\tools\\partitions\\default.csv" "C:\\Users\\AHMADN~1\\AppData\\Local\\Temp\\arduino_build_578325\\partitions.csv"
        1 file(s) copied.
Detecting libraries used...
"C:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/config" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/app_trace" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/app_update" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/asio" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bootloader_support" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/bt" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/coap" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/console" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/driver" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/efuse" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp-tls" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp32" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_event" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_http_client" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_http_server" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_https_ota" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/sdk/include/esp_https_server" "-IC:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espr [... cut ...] 1\\AppData\\Local\\Temp\\arduino_build_578325/GetChipID.ino.partitions.bin"
"C:\\Users\\Ahmad Naweed Rustami\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-size" -A "C:\\Users\\AHMADN~1\\AppData\\Local\\Temp\\arduino_build_578325/GetChipID.ino.elf"
Sketch uses 207832 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13476 bytes (4%) of dynamic memory, leaving 314204 bytes for local variables. Maximum is 327680 bytes.

prenticedavid, 2020-12-01:
File->Preferences: verbose compile
**File->Preferences: compiler warnings : Default**

Copy-paste the "Build sketch" command line
Copy-paste the Error line(s)

If you paste 1000 lines again I will not reply to you.

11Naweed11, 2020-12-02:
well I managed to fix that problem thanks a lot. but now after uploading the graphictest_kbv there is nothing on the display and I continuously get this message in the serial monitor:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

11Naweed11, 2020-12-02:
and there is one more thing if the pins D0 and D1 of the display are connected to the D12 and D13 of the esp32 the sketch fails to upload, so to upload the sketch I need to disconnect them. can I change these 2 pins to be connected to some other pins?

11Naweed11, 2020-12-02:
one more thing the new 2.9.9 version of the library didn't work with any of the boards, I reverted back to 2.9.8.

prenticedavid, 2020-12-02:
Please be more specific.   I seldom use a Parallel Shield with the ESP32.

I would expect v2.9.9-Release to work out of the box.   If you can describe your hardware I can see if I can replicate your problems.
The easiest way is to post links to the hardware items e.g. Ebay sale page(s)

David.

11Naweed11, 2020-12-03:
OK, I am sorry if this is too long.
recently I started working on project to use esp32 as a graphics driver such that it receives data through a serial communication protocols from external sources and displays it on the screen. Because the microcontrollers used in Arduino boards don't have the horsepower to drive the displays at a reasonable frame rates and do some other tasks at the same time even if you use SPI displays but esp32 does. I tried to run 3 different displays with esp32 using your library (v 2.9.8) but were unsuccessful. one of them is this one:

2.4 inch size, LGDP4532 driver IC, 240x320 res, 8-bit parallel interface
[image]

this display works fine with all Arduino boards using your library v2.9.8. somehow I couldn't get the new version to work. Also using v2.9.8 with Arduino Mega and defining "use mega 8 bit parallel protosheild" does increase the speed but messes with the touch. the values keep jumping to some negative values.

prenticedavid, 2020-12-03:
Thanks for the links.    I don't have an LGDP4532 display.  But if it works on a Uno with v2.9.9-Release it should work on ESP32 too.

I suggest that you report the ID that is shown on the Serial Terminal with Uno and with ESP32.
If it does not read the ID correctly,  force it with `tft.begin(0x4532);`
Likewise copy-paste the Serial Terminal when running the Touch Calibration sketch on a Uno.

As it happens,   I plugged a 320x480 ST7795S Shield into my ESP32 (TTGO R32) today.
Regular sketches ran fine.   But Adafruit Touchscreen.h does not work.   Touchscreen_kbv.h does work.

Seriously.    Most sketches should run pretty well on Uno, Zero, Due, STM32, ESP32,
Uno Shields are crap on Mega2560 and Leo due to the random pin mapping.

So I would concentrate on designing your project.   e.g. updating minimal areas of the screen and _only_ when changed.

Yes,   ESP32, ESP8266 work far better with SPI screens.   And you have spare pins left for other electronics. 

David.

11Naweed11, 2020-12-03:
I get a 0x4532 on the serial monitor no matter what board I use and unfortunately the display doesn't work with v2.9.9 with any board. and thanks for the advice about the touch library I will do it and let you know the results.

11Naweed11, 2020-12-13:
hello, I finally got it working by switching to another display with 0x9340 driver ic. I couldn't get the 0x4532 working no matter what.

[image]

but still there is a small problem. if the display's D0 and D1 are connected to esp32, the sketch fails to upload so I have to disconnect them each time when I upload the sketch and connect them back and reset the mcu.
I am not so satisfied with the results though, I mean I didn't get much improvements by jumping from 2KB to 512KB of ram, 16Mhz to 240Mhz of cpu frequency, 8bit to 32bit architecture. Arduino uno finishes the test in 5.5 secs and esp32 in 2.5 secs. is there anyway to improve this?

prenticedavid, 2020-12-13:
I don't have an LGDP4532 display. But if it works on a Uno with v2.9.9-Release it should work on ESP32 too.
Note that the "older" controllers like LGDP4532 might find ESP32 to be too fast.   You can always change the ESP32 speed to 240MHz,  160MHz or 80MHz.   Does 80MHz make a difference?

Yes,  I would expect ILI9340 to behave much like the ILI9341.   i.e. fine on the ESP32.

ESP32 does not like external electronics on GPIO12.   But that is what TTGO R32 headers use for LCD_D0.   You can connect a pull-down on LCD_D0 during program upload. (I think)

Regarding speed.   The Adafruit Tests take 0.76 seconds for an overclocked Xmega.  (62MHz)
There is little advantage in running at top speed.    Not all controllers accept this speed.

Oh,   this library was originally written for Uno and expected a to support a handful of controllers.    Over the years it has added many controllers and many target MCUs.

David.

11Naweed11, 2020-12-13:
Thanks David. there is something that makes me curious. I have an oscilloscope DSO138, it's an open source device so you can find any information about it on the internet even the source code. this thing uses STM32F103C8 (ARM Cortex M3) muc and a display with ili9341, 240x320 res, 37 pin fpc connection. the refresh rates of the graphs on the display are above 30fps.
So whats the secret here? the hardware obviously is not so high end. how do they get that fast refresh rates?

11Naweed11, 2020-12-13:
I changed the cpu frequency to 80Mhz, the test took 7.45 secs but the F_CPU still shows me 240Mhz on the display, while getCpuFrequencyMhz() shows 80Mhz.

prenticedavid, 2020-12-13:
I get F_CPU=80MHz in the Adafruit_Results page.

The display speed depends on F_CPU for the target.   I introduce delays on anything above 64MHz.

There is no point in worrying about tests that take 0.76 sec or 2.0 secs.

If you want your application to run faster,   sit down and design it properly.

David.

11Naweed11, 2020-12-13:
sure! I am not the kind of person that gives up quickly.

SelfForm, 2020-12-22:
Hi 11Naweed11, could you please post a picture or a map of the wiring from your display to the ESP 32? Regards Alfred

closed 2021-08-24 by prenticedavid

ealsons, 2025-01-20:
chup loray naveed