prenticedavid/MCUFRIEND_kbv #67: Arduino mega2560 pin change

TJWi, 2018-10-15:
Hi, i need a pin change because i damage the pin on my Mega2560

#define LCD_RST A12
#define LCD_CS A11
#define LCD_RS A10
#define LCD_WR A9
#define LCD_RD A8

#define LCD_D0 22
#define LCD_D1 23
#define LCD_D2 24
#define LCD_D3 25
#define LCD_D4 26
#define LCD_D5 27
#define LCD_D6 28
#define LCD_D7 29

I've tried tried to search for solution in your library, but i didn't find any suitable solution that would help.

prenticedavid, 2018-10-15:
There is no ready-made shield with that pinout.

If you have hand-wired with that custom arrangement,  verify the wiring via LCD_ID_readreg sketch.

David.

TJWi, 2018-10-17:
it is a custom, since i damage the pin on the mega board, i have to bridge and use other pins instead

TJWi, 2018-10-17:
anyway, here is the result

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 95 95	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 3F 3F 3F 3F 3F	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 01 01	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03	GAMMA-P
reg(0x00E1) 06 06 06 06 06 06 06 06 06 06 06 06 06 06 06 06	GAMMA-N
reg(0x00EF) 14 14 14 14 14 14	ILI9327
reg(0x00F2) 08 08 08 08 08 08 08 08 08 08 08 08	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

TJWi, 2018-10-17:
ok nvm I got it working for some reason, something came up in my mind and i happen to got it working for me, here is what i have changed with the help of this schematic:

http://yourduino.com/docs/MegaPinOut.png

#elif defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1280__)       //regular UNO shield on MEGA2560
#define RD_PORT PORTK   //PortK
#define RD_PIN  0              //pin 0 of Port K
#define WR_PORT PORTK
#define WR_PIN  1              //pin 1 of Port K
#define CD_PORT PORTK
#define CD_PIN  2              //pin 2 of Port K
#define CS_PORT PORTK
#define CS_PIN  3              //pin 3 of Port K
#define RESET_PORT PORTK
#define RESET_PIN  4         //pin 4 of Port K

#define write_8(x)   { PORTA = x;} //all pins of Port A

#define read_8()      ( PINA )
#define setWriteDir() { DDRA = 0xFF; }
#define setReadDir()  { DDRA = 0x00; }
#define write8(x)     { write_8(x); WR_ACTIVE; WR_STROBE; } // HX8357-D is slower
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { RD_STROBE; dst = read_8(); RD_IDLE; RD_STROBE; dst = (dst<<8) | read_8(); RD_IDLE; }

#define PIN_LOW(p, b)        (p) &= ~(1<<(b))
#define PIN_HIGH(p, b)       (p) |= (1<<(b))
#define PIN_OUTPUT(p, b)     *(&p-1) |= (1<<(b))

I am a beginner as a programmer and i wish there is a simpler way to change pin but i understood what you did for pin assignment after studying it, so it is justify able.

The original port that you type for mega2560 is one hell of a bit manipulation, I can never think of coding something like that so great job! I appreciate this library a lot more now!

prenticedavid, 2018-10-17:
That looks exactly like my USE_MEGA_8BIT_PROTOSHIELD from mcufriend_special.h
i.e. control pins on A0-A4, data bus moved to D22-D29

It is much easier to just quote your LCD_ID_readreg defines verified by proper report from LCD_ID_readreg.

closed 2018-10-18 by TJWi

TJWi, 2018-10-18:
yes, before i tried ask for your help, i actually type my own code along those lines to tried and get it to work which it did not. i later found one this set of code above when i was searching around the special but it still didn't work until i change the port to K which is how i've wired it.