prenticedavid/MCUFRIEND_kbv #69: ATmega64/128 & TFT

kluv58, 2018-11-29:
Dear David.
Today I decided to supplement your library with a new controller: ATmega64 / 128. This is a worthy alternative to the ATmega2560, but with a smaller number of pins, which makes it much easier to work with it.
To implement your idea, the following block was added to your library:

//################################### UNO SHIELD on  AVR_ATmega128/64#############

#elif defined(__AVR_ATmega128__) || defined(__AVR_ATmega64__) 
#define RD_PORT PORTF 
#define RD_PIN  0
#define WR_PORT PORTF
#define WR_PIN  1
#define CD_PORT PORTF
#define CD_PIN  2
#define CS_PORT PORTF
#define CS_PIN   3
#define RESET_PORT PORTF
#define RESET_PIN   4

#define write_8(x)   {PORTC = x;}
#define read_8()      (PINC)
#define setWriteDir() { DDRC = 255;  }
#define setReadDir()  { DDRC = 0;  }

#define write8(x)     { write_8(x); WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

#define PIN_LOW(p, b)        (p) &= ~(1<<(b))
#define PIN_HIGH(p, b)       (p) |= (1<<(b))
#define PIN_OUTPUT(p, b)     *(&p-1) |= (1<<(b))
---------------------------------------------------------------------------------------------------------------
display connection:
#define LCD_RST  A4
#define LCD_CS    A3
#define LCD_RS    A2
#define LCD_WR   A1
#define LCD_RD    A0

#define LCD_D0     28 //PC0
#define LCD_D1     29 //PC1
#define LCD_D2     30 //PC2
#define LCD_D3     31 //PC3
#define LCD_D4     32 //PC4
#define LCD_D5     33 //PC5
#define LCD_D6     34 //PC6
#define LCD_D7     35 //PC6

Here is the result of the work:

[image]

kluv58, 2018-11-29:
In the source file everything was so
#elif defined(__AVR_ATmega128__) || defined(__AVR_ATmega64__) 
Apparently an error occurred while copying.
I know only one ready board based on ATmega128, which is made by several companies.
http://archiwumalle.pl/moduł_uruchomieniowy_atmega128a_jtag_isp_128-8_4887939299.html
but this is not the arduino form factor

prenticedavid, 2018-11-29:
The missing underscore feature is part of this GitHub message system.    You should use the  <> Code icon when quoting any code.

Your mega128 board does not have Aduino headers.    So your "special" is not appropriate for the "shield.h" file.

However you mega128 should work faster than a Uno and MUCH faster than a Mega.

David.

closed 2018-12-04 by kluv58