prenticedavid/MCUFRIEND_kbv #71: Data pins issue

sh-raza, 2018-12-07:
Hello Guys

I am using ILI9325  240x320  ID=0x9325 using with Arduino Mega 2560 R3 working fine with it.! Now I want to know that which pins of Arduino Mega 2560 R3 can be used for LCD_D2 and LCD_D3.

LCD Image: http://www.yggenyk.dk/wiki/images/thumb/Mcufriend_3.5%27_TFT_LCD_for_arduino_uno.jpg/609px-Mcufriend_3.5%27_TFT_LCD_for_arduino_uno.jpg

prenticedavid, 2018-12-08:
God invented shields.   You plug the shield into your Mega2560.    All the male pins are happy because they have female sockets to mate with.   e.g. LCD_D2 mates with digital#2 PE4)

I do NOT believe that you have a 3.5 inch shield with an ILI9325 controller.
Most 3.5 shields are 320x480.

David.

sh-raza, 2018-12-08:
Thanks for your response!

prenticedavid, 2018-12-08:
Please explain what you want to do.   It might be better to ask on Arduino Displays Forum.

I suspect that you want to use digital#2 and digital#3 for interrupts.  Just say.   

You can use Pin Change interrupts on any digital pin of a modern AVR.
However the External INT0, INT1, INT2, INT3, INT4, INT5, INT6, INT7 need specific pins.

David.

prenticedavid, 2019-01-31:
Please explain what you want to do.   

If you have solved your problem,  please say HOW you solved it before Closing the Issue.

David.

sh-raza, 2019-02-01:
Actually I want to use the M6E RFID reader and LCD both on the same board but issue is that I want to know which pins can be used as alternative for LCD_D2 and LCD_D3.

sh-raza, 2019-02-01:
You should have to believe that you have a 2.8-inch shield with an ILI9325 controller which is I bought from Chinese electronic market/store

prenticedavid, 2019-02-01:
My apologies.   Your original message showed a JPEG of a 3.5 inch pcb.    I can view the original message before you edited it.
There is nothing wrong with showing a 3.5 pcb when you own a 2.8 inch pcb.    Just say so e.g. say your pcb is smaller.
It is always wise to clarify any points of confusion.    You can edit messages, correct typos, correct links etc.

I can move any TFT pin that you want.    You have to test your wiring with LCD_ID_readreg and tell me.   I will write the SPECIAL for you.
But life is much easier if you stick with the shield wiring.    A mated shield is mechanically and electrically robust.  

David.

sh-raza, 2019-02-04:
Thanks, David for your kind support. I apologise, for not clearly asking the question. Unfortunately currently not working on this project because my supervisor has changed the scope.

closed 2021-08-24 by prenticedavid