prenticedavid/MCUFRIEND_kbv #74: STM32F103C bluepill with Hx8352A

cchandel, 2019-01-09:
I'm trying to get a bluepill board to work with a opensmart 3.2" hx 8352A LCD (as printed on the LCD).

    When running LCD_ID_readreg.ino I get the following :-

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 52 52 ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00 Manufacturer ID
reg(0x0009) 8F 8F 8F 8F 8F Status Register
reg(0x000A) 52 52 Get Power Mode
reg(0x000C) 01 01 Get Pixel Format
reg(0x0061) FF FF RDID1 HX8347-G
reg(0x0062) 1F 1F RDID2 HX8347-G
reg(0x0063) FF FF RDID3 HX8347-G
reg(0x0064) 1F 1F RDID1 HX8347-A
reg(0x0065) FF FF RDID2 HX8347-A
reg(0x0066) 1F 1F RDID3 HX8347-A
reg(0x0067) FF FF RDID Himax HX8347-A
reg(0x0070) 30 30 Panel Himax HX8347-A
reg(0x00A1) 08 08 08 08 08 RD_DDB SSD1963
reg(0x00B0) 00 00 RGB Interface Signal Control
reg(0x00B4) 5D 5D Inversion Control
reg(0x00B6) 64 64 64 64 64 Display Control
reg(0x00B7) 68 68 Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00 ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00 Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00 GAMMA
reg(0x00CC) 00 00 Panel Control
reg(0x00D0) 00 00 00 Power Control
reg(0x00D2) 00 00 00 00 00 NVM Read
reg(0x00D3) 00 00 00 00 ILI9341, ILI9488
reg(0x00D4) 00 00 00 00 Novatek ID
reg(0x00DA) 00 00 RDID1
reg(0x00DB) 00 00 RDID2
reg(0x00DC) 00 00 RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 GAMMA-N
reg(0x00EF) 00 00 00 00 00 00 ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00 Adjust Control 2
reg(0x00F6) 00 00 00 00 Interface Control

    However, I see nothing on the LCD (the backlight is on) when running graphictest_kbv.ino. The ID returned there is D3D3.

prenticedavid, 2019-01-09:
BluePill has an "official" wiring.    Connect to your Open-Smart pins correctly and it will read the ID ok.
e.g. LCD_RD=PB0, LCD_WR=PB6, LCD_RS=PB7, LCD_CS=PB8, LCD_RST=PB9
e.g. LCD_D0 - LCD_D7 = PA0 - PA7

You must enable SUPPORT_8352 in MCUFRIEND_kbv.cpp for tft.begin(0x5252) to work.

David.

prenticedavid, 2019-01-11:
You have to diagnose the Touch pins with a Uno.
Or just measure the resistance with a DMM.

It is not possible to diagnose Touch pins on an ARM.   But the TouchScreen library will work if you use the correct constructor.

David.

prenticedavid, 2019-01-29:
Are you using the ST Core from STMicroelectronics or Roger's weird Maple Core?

Most BluePills have the wrong resistors on the USB lines.
Roger's Serial USB needs a pause to initialise properly.

I would expect the Calibration sketch to work.

David.

closed 2021-08-24 by prenticedavid