prenticedavid/MCUFRIEND_kbv #75: Teensy++ 2.0 support

tysonlt, 2019-01-27:
Hello! Your library works well with Teensy 3x, but I was wondering if it could support T++ 2.0. I have an 0x8230, and the screen fills (with blue, but hey) during the LCD_ID_readreg test.

When I compile graphicstest with T++ I get:

`
/Users/Tyson/Documents/Arduino/libraries/Adafruit_GFX/Adafruit_GFX.cpp: In member function 'boolean Adafruit_GFX_Button::contains(int16_t, int16_t)':
/Users/Tyson/Documents/Arduino/libraries/Adafruit_GFX/Adafruit_GFX.cpp:602:11: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if ((x < (_x - _w/2)) || (x > (_x + _w/2))) return false;
           ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_GFX/Adafruit_GFX.cpp:602:32: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if ((x < (_x - _w/2)) || (x > (_x + _w/2))) return false;
                                ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_GFX/Adafruit_GFX.cpp:603:11: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if ((y < (_y - _h)) || (y > (_y + _h/2))) return false;
           ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_GFX/Adafruit_GFX.cpp:603:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
    if ((y < (_y - _h)) || (y > (_y + _h/2))) return false;
                              ^
In file included from /Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp:19:0:
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/pin_magic.h:166:28: warning: backslash and newline separated by space
   #define write8inline(d) {\                        

                            ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/pin_magic.h:172:30: warning: backslash and newline separated by space
    PORTG |= (d & 0x10) << 1; \ 

                              ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/pin_magic.h:287:33: warning: backslash and newline separated by space
   #define read8inline(result) { \    

                                 ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/pin_magic.h:355:3: error: #error "Board type unsupported / not recognized"
  #error "Board type unsupported / not recognized"

   ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp: In member function 'void Adafruit_TFTLCD::begin(uint16_t)':
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp:343:14: warning: unused variable 'a' [-Wunused-variable]
     uint16_t a, d;

              ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp:343:17: warning: unused variable 'd' [-Wunused-variable]
     uint16_t a, d;

                 ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp: In member function 'void Adafruit_TFTLCD::write8(uint8_t)':
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp:1031:21: error: 'write8inline' was not declared in this scope
   write8inline(value);

                     ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp: In member function 'uint8_t Adafruit_TFTLCD::read8fn()':
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp:1038:21: error: 'read8inline' was not declared in this scope
   read8inline(result);

                     ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp: In member function 'void Adafruit_TFTLCD::setWriteDir()':
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp:1045:21: error: 'setWriteDirInline' was not declared in this scope
   setWriteDirInline();

                     ^
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp: In member function 'void Adafruit_TFTLCD::setReadDir()':
/Users/Tyson/Documents/Arduino/libraries/Adafruit_TFTLCD/Adafruit_TFTLCD.cpp:1051:20: error: 'setReadDirInline' was not declared in this scope
   setReadDirInline();

                    ^
Error compiling for board Teensy++ 2.0.
`

I tried hacking the include files to support the chipset, but I wasn't sure what to do!!!

Thank you

prenticedavid, 2019-01-27:
Why are you compiling Adafruit_TFTLCD.cpp ?
Your question is about MCUFRIEND_kbv.

Teensy3.x have an "official" baseboard from SparkFun with Arduino headers.
If there was a commercially produced baseboard for Teensy++ 2.0 I would put the support in "utility/mcufriend_shield.h"

I do not own a Teensy++ 2.0 but it appears to use a AT90USB1286.   I have to rely on owner feedback.
If you post the defines from your readreg sketch with the correct sketch output,   I can add support as a SPECIAL.

David.

tysonlt, 2019-01-28:
You sir, are a scholar and a gentleman. It worked first time!!! I thank you many times, I am a very happy man.

I had to change the include to check for __AVR_AT90USB1286__ instead of __AVR_ATmega1286__.

For completeness, here is the result of LCD_ID_readnew:

`reg(0x0000) 82 30	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 84 10	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
`

closed 2019-01-28 by tysonlt