prenticedavid/MCUFRIEND_kbv #78: ...any chance ili9335 to work on esp32 R32 uno style board.?

SurenBono, 2019-02-14:


SurenBono, 2019-02-14:
https://github.com/Bodmer/TFT_eSPI/issues/307

prenticedavid, 2019-02-14:
I seem to have replied on Bodmer's Github and not mine.
MCUFRIEND_kbv supports over 30 different controllers.   Bodmer only supports a few specific MIPI controllers.   ILI9335 is an "ILI9320-style" controller.

David.

SurenBono, 2019-02-18:
..yeah sorry bro...i got confused..... is there ANY chance that the display ILI9335 tft shield that worked on UNO to work on similar UNO syle esp32 r32 board...... logically it cant run on wemos d1 r1 or r2...but it should work on esp32 r32 if i know how to extract the driver from MCUFRIEND_KBV and imbue it on Bodmers eSPI_TFT... but im not expert ..thats why im here...

prenticedavid, 2019-02-18:
I support ILI9335.    If you want to use Bodmer's TFT_eSPI library,   I suggest that you buy a shield with ILI9341.

Quite honestly,   an SPI ILI9341 with XPT2046 Touch controller is far more appropriate.
It works with TFT_eSPI on ESP32, ESP8266 without wasting GPIO pins.

David.

SurenBono, 2019-02-19:
Thanks for the info David...

closed 2019-02-24 by prenticedavid