prenticedavid/MCUFRIEND_kbv #82: Wiring problem

verbessern, 2019-03-03:
Hi community, i have Wemos Lolin32, and a TFT display recognized by id 0x4532 (mcufriend 2.4 inch TFT LCD shield). I have connected just RS to 15, CS to 33 and RST to 32, but the display is very dark. I found the wiring to be this: 

LCD_RD 2 //LED
LCD_WR 4
LCD_RS 15 //hard-wired to A2 (GPIO35)
LCD_CS 33 //hard-wired to A3 (GPIO34)
LCD_RST 32 //hard-wired to A4 (GPIO36)
LCD_D0 12
LCD_D1 13
LCD_D2 26
LCD_D3 25
LCD_D4 17
LCD_D5 16
LCD_D6 27
LCD_D7 14

my question is, do GPIO33 and GPIO34 have to be connected both to LCD_CS? Same for RST--32--36, RS--15--35. I'm worried that connecting two pins (15-35 for example) may kill my hardware (like short circuit it), if i don't understand the comments correctly, even that is directly written. I'm new in hardware. Is it 'ok' to connect all this pins together? Please help.

prenticedavid, 2019-03-03:
No,  you don't need to connect GPIO35, GPIO34, GPIO36.

It is simply that these pins are already wired to A2, A3, A4 on the TTGO board.

Personally,   I think it is fairly pointless to use a parallel display with ESP32.   Not possible at all with ESP8266.
Buy an SPI ILI9341 display from Ebay.

What do you mean by dark?
All the programs should work correctly (except for Band Scroll)

Please post a link to your display that shows a photo of the pcb.

David.

verbessern, 2019-03-03:
Thank you for the answer. Here are some photos. The display is very dark, and the graphics are hardly visible. I connected wrongly the first time, and the display was only very bright, far more then after i connected it correctly. Even in a dark room, its barely visible.
[image]
[image]
[image]

prenticedavid, 2019-03-03:
The original Mcufriend shields had an AMS1117-3.3 LDO regulator connected to the 5V pin.
The backlight LED was fed from the AMS1117 output e.g. 50mA or 200mA for bigger screens.
The 3.3V pin powers the 74HC245 buffers and TFT panel e.g. 10mA or so.

The current pcb (like your photo) has no LDO.
It powers the LED from the 3.3V pin.    Which is **WRONG** for Uno or Mega.
The 5V pin is unused.

Check that you have 3.3V on the 3.3V pin. 
Measure current drawn by 3.3V pin.
Check that there is no current drawn by 5V pin.

David.

verbessern, 2019-03-04:
I always had 3.3v and 5v connected. Now i have only 3.3v and it works. Maybe my wires were somehow not connected properly even that they are new. It is all very strange, because I have tested many times. Now everything works with only 3.3v connected but the button of the display stopped to work, it was working before, and was restarting the board when clicked, but i will start new issue if i don't manage to make it work. I have also changed GPIO 12 to 0, because Wemos Lolin32 can't boot when 12 is used (https://github.com/espressif/esptool/issues/305). Thank you for your help.

prenticedavid, 2019-03-04:
I am horrified by the cavalier attitude of mcufriend.com
The displays are sold as "Uno Shields".   So they should use LV245 and not HC245 buffers to be input-tolerant of 5V GPIO.   

And they should use AMS1117-3.3 regulators for the backlight.
The empty Flash memory footprint can only be used with 3.3V GPIO.

It would cost them nothing to design their universal pcbs properly.   e.g. solder-bridges to configure different series resistors to limit the backlight current.

Please check whether your 5V pin is connected with any copper traces to the display.

David.

verbessern, 2019-03-04:
It does not have any connection from the both sides of the PCB.

prenticedavid, 2019-03-04:
In which case it would not matter whether the 5V pin is wired or not.

I have never seen a LGDP4532 shield.    I rely on user feedback.
I was asked for LGDP4532 support four years ago.   I had assumed that batch had all been sold by now.
Your "modern pcb" implies that there is a new batch of LGDP4532 screens on the Ebay market.

David.

verbessern, 2019-03-04:
I have this sins a year, but never had time to test it. I really hope that i will drive all its functionalities. There is SD card reader, touch, looks very nice to my basic hardware skills. Its a little bright, I may try to reduce the 3.3v at some point to see is there effect.

prenticedavid, 2019-03-04:
Measure the current taken by the 3.3V pin.

Some mcufriend shields take far too much current for the backlight.
Does the screen feel warm?
Most 3.3V Arduinos can deliver 300mA to the 3.3V pin.
But a 5V Uno or Mega has a wimpy 3.3V regulator that is limited to 50mA

I suggest that you identify the trace to the backlight.    Cut the trace and insert a series resistor e.g. 22R.

verbessern, 2019-03-04:
Its 75mA during the tests. I will delay the trace cutting if i don't get use to the brightness. Thank you for the advice. I'm trying the touch now, automatic detection does not seem to work in my case, but I'm still investigating, found your posts in one other forum (https://forum.arduino.cc/index.php?topic=496950.0).

prenticedavid, 2019-03-04:
75mA is clearly far too much for Uno, Mega regulators.
It is not unreasonable for LEDs.    They are unlikely to fail.   Or make the screen warm.   Or too bright.
40-50mA might be a more convenient brightness.

The 3.5 inch RM68140 mcufriend shield that takes 400mA is seriously **WRONG**.
The 3.2 inch ILI9481 gets hot.

David.

closed 2019-03-04 by verbessern