prenticedavid/MCUFRIEND_kbv #83: Touch screen on 2.4" TFT

verbessern, 2019-03-04:
Hi community, I'm trying to read the touch screen values of 2.4" TFT display with no success. The display visual tests are running successfully. I have written a small script based on 'diagnose_touchpins" example, because I use Wemos Lolin32 board and the pins are different. Here is the code:

```
// connected pins from 2.4" MCUFRIEND TFT LCD display to the Wemos Lolin32
// the values are from mcufriend_shield.h for ESP32
#define LCD_RD  2   
#define LCD_WR  4
#define LCD_RS  15
#define LCD_CS  33
#define LCD_RST 32

#define LCD_D0 0   // changed from 12 to 0, because Lolin32 does not boot
#define LCD_D1 13
#define LCD_D2 26
#define LCD_D3 25
#define LCD_D4 17
#define LCD_D5 16
#define LCD_D6 27
#define LCD_D7 14

// the two sets of pins
int alpha[] =  { LCD_RD, LCD_WR, LCD_RS, LCD_CS, LCD_RST };
int beta[] =  { LCD_D0, LCD_D1, LCD_D2, LCD_D3, LCD_D4, LCD_D5, LCD_D6, LCD_D7 };

// helper
#define DIM(a) (sizeof(a) / sizeof(*a))

void setup()
{
  const int len = DIM(alpha) * DIM(beta);
  int i, j, value, Apins[len], Dpins[len], Values[len], found = 0;

  Serial.begin(9600);
  Serial.println("START");

  for (i = 0; i < DIM(alpha); i++)
  {
    Serial.printf("alpha %d\n", alpha[i]);
    pinMode(alpha[i], INPUT_PULLUP);
  }

  for (i = 0; i < DIM(beta); i++)
  {
    Serial.printf("Beta %d\n", beta[i]);
    pinMode(beta[i], INPUT_PULLUP);
  }

  for (i = 0; i < DIM(alpha); i++)
  {
    for (j = 0; j < DIM(beta); j++)
    {
      // set low
      pinMode(beta[j], OUTPUT);
      digitalWrite(beta[j], LOW);
      delay(10);

      // get
      value = analogRead(alpha[i]);   // ignore first reading
      value = analogRead(alpha[i]);

      Serial.printf("Value %d:%d = %d\n", alpha[i], beta[j], value);

      // restore
      pinMode(beta[j], INPUT_PULLUP);
    }
  }
}

void loop()
{

}
```
and that is the results, but many values are different at every execution of the code, where is the touch connected?:
```
START
alpha 2
alpha 4
alpha 15
alpha 33
alpha 32
Beta 0
Beta 13
Beta 26
Beta 25
Beta 17
Beta 16
Beta 27
Beta 14
Value 2:0 = 2967
Value 2:13 = 2883
Value 2:26 = 0
Value 2:25 = 3095
Value 2:17 = 3287
Value 2:16 = 3157
Value 2:27 = 2331
Value 2:14 = 1581
Value 4:0 = 4095
Value 4:13 = 4095
Value 4:26 = 4095
Value 4:25 = 4095
Value 4:17 = 4095
Value 4:16 = 4095
Value 4:27 = 0
Value 4:14 = 4095
Value 15:0 = 4095
Value 15:13 = 4095
Value 15:26 = 4095
Value 15:25 = 4095
Value 15:17 = 4095
Value 15:16 = 4095
Value 15:27 = 4095
Value 15:14 = 0
Value 33:0 = 3365
Value 33:13 = 0
Value 33:26 = 0
Value 33:25 = 0
Value 33:17 = 0
Value 33:16 = 3251
Value 33:27 = 3207
Value 33:14 = 3295
Value 32:0 = 3195
Value 32:13 = 3343
Value 32:26 = 3407
Value 32:25 = 3346
Value 32:17 = 3383
Value 32:16 = 3327
Value 32:27 = 3367
Value 32:14 = 3367

START
alpha 2
alpha 4
alpha 15
alpha 33
alpha 32
Beta 0
Beta 13
Beta 26
Beta 25
Beta 17
Beta 16
Beta 27
Beta 14
Value 2:0 = 3085
Value 2:13 = 0
Value 2:26 = 616
Value 2:25 = 0
Value 2:17 = 0
Value 2:16 = 2987
Value 2:27 = 3306
Value 2:14 = 3407
Value 4:0 = 4095
Value 4:13 = 4095
Value 4:26 = 4095
Value 4:25 = 4095
Value 4:17 = 4095
Value 4:16 = 4095
Value 4:27 = 0
Value 4:14 = 4095
Value 15:0 = 4095
Value 15:13 = 4095
Value 15:26 = 3755
Value 15:25 = 3799
Value 15:17 = 3771
Value 15:16 = 3787
Value 15:27 = 4095
Value 15:14 = 0
Value 33:0 = 3174
Value 33:13 = 2559
Value 33:26 = 2559
Value 33:25 = 2559
Value 33:17 = 1264
Value 33:16 = 1446
Value 33:27 = 1489
Value 33:14 = 1449
Value 32:0 = 3179
Value 32:13 = 0
Value 32:26 = 0
Value 32:25 = 0
Value 32:17 = 0
Value 32:16 = 2650
Value 32:27 = 2655
Value 32:14 = 2559

```
Please help.

prenticedavid, 2019-03-04:
Go on.   You could either plug the shield into a Uno and run the whole calibration sketch on the Uno.

Or just measure resistance with your DMM.     This will determine the touch pins.
Then run the calibration sketch on the ESP32 with the **known** XM, YP, XP, YM pins.

Make sure that XM, YP pins can be used for both Analog and Digital.    XP, YM only need to be Digital.

David.

verbessern, 2019-03-04:
I don't have Uno. I never measured resistance, but it seems one wire from the DMM goes to ground other to the pin. Do you mean to make LOW one of the pins and measure the others, till i find which one changes when some pin is LOW?

prenticedavid, 2019-03-04:
Most shields use A1, A2, D6, D7 for the touch pins.

Unplug the shield.   Measure the resistance between LCD_WR (A1) and LCD_D6, LCD_D7, LCD_D0, LCD_D1.   i.e. one DMM probe on A1.  place the other probe on D6, ....
Then do the same with LCD_RS (A2).

You should find that you get about 300 ohms for one pair.   And about 500 ohms for the other pair.
Make notes of the results.

David.

verbessern, 2019-03-04:
Awesome! I found out that putting GPIO 14 LOW i can measure Y touch value on the display (by my orientation)! Great! I have to find now X and Pressure :).

prenticedavid, 2019-03-04:
What are the resistance values?

verbessern, 2019-03-04:
Yes that is true for the pins: WR, RS, D6, D7. But I'm getting very twisted space. Its not orthogonal and I will have to invert by a matrix. Is that normal? I see the examples of the calibration do linear interpolation.

verbessern, 2019-03-04:
The resistance i did not measure, because the wires are so many to reconnect, and the pins are already known.

prenticedavid, 2019-03-04:
Go on.   Just say what the resistances are.   
e.g. 345 ohms between LCD_WR and LCD_D7
e.g. 567 ohms between LCD_RS and LCD_D6

verbessern, 2019-03-04:
Also Lolin32 i think have 12 bit ADC and analogRead returns 0..4095, but the touchscreen.h for arduino that is used into the calibration code used 0..1023 so my results are even more twisted out of scale then usually.

prenticedavid, 2019-03-04:
I will ask once more.

Just say what the resistances are.

verbessern, 2019-03-04:
I did not measure, im measuring now. One moment, i will edit this text here.

verbessern, 2019-03-04:
One question, what does XP, YP, XM, YM mean? I.e. what are this abbreviations?

verbessern, 2019-03-04:
I think i get it. The touch works by changing the resistance. Probably one pin is connected to top, one bottom, left, right and the 'magic' is putting electricity on one pin and measuring on the other how much is received. Awesome.

verbessern, 2019-03-04:
Ah...Plus and minus

verbessern, 2019-03-04:
Thank you very much for your help today @prenticedavid.

closed 2019-03-04 by verbessern

spilz87, 2019-08-09:
hello,
I having a similar issue, I want to use the same screen with an esp32, display works well, but touch doesn't work. (it's not a problem on shield, it works well on Arduino uno)

if you succed using the display and touch with esp32, can you give me the wire you use and explain if you change Something on Library for pin mapping ?

thanks in advance.

spilz87, 2019-08-09:
i use this wiring :

> #define LCD_RD  2  //LED
> #define LCD_WR  4
> #define LCD_RS 15  //hard-wired to A2 (GPIO35) 
> #define LCD_CS 33  //hard-wired to A3 (GPIO34)
> #define LCD_RST 32 //hard-wired to A4 (GPIO36)
> 
> #define LCD_D0 12
> #define LCD_D1 13
> #define LCD_D2 26
> #define LCD_D3 25
> #define LCD_D4 17
> #define LCD_D5 16
> #define LCD_D6 27
> #define LCD_D7 14

and 

> int XP = 27, YP = 4, XM = 15, YM = 14;

spilz87, 2019-08-10:
i change touch pin to 

> int XP = 12, YP = 13, XM = 15, YM = 33; 

it seems to work :)

but I really don't understand where pin are defined and I don't understand the logic with Arduino UNO pin (LCD pin vs TOUCH pin)

prenticedavid, 2019-08-11:
Your Shield appears to have
```
LCD_D0 XP
LCD_D1 YM
LCD_RS XM
LCD_CS YP

```
You could have plugged the shield into a Uno and the sketch would diagnose the pins.

An ARM or ESP32 can't diagnose.   So you simply measure the resistance with a DMM.

David.