prenticedavid/MCUFRIEND_kbv #84: MCUFriend with Intel Edison?

jsweet13, 2019-03-21:
I was wondering if an Intel Edison could handle drawing to a 9341 display.

I was wanting to create a game engine and hold pixel color data for a level map in an array, build a buffered screenshot and then print the buffer directly to the screen.

The Edison has a 500mhz dual core Intel Atom, and a 100mhz Quark MCU though I think the Arduino sketches run on the MCU while the CPU handles USB, UART and micro sd card.

Thanks.

prenticedavid, 2019-03-21:
I know nothing about Edison.
I have no intention of buying an Edison.

So I don't think it is a practical proposition.
It would need an "Edison-specific" block in mcufriend_shield.h

David.

jsweet13, 2019-03-21:
OK, I might try to add a quick block to the mcufriend_shield.h for the Edison and think the existing code for Arduino Uno just might work with it as is.

It might be odd since it is a x86-64 board that uses Arduino sketches.

jsweet13, 2019-03-25:
Looking further into it, there is a FCPU error when working with the Edison so it would take a complete rewrite for compatibility. It would be quite the effort to make it work.

closed 2021-08-24 by prenticedavid