prenticedavid/MCUFRIEND_kbv #85: STM32F103

nikjayswal, 2019-03-26:
Pinmapping not working on stm32f103 LCD ID : 0x4747 (HX8347D) 

I want to change pins from this :------

#define LCD_CS A3 // Chip Select goes to Analog 3
#define LCD_CD A2 // Command/Data goes to Analog 2
#define LCD_WR A1 // LCD Write goes to Analog 1
#define LCD_RD A0 // LCD Read goes to Analog 0
#define LCD_RESET A4 // Can alternately just connect to Arduino's reset pin

To this : -----

#define LCD_CS PA0 // Chip Select goes to Analog 3
#define LCD_CD PA1 // Command/Data goes to Analog 2
#define LCD_WR PA2 // LCD Write goes to Analog 1
#define LCD_RD PA3 // LCD Read goes to Analog 0
#define LCD_RESET PA4 // Can alternately just connect to Arduino's reset pin

#define LCD_D0 PB6
#define LCD_D1 PB7
#define LCD_D2 PB8
#define LCD_D3 PB9
#define LCD_D4 PB10
#define LCD_D5 PB11
#define LCD_D6 PB14
#define LCD_D7 PB15

prenticedavid, 2019-03-26:
**I would be happier if you had verified your wiring with LCD_ID_readreg first.**

Here is your SPECIAL

```
#elif defined(USE_NIKJAYSWAL_BLUEPILL) && (defined(ARDUINO_GENERIC_STM32F103C) || defined(ARDUINO_NUCLEO_F103C8))
#warning Using USE_NIKJAYSWAL_BLUEPILL

LCD pins  |D7  |D6  |D5  |D4  |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST|
STM32 pin |PB15|PB14|PB11|PB10|PB9|PB8|PB7|PB6| |PA3|PA2|PA1|PA0|PA4|

#if defined(ARDUINO_NUCLEO_F103C8)   //regular CMSIS libraries
#define REGS(x) x
#define GPIO_INIT()   { RCC->APB2ENR |= RCC_APB2ENR_IOPAEN | RCC_APB2ENR_IOPBEN | RCC_APB2ENR_IOPCEN | RCC_APB2ENR_IOPDEN | RCC_APB2ENR_AFIOEN; \
        AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1;}
#else                                                                  //weird Maple libraries
#define REGS(x) regs->x
#endif

#define WRITE_DELAY { }
#define READ_DELAY  { RD_ACTIVE; }
#define GROUP_MODE(port, reg, mask, val)  {port->REGS(reg) = (port->REGS(reg) & ~(mask)) | ((mask)&(val)); }
#define GP_OUT(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x33333333)
#define GP_INP(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x44444444)
#define PIN_OUTPUT(port, pin) {\
        if (pin < 8) {GP_OUT(port, CRL, 0xF<<((pin)<<2));} \
        else {GP_OUT(port, CRH, 0xF<<((pin&7)<<2));} \
    }
#define PIN_INPUT(port, pin) { \
        if (pin < 8) { GP_INP(port, CRL, 0xF<<((pin)<<2)); } \
        else { GP_INP(port, CRH, 0xF<<((pin&7)<<2)); } \
    }
#define PIN_HIGH(port, pin)   (port)-> REGS(BSRR) = (1<<(pin))
#define PIN_LOW(port, pin)    (port)-> REGS(BSRR) = (1<<((pin)+16))

#define RD_PORT GPIOA
#define RD_PIN  3
#define WR_PORT GPIOA
#define WR_PIN  2
#define CD_PORT GPIOA
#define CD_PIN  1
#define CS_PORT GPIOA
#define CS_PIN  0
#define RESET_PORT GPIOA
#define RESET_PIN  4

// configure macros for the data pins
#define BMASK 0xCFC0
#define write_8(d)    { GPIOB->REGS(BSRR) = BMASK << 16; \
                        GPIOB->REGS(BSRR) = (((d) & 0xC0) << 8); \
                        GPIOB->REGS(BSRR) = (((d) & 0x3F) << 6); \
                       }
#define read_8()      (((GPIOB->REGS(IDR) & (0xC000)) >> 8) | ((GPIOB->REGS(IDR) & (0x0FC0)) >> 6) | ((GPIOB->REGS(IDR) & (3<<6)) >> 4))
//                                PB15-PB14,PB11-PB8                         PB7-PB6
#define setWriteDir() {GP_OUT(GPIOB, CRH, 0xFF00FFFF); GP_OUT(GPIOB, CRL, 0xFF000000); }
#define setReadDir()  {GP_INP(GPIOB, CRH, 0xFF00FFFF); GP_INP(GPIOB, CRL, 0xFF000000); }

#define write8(x)     { write_8(x); WRITE_DELAY; WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; READ_DELAY; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }
```

Untested.

David.

nikjayswal, 2019-03-26:
not working i had add your code and remove stm32code but dosent work i had attached mcufriend_special plz check it

[mcufriend_special.docx](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3009637/mcufriend_special.docx)

prenticedavid, 2019-03-26:
I have no intention of opening a DOCX file.
Either attach the H file or put it in a ZIP.

You have contacted me on GitHub.   This implies that you are able to fork or clone the repository.

I would expect you to be able to edit a file.   And test the LCD_ID_readreg sketch for your defines.

There is no point in attempting a SPECIAL without testing your wiring first.

David.

nikjayswal, 2019-03-26:
LCD_ID_readreg

e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 47 47	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 3F 3F 3F 3F 3F	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 01 01	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 80 80	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03	GAMMA-P
reg(0x00E1) 49 49 49 49 49 49 49 49 49 49 49 49 49 49 49 49	GAMMA-N
reg(0x00EF) 49 49 49 49 49 49	ILI9327
reg(0x00F2) 10 10 10 10 10 10 10 10 10 10 10 10	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

nikjayswal, 2019-03-26:
[mcufriend_special.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3009704/mcufriend_special.zip)

nikjayswal, 2019-03-26:
[mcufriend_special.txt](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3009711/mcufriend_special.txt)

nikjayswal, 2019-03-26:
[mcufriend_special.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3009716/mcufriend_special.zip)

nikjayswal, 2019-03-26:
i had sent you LCD_ID_readreg plz check it

prenticedavid, 2019-03-26:
Thanks for the readreg output.   It looks fine.
Thanks for the H file.   I have edited my SPECIAL to correct the STM macro.

You have never said whether you are using the STM core or Roger's core.
Nor have you said whether it compiled or what warning messages you got.

David.

nikjayswal, 2019-03-26:
used core hardweare from : https://github.com/rogerclarkmelbourne/Arduino_STM32
i had compiled from arduino no error code but display only showing white if i connect pins using your information on this link : https://github.com/prenticedavid/MCUFRIEND_kbv/issues/74      it's working but i want to change pins

Arduino compiled log;

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\HP\AppData\Local\Arduino15\packages -hardware C:\Users\HP\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\HP\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\HP\Documents\Arduino\libraries -fqbn=Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10800 -build-path C:\Users\HP\AppData\Local\Temp\arduino_build_149507 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\HP\Documents\Arduino\libraries\MCUFRIEND_kbv-master\examples\graphictest_kbv\graphictest_kbv.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\HP\AppData\Local\Arduino15\packages -hardware C:\Users\HP\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\HP\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\HP\Documents\Arduino\libraries -fqbn=Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10800 -build-path C:\Users\HP\AppData\Local\Temp\arduino_build_149507 -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\HP\Documents\Arduino\libraries\MCUFRIEND_kbv-master\examples\graphictest_kbv\graphictest_kbv.ino
Using board 'genericSTM32F103C' from platform in folder: C:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1
Using core 'maple' from platform in folder: C:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1
Detecting libraries used...
"C:\Users\HP\AppData\Local\Arduino15\packages\intorobot\tools\arm-none-eabi-gcc\4.9-2015q3/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10800 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c" "C:\Users\HP\AppData\Local\Temp\arduino_build_149507\sketch\graphictest_kbv.ino.cpp" -o "nul"
"C:\Users\HP\AppData\Local\Arduino15\packages\intorobot\tools\arm-none-eabi-gcc\4.9-2015q3/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10800 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/stm32f1/include" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/stm32f1" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/usb/usb_lib" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\variants\generic_stm32f103c" "-IC:\Users\HP\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\SPI\src" "C:\Users\HP\AppData\Local\Temp\arduino_build_149507\sketch\graphictest_kbv.ino.cpp" -o "nul"
[... cut ...]
Wrote address 0x08009f00 (93.03%) 
Wrote address 0x0800a000 (93.62%) 
Wrote address 0x0800a100 (94.20%) 
Wrote address 0x0800a200 (94.79%) 
Wrote address 0x0800a300 (95.37%) 
Wrote address 0x0800a400 (95.96%) 
Wrote address 0x0800a500 (96.54%) 
Wrote address 0x0800a600 (97.13%) 
Wrote address 0x0800a700 (97.71%) 
Wrote address 0x0800a800 (98.30%) 
Wrote address 0x0800a900 (98.88%) 
Wrote address 0x0800aa00 (99.47%) 
Wrote address 0x0800aae8 (100.00%) Done.

Starting execution at address 0x08000000... done.

prenticedavid, 2019-03-26:
Ok.   So you are using Roger's core.   So my edit would not have made any difference.

Please use some common sense when pasting build output.
It is useful to see what -Dxxxxx arguments are passed to the compiler.
It is useful to see ORANGE warning messages.

Your massive paste contains nothing about building MCUFRIEND_kbv library.   It used a "previously compiled" object file.

You can force a library compile by changing a dependent file e.g. add a space to the special.h and then delete the space.

I have no idea what your "Wrote address 0x08000100 (0.59%)" garbage means.   Unless you just want to irritate me.

David.

nikjayswal, 2019-03-27:
" Special.h " doesn't have this type of file it is only contains mcrfriend_special.h
I already had edit it and add your code in between generic stm32f10xx bluepill. i had add comment in " //#define USE_MY_BLUEPILL " and add #define USE_NIKJAYSWAL_BLUEPILL

prenticedavid, 2019-03-27:
From mcufriend_how_to.txt
```

17. If you do not have a standard Uno Shield, you can add a SPECIAL to the mcufriend_special.h
    Edit mcufriend_shield.h:  #define USE_SPECIAL
    Edit mcufriend_special.h: e.g. #define USE_MEGA_16BIT_SHIELD
    If your "special" is write-only,  the library can not read the ID.  It always returns 0xD3D3
	
```
I forced a re-compile of MCUFRIEND_kbv.cpp by making an edit/save of mcufriend_special.h

```
Compiling library "Mcufriend_kbv"
"C:\Users\David Prentice\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10801 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ "-IC:\Users\David Prentice\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.5.11\system/libmaple" "-IC:\Users\David Prentice\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.5.11\system/libmaple/include" "-IC:\Users\David Prentice\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.5.11\system/libmaple/stm32f1/include" "-IC:\Users\David Prentice\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.5.11\system/libmaple/usb/stm32f1" "-IC:\Users\David Prentice\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.5.11\system/libmaple/usb/usb_lib" "-IC:\Users\David Prentice\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.5.11\cores\maple" "-IC:\Users\David Prentice\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.5.11\variants\generic_stm32f103c" "-IC:\Users\David Prentice\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.5.11\libraries\SPI\src" "-IC:\Users\David Prentice\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv" "-IC:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility" "C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\MCUFRIEND_kbv.cpp" -o "C:\Users\DAVIDP~1\AppData\Local\Temp\arduino_build_852227\libraries\Mcufriend_kbv\MCUFRIEND_kbv.cpp.o"
In file included from C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility/mcufriend_shield.h:16:0,

                 from C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\MCUFRIEND_kbv.cpp:35:

C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility/mcufriend_special.h:817:2: warning: #warning Uno Shield on MY BLUEPILL [-Wcpp]

 #warning Uno Shield on MY BLUEPILL

  ^

In file included from C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\MCUFRIEND_kbv.cpp:35:0:

C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility/mcufriend_shield.h:18:2: warning: #warning WE ARE USING A SPECIAL CUSTOM DRIVER [-Wcpp]

 #warning WE ARE USING A SPECIAL CUSTOM DRIVER

  ^

```
Please note the **ORANGE** warning messages:

**#warning Uno Shield on MY BLUEPILL**

**#warning WE ARE USING A SPECIAL CUSTOM DRIVER**

The first warning means that I had enabled #define USE_MY_BLUEPILL in mcufriend_special.h
The second warning means that I had enabled #define USE_SPECIAL in mcufriend_shield.h

Ignore all the opensmart_kbv.cpp warnings,   Opensmart_kbv is only relevant to Uno, Mega and is experimental anyway.

Please understand that I provide a SPECIAL out of the kindness of my heart.
You appear to understand GitHub Issues,  editing files, ...
Please understand that deliberately posting reams of useless messages and DOCX files is not tactful.

I am 99.9% confident that the SPECIAL I posted earlier should work.   If it does not,   please quote **your** ORANGE messages.

David.

nikjayswal, 2019-03-27:
sorry prenticedavid it's my mistake i forgot to uncomment "//#define USE_SPECIAL '' in mcufriend_shield.h its working now thnx a lot  prenticedavid.......

"''''' but my another question is why pin mapping not working in arduino ide why need to modify mcufriend.H and .cpp file why not pin mapping not working directly in arduino ide '''''''''''''''

prenticedavid, 2019-03-27:
The library is intended for Arduino Shields plugged into regular Arduino headers.
The wiring is fixed.   The Arduino user simply has to learn how to mate male pins with female sockets.

You have chosen to use a BluePill.   It has no official Arduino-compatible headers.   However a stm32duino Forum member made an Adapter board for the BluePill.

If you use this "Adapter" wiring,   the BluePill works out of the box.
Just like Uno, Zero, Mega, Due, Nucleo, ...

I am confused by your "experience".   You have chosen to ask on GitHub rather than the Arduino.cc Forum or Roger Clark's stm32duino Forum. 

If you want to use custom wiring,   you can write your own SPECIAL.   Or ask me.
It is pretty easy to change the control pins.    No Arduino library allows you to alter the data bus.

David.

nikjayswal, 2019-03-27:
ok

nikjayswal, 2019-03-27:
ok if i want to write my own custom wiring for pin mapping how can i make ? can you pls provide some tutorial or information.

prenticedavid, 2019-03-27:
Go on.   If you can't understand the RD_PORT, RD_PIN defines,  you should look for another hobby.

Altering the data bus macros: write_8(), read_8(), setWriteDir(), setReadDir() is **very fiddly**
The STM32F103 is painful to select pin modes etc.
Even more painful when you support Roger's weird Maple Core.

David.

closed 2019-03-27 by nikjayswal