prenticedavid/MCUFRIEND_kbv #86: stm32f103 TouchScreen problem

nikjayswal, 2019-03-29:
the y axis value dasen't change the x axis value is ok but y axis position is same in each touch 
[image]

LCD_ID_readreg:
ngle 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 47 47	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 3F 3F 3F 3F 3F	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 01 01	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 80 80	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03	GAMMA-P
reg(0x00E1) 49 49 49 49 49 49 49 49 49 49 49 49 49 49 49 49	GAMMA-N
reg(0x00EF) 49 49 49 49 49 49	ILI9327
reg(0x00F2) 10 10 10 10 10 10 10 10 10 10 10 10	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

My Arduino File :

[stm_test_int.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3023093/stm_test_int.zip)

prenticedavid, 2019-03-29:
First off.   Have you identified the X, Y pins with a DMM ?

The Red Mcufriend boards usually use LCD_D6, LCD_D7, LCD_RS, LCD_WR.
The Blue 2.8 inch Mcufriend boards often use LCD_D0, LCD_D1, LCD_CS, LCD_RS.

Your pins look "normal".    Only you can test with a DMM.

I suggest that you run the "Touch_shield_new.ino" example and edit e.g.

```
#include <MCUFRIEND_kbv.h>
MCUFRIEND_kbv tft;       // hard-wired for UNO shields anyway.
#include <TouchScreen_kbv.h>  

char *name = "Please Calibrate.";  //edit name of shield
const int XP=PB14,XM=PA2,YP=PA1,YM=PB15; //ID=0x9341
const int TS_LEFT=907,TS_RT=136,TS_TOP=942,TS_BOT=139;

TouchScreen_kbv ts = TouchScreen_kbv(XP, YP, XM, YM, 300);
TSPoint_kbv tp;

```

The Calibration directions might be wrong but you should see X, Y values alter.

Seriously,   you should really ask these questions on www.Arduino.cc/Displays

It is always wise to start with a library example.   Everyone is familiar with the code.

It is painful for me to hook up a Mcufriend display to a BluePill.   And even more painful to use Roger's MapleCore.

David.

prenticedavid, 2019-03-31:
Did you get everything working 100% ?

Did you measure X, Y resistance with a multimeter?
Did you try the library example?

If you ask questions on GitHub people expect you to provide intelligent feedback.
If you are not experienced with programming,  it is better to ask questions on Arduino forum.

David.

closed 2019-05-07 by nikjayswal