prenticedavid/MCUFRIEND_kbv #87: STM32L476RG Nucleo screen garbage

patricklaf, 2019-04-03:
Hello David,
I tested an ILI9341 2.8" TFT shield. It works well with several boards: Uno, Mega and also STM32F429ZI Nucleo  (you can suppress the warning about delay).
[image]
But with a STM32L476RG Nucleo I get garbage on screen.
When I run graphictest_kbv I can recognize the figures (rectangles, triangles) but the display is somewhat distorted.
[image]
See the penguin icon garbage.
The basic communication is OK as I can see with serial log and other examples:
> Connect to serial port COM35 at 9600
> Serial took 0ms to start
> ID = 0x9341

I tested with latest Arduino STM core release 1.5.0 and both latest MCUFRIEND_kbv release 2.9.8 and current GIT version.

Patrick

prenticedavid, 2019-04-03:
I tried your shield on a L476RG Nucleo-64 with both v2.9.8 Release and the current "master".
It worked fine.  The Adafruit Report screen says Total time = 2.00secs.  F_CPU=80MHz

Nucleo-144 boards were not supported in the v2.9.8 Release.
I don't have a Nucleo-144 F429 but do have a F767.   This takes 0.68sec @ 216MHz
I do have a Nucleo-64 F446.   The 64 pinout is different to the 144.   However it runs @ 180MHz like your F429.    So I would expect similar timing.

Your screenshot shows the F429 running at 168MHz.   I was expecting 180MHz.
What speed are you running your L476 ?

I have never overclocked ST controllers.    Paul Stoffrogen overclocks Kinetis controllers on his Teensy3.x
I overclock 32MHz Xmegas to 64MHz.     Other people go faster.

My Blue 2.8 inch ILI9341 is several years old.   It has HC245 and AMS1117-3.3 chips on the pcb.    The backlight runs warm.    How does your Blue Shield behave?

David.

patricklaf, 2019-04-03:
I use default clock. It gives me 80 MHz for L476RG and 168 MHz for 429ZI. From STM32CubeMX, it is the default clock configuration.
My shield is totally new. It has LVC245 and AMS1117 3.3. It is not warm when running.

prenticedavid, 2019-04-03:
Woo-hoo.   Mcufriend have finally put the correct buffer chips on the shield.
HC245 are fine for STM32 but they are not input voltage tolerant for 5V Uno.

This is from my build report.
```

Using library SPI at version 1.0 in folder: C:\Users\David\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.4.0\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.4.8 in folder: C:\Users\David\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library MCUFRIEND_kbv at version 2.9.9-Beta in folder: C:\Users\David\Documents\Arduino\libraries\MCUFRIEND_kbv 
"C:\\Users\\David\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\arm-none-eabi-gcc\\6-2017-q2-update/bin/arm-none-eabi-size" -A "C:\\Users\\David\\AppData\\Local\\Temp\\arduino_build_305799/graphictest_kbv.ino.elf"
Sketch uses 46908 bytes (4%) of program storage space. Maximum is 1048576 bytes.
Global variables use 2188 bytes (2%) of dynamic memory, leaving 96116 bytes for local variables. Maximum is 98304 bytes.

```

I see that it is using 1.4.0 on this Laptop.    I will upgrade the STM Core to 1.5.0 and re-build.

David.

patricklaf, 2019-04-03:
I don't know what is going wrong with my L476RG. If someone else can test it.
Basic communication is OK, I can dump register sand so on. Basic operations like set rotation are also working. The only problem seems to be when I send raw data, like we can see on picture with the penguin icon.

prenticedavid, 2019-04-03:
I presume you are using Win10-64.

Please post the times for fillScreen, Lines,  Total from the Adafruit Tests Screen.
And paste the "Using library" and memory usage lines from your build e.g. like my post.

Do you have any other Nucleo boards?
I would expect all my examples to work on the L476.    Please can you report any problems.
The Nucleo-144 boards might have issues with fast GPIO @ 216MHz.

David.

patricklaf, 2019-04-03:
I am using Seven, Eclipse 2019-03 and Sloeber 4.3.1.
I only have L476RG and F429ZI (the shield doesn't fit my L432KC).
I tried to build with Arduino IDE 1.8.9 and then 1.8.8 with no luck. 
It just tell me:
>Carte Nucleo_64 (plateforme stm32, package STM32) est inconnue
>Erreur de compilation pour la carte Nucleo-64
I am pretty happy with the L429ZI, I don't care about yellow... But you are the boss.
I will do the tests.

prenticedavid, 2019-04-03:
I get 
FillScreen 168341
Lines    431168
Total    2.00sec

The Arduino IDE version should make little difference.    The STM Release version will be significant.

I would expect you to get exactly the same times as me.    I would also expect you to be able to install Arduino IDE version  and STM Core painlessly.

The STM Core and IDE work fine.     I would not trust anything to do with Eclipse.

I suggest that you use the IDE that God gave you.     It should work identically on Windoze, Linux or MacOs.
Report any problems from tbe IDE first.    When all the examples work 100% in the IDE you can build again in Sloeber.

There should be several L476 owners out there.   It would be helpful to hear their reports.

David.

patricklaf, 2019-04-04:
Hello David,
I made a fresh installation on Windows 10-64: Arduino 1.8.9, STM Core 1.5.0 and MCUFriend_kbv 2.98.
It works well with L476RG: 1.99s total at 80 MHz.
It does not work with F429ZI, as expected.

I upgraded to MCUFriend_kbv 2.99.
It works well with L476RG: total 2.00s at 80 MHz (curiously, 1 second slower).
It works well with F429ZI: 1.21 total at 168 MHz (the default frequency is 168 not 180).

It seems that the problem is related to Eclipse and / or Sloeber and that the support of the F429ZI module is perfectly functional.
Thank you for your library and your help.
Unless otherwise stated, I will close the subject.

Patrick

prenticedavid, 2019-04-04:
0.01 seconds is 10ms.   Yes,  there is probably some extra code that is slowing the Tests slightly.

I have always found Eclipse HORRIBLE.    It probably works but my life is worth living.

I know nothing about Sloeber.    If it presents the same arguments to the same G++ compiler,   it should produce exactly the same ELF and HEX files for the STM32.

You can look in mcufriend_shield.h
I use several macro symbols to determine which Target and which Core is being used:
```
//####################################### STM32 ############################
// NUCLEO:   ARDUINO_NUCLEO_xxxx from ST Core or ARDUINO_STM_NUCLEO_F103RB from MapleCore
// BLUEPILL: ARDUINO_NUCLEO_F103C8 / ARDUINO_BLUEPILL_F103C8 from ST Core or ARDUINO_GENERIC_STM32F103C from MapleCore
// MAPLE_REV3: n/a from ST Core or ARDUINO_MAPLE_REV3 from MapleCore
// ST Core:   ARDUINO_ARCH_STM32
// MapleCore: __STM32F1__
```

But I also use `defined(STM32L476xx)` and `defined(ARDUINO_NUCLEO_L476RG)`

If Sloeber passes different defines on the command line,   the shield code will build differently.

Please can you paste one single Sloeber command.   Remember to use the [<>] icon to preserve the text.

David.

patricklaf, 2019-04-05:
Hello David,
This is the command line from Sloeber:
>"C:\Eclipse\arduinoPlugin\packages\STM32\tools\arm-none-eabi-gcc\6-2017-q2-update/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:/Users/Patrick/Arduino/NucleoTFT/Release/sketch/build_opt.h" -c -Os -Wall -Wextra -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD "-I" "-IC:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/avr" "-IC:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32" "-IC:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/LL" "-IC:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb" "-IC:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb/hid" "-IC:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino/stm32/usb/cdc" "-IC:\Eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/STM32L4xx_HAL_Driver/Inc/" "-IC:\Eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/STM32L4xx_HAL_Driver/Src/" "-IC:\Eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/STM32L4xx/" "-IC:\Eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\Eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32L4xx -DARDUINO=10802 -DARDUINO_NUCLEO_L476RG -DARDUINO_ARCH_STM32 -DBOARD_NAME="NUCLEO_L476RG"  -DSTM32L476xx -DUSBCON  -DUSBD_VID=0x0483 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT="NUCLEO_L476RG"" -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DDISABLE_GENERIC_SERIALUSB -DHAL_UART_MODULE_ENABLED "-IC:\Eclipse\arduinoPlugin\packages\STM32\tools\CMSIS\5.3.0/CMSIS/Core/Include/" "-IC:\Eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-IC:\Eclipse\/arduinoPlugin/packages/STM32/hardware/stm32/1.5.0/system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/"   -I"C:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\cores\arduino" -I"C:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\variants\NUCLEO_L476RG" -I"C:\Eclipse\arduinoPlugin\libraries\Adafruit_GFX_Library\1.4.8" -I"C:\Eclipse\arduinoPlugin\libraries\MCUFRIEND_kbv\2.9.8\utility" -I"C:\Eclipse\arduinoPlugin\libraries\MCUFRIEND_kbv\2.9.8" -I"C:\Eclipse\arduinoPlugin\packages\STM32\hardware\stm32\1.5.0\libraries\SPI\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"

It seems that  `STM32L476xx` and `ARDUINO_NUCLEO_L476RG` are defined.

Patrick

prenticedavid, 2019-04-05:
The commandline defines look ok.    I will have to study all the other arguments when I get my blood pressure up.

The "build" must be reasonably ok.   Otherwise you would get no graphics at all.

I am fascinated by people's desire to use Sloeber.    The Arduino IDE works pretty well.    You can always use an external editor if you want.    (The Arduino IDE "find" is horrible)

David.

patricklaf, 2021-03-10:
It took a while, but I finally tested the latest versions of MCUFRIEND_kbv (v2.9.9 Release) with the Arduino Core for STM32 (1.9.0).
The L476RG works well with this configuration.
The display is perfect. I close this issue.

closed 2021-03-10 by patricklaf