prenticedavid/MCUFRIEND_kbv #88: The problem with connecting 3.2 'TFT Shield to Arduino Mega2560

kluv58, 2019-04-08:
Dear David!
I again appeal to you with another problem.
I'm trying to connect a regular 3.2 'screen on the ILI9481 (480X320) 16bit controller to the Arduino Mega2560, but I can't do anything. After downloading examples from your library, the screen remains white.
[image]
[image]
[image]
Tell me what could be the problem?
port monitor message:
Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt

tft.readID() finds: ID = 0xD3D3

MCUFRIEND_kbv version: 2.9.9

Probably a write-only Mega2560 Shield
Try to force ID = 0x9481

PORTRAIT is 320 x 480

Run the examples/graphictest_kbv sketch
All colours, text, directions, rotations, scrolls
should work.  If there is a problem,  make notes on paper
Post accurate description of problem to Forum
Or post a link to a video (or photos)

I rely on good information from remote users

prenticedavid, 2019-04-08:
It should work with:
```
USE_SPECIAL
USE_MEGA_16BIT_SHIELD
tft.begin(0x9481);

```
It should also work with UTFT or Bodmer's TFT_HX8357 libraries.

David.

kluv58, 2019-04-08:
I load it like this:

#include "Adafruit_GFX.h"
#include <MCUFRIEND_kbv.h>
MCUFRIEND_kbv tft;

#define  BLACK   0x0000
#define BLUE    0x001F
#define RED     0xF800
#define GREEN   0x07E0
#define CYAN    0x07FF
#define MAGENTA 0xF81F
#define YELLOW  0xFFE0
#define WHITE   0xFFFF
#define ACTIVE  RED
#define PASSIVE GREEN

void setup() {

  tft.begin(0x9481);
  tft.fillScreen(RED);
  
}

void loop() {

}

doesn't work anyway
with UTFT and TFT_HX8357 everything works fine, but does not want to here

prenticedavid, 2019-04-08:
Have you enabled the #defines?

From the how_to file:
```
17. If you do not have a standard Uno Shield, you can add a SPECIAL to the mcufriend_special.h
    Edit mcufriend_shield.h:  #define USE_SPECIAL
    Edit mcufriend_special.h: e.g. #define USE_MEGA_16BIT_SHIELD
    If your "special" is write-only,  the library can not read the ID.  It always returns 0xD3D3

```

Note that mcufriend_shield.h is in the uitils folder e.g. utils/mcufriend_shield.h

David.

kluv58, 2019-04-08:
all is well!!!
The lines you specified were compiled.
thanks for the help.
all blessings to you

closed 2019-04-08 by prenticedavid

kluv58, 2019-04-08:
Dear David.
Now the same question with the same display, but with the Arduino DUE board
In mcufriend_shield.h: USE_SPECIAL definition string is expanded
in mcufriend_special.h: the USE_MEGA_16BIT_SHIELD definition string is disclosed
in mcufriend_special.h: the definition string USE_DUE_16BIT_SHIELD is expanded.
But the display still does not work.

prenticedavid, 2019-04-08:
Your display should plug into the Due headers.   Yes you use USE_MEGA_16BIT_SHIELD

And it should run ok too.    Do **not** use USE_DUE_16BIT_SHIELD.  That is for CTE Due Adapter.

David.

kluv58, 2019-04-08:
Sorry, David, but the issue has been resolved

kluv58, 2019-04-08:
Sorry David, but the issue has already been resolved, it was just in closing this line