prenticedavid/MCUFRIEND_kbv #89: 4.3 'inches 480 * 272 TFT (SSD1963) + Arduino DUE

kluv58, 2019-04-12:
Dear David.
I'm having difficulty connecting a 4.3 'TFT on an SSD1963 controller to an Arduino DUE.
I have:
1. [https://ru.aliexpress.com/item/4-3-480-272-TFT/32920127970.html?spm=a2g0v.search0104.3.8.207665b86mSl1k&transAbTest=ae803_5&ws_ab_test=searchweb0_0%2Csearchweb201602_6_10065_10068_319_317_10696_453_10084_454_10083_10618_10307_10301_537_536_10059_10884_10889_10887_321_322_10915_10103_10914_10911_10910%2Csearchweb201603_6%2CppcSwitch_0&algo_pvid=6d869d2e-d908-4dd6-b2ac-a894aef4448f&algo_expid=6d869d2e-d908-4dd6-b2ac-a894aef4448f-1](url)
2. Arduino DUE
Connection:
#elif defined(__SAM3X8E__) && defined(USE_DUE_16BIT_SHIELD)  //regular CTE shield on DUE
#warning USE_DUE_16BIT_SHIELD
#define USES_16BIT_BUS
// configure macros for the control pins
#define RD_PORT PIOA
#define RD_PIN  15     //D24 Graham
#define WR_PORT PIOD
#define WR_PIN  1      //D26
#define CD_PORT PIOD
#define CD_PIN  0      //D25
#define CS_PORT PIOD
#define CS_PIN  2      //D27
#define RESET_PORT PIOD
#define RESET_PIN  3   //D28
// configure macros for data bus 
// DB0..DB7 on PIOC1..PIOC8(D33...D40),  DB8..DB15 on PIOC12..PIOC19(D51...D44)

#define USE_DUE_16BIT_SHIELD - open
#define USE_SPECIAL  - open     

 After loading, the screen remains white.     
Tell me what I'm doing wrong.
Where is the mistake?

prenticedavid, 2019-04-12:
I use the CTE Adapter shield.   I modify it to enable it to use the RD pin.
If you do not connect the RD pin,  the ID will appear as 0xD3D3 and you have to force `tft.begin(0x1963);`

If you wire all the specified pins from your 40-pin display exactly like CTE,  it should work fine.
It defaults to 5 inch 800x480

David.

kluv58, 2019-04-12:
In the UTFT library everything loads well.
TFT is defined as ITDB43
I load the code:

#include <MCUFRIEND_kbv.h>
MCUFRIEND_kbv tft;

#define  BLACK   0x0000
#define BLUE    0x001F
#define RED     0xF800
#define GREEN   0x07E0
#define CYAN    0x07FF
#define MAGENTA 0xF81F
#define YELLOW  0xFFE0
#define WHITE   0xFFFF
#define ACTIVE  RED
#define PASSIVE GREEN
uint16_t g_identifier;

void setup() {
  tft.begin(0x1963);
  tft.setRotation(3);
  }
void loop() {
  tft.fillScreen(RED);
  tft.setCursor(50, 100);
  tft.setTextColor(WHITE, RED);
  tft.setTextSize(10);
  tft.print("SSD1963");
  delay(1000);
  tft.fillScreen(GREEN);
  tft.setCursor(50, 100);
  tft.setTextColor(WHITE, GREEN);
  tft.setTextSize(10);
  tft.print("SSD1963");
  delay(1000);
  tft.fillScreen(BLUE );
  tft.setCursor(50, 100);
  tft.setTextColor(WHITE, BLUE);
  tft.setTextSize(10);
  tft.print("SSD1963");
  delay(1000);
  tft.fillScreen(MAGENTA);
  tft.setCursor(50, 100);
  tft.setTextColor(WHITE, MAGENTA);
  tft.setTextSize(10);
  tft.print("SSD1963");
  delay(1000);
}

prenticedavid, 2019-04-12:
If it works with UTFT,  please quote which constructor you have used.    And how you have configured `hardware/arm/HW_ARM_defines.h`

David.

kluv58, 2019-04-12:
Do you mean these lines?
// CTE TFT LCD/SD Shield for Arduino Due
// -------------------------------------
// Uncomment the following line if you are using this shield
//#define CTE_DUE_SHIELD 1
//
// For this shield: RS=25, WR=26, CS=27, RST=28
//********************************************************************

// ElecHouse TFT LCD/SD Shield for Arduino Due
// -------------------------------------
// Uncomment the following line if you are using this shield
//#define EHOUSE_DUE_SHIELD 1
//
// For this shield: RS=22, WR=23, CS=31, RST=33
//********************************************************************

so I did not edit them.

kluv58, 2019-04-12:
UTFT wiring:
UTFT myGLCD(ITDB43,38,39,40,41);
DB0...DB7 -  D37...D30, DB8...DB15 - D22...DD29

prenticedavid, 2019-04-12:
In which case I assume that you are using a regular Mega Adapter shield instead of the CTE Due Adapter.

USE_SPECIAL
USE_MEGA_16BIT_SHIELD

Please post a link to the Adapter you are using.    Do not use the link icon.   It is easier to paste the link. 

David.

kluv58, 2019-04-12:
there is no adapter as such, everything is connected via a loop
[image]

prenticedavid, 2019-04-12:
You have clearly used the regular Mega Adapter wiring if UTFT works.

kluv58, 2019-04-12:
Yes, but I want to use your library with this display. what with any wiring
that you advise

kluv58, 2019-04-12:
Perhaps I did not understand your answer.
with the UTFT library I use normal wiring Mega Adapter,
and with your library I use the wiring specified in my post number 1

prenticedavid, 2019-04-12:
I told you to use:
`USE_MEGA_16BIT_SHIELD`

kluv58, 2019-04-12:
I will try, but I have doubts

kluv58, 2019-04-12:
result with your library:
[image]
[image]

result with UTFT library:
[image]
[image]

prenticedavid, 2019-04-12:
USE_SPECIAL
USE_MEGA_16BIT_SHIELD

tft.begin(0x1963);

kluv58, 2019-04-12:
Yes, everything is set:
#define USE_SPECIAL             //check for custom drivers
#define USE_MEGA_16BIT_SHIELD     // 2.14sec Mega2560 Shield
RESET_PIN  9   //D41- installed

error reporting(part of the mistakes):
In file included from C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\utility/mcufriend_shield.h:14:0,
RESET_PIN  9   //D41 - established
                 from C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:33:

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\utility/mcufriend_special.h:392:2: warning: #warning USE_MEGA_16BIT_SHIELD [-Wcpp]

 #warning USE_MEGA_16BIT_SHIELD

  ^

In file included from C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:33:0:

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\utility/mcufriend_shield.h:16:2: warning: #warning WE ARE USING A SPECIAL CUSTOM DRIVER [-Wcpp]

 #warning WE ARE USING A SPECIAL CUSTOM DRIVER

  ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp: In member function 'uint16_t MCUFRIEND_kbv::readReg(uint16_t, int8_t)':

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:141:13: warning: unused variable 'lo' [-Wunused-variable]

     uint8_t lo;

             ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp: In member function 'uint16_t MCUFRIEND_kbv::readID()':

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:173:19: warning: unused variable 'ret2' [-Wunused-variable]

     uint16_t ret, ret2;

                   ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp: In member function 'int16_t MCUFRIEND_kbv::readGRAM(int16_t, int16_t, uint16_t*, int16_t, int16_t)':

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:265:19: warning: variable 'dummy' set but not used [-Wunused-but-set-variable]

     uint16_t ret, dummy, _MR = _MW;

                   ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:267:22: warning: unused variable 'tmp' [-Wunused-variable]

     uint8_t r, g, b, tmp;

                      ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp: In member function 'virtual void MCUFRIEND_kbv::setRotation(uint8_t)':

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:327:29: warning: unused variable 'REV' [-Wunused-variable]

     uint16_t GS, SS_v, ORG, REV = _lcd_rev;

                             ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp: In member function 'void MCUFRIEND_kbv::begin(uint16_t)':

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:1522:30: warning: unused variable 'ST7789_regValues_arcain6' [-Wunused-variable]

         static const uint8_t ST7789_regValues_arcain6[] PROGMEM = {

                              ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:2400:30: warning: unused variable 'ILI9341_regValues_ada' [-Wunused-variable]

         static const uint8_t ILI9341_regValues_ada[] PROGMEM = {        // Adafruit_TFTLCD only works with EXTC=0

                              ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:2480:30: warning: unused variable 'ILI9481_CPT29_regValues' [-Wunused-variable]

         static const uint8_t ILI9481_CPT29_regValues[] PROGMEM = {    // 320x430

                              ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:2492:30: warning: unused variable 'ILI9481_PVI35_regValues' [-Wunused-variable]

         static const uint8_t ILI9481_PVI35_regValues[] PROGMEM = {    // 320x480

                              ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:2504:30: warning: unused variable 'ILI9481_AUO317_regValues' [-Wunused-variable]

         static const uint8_t ILI9481_AUO317_regValues[] PROGMEM = {    // 320x480

                              ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:2515:30: warning: unused variable 'ILI9481_CMO35_regValues' [-Wunused-variable]

         static const uint8_t ILI9481_CMO35_regValues[] PROGMEM = {    // 320480

                              ^

C:\Users\User\Documents\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp:2529:30: warning: unused variable 'ILI9481_RGB_regValues' [-Wunused-variable]

         static const uint8_t ILI9481_RGB_regValues[] PROGMEM = {    // 320x480

prenticedavid, 2019-04-12:
Please don't post reams of messages.
I just wanted you to check the USE_SPECIAL messages. e.g.
```
#warning USE_MEGA_16BIT_SHIELD
#warning WE ARE USING A SPECIAL CUSTOM DRIVER

```

Have you connected LCD_RD pin?
Have you forced tft.begin(0x1963) ?
Are you using the current Beta ?
Or any particular date?

I can test a regular write-only 16BIT_SHIELD on a Due.
I can test a CTE_Adapter on a Due with 800x480 SSD1963

I don't feel like wasting my time if you don't answer my questions

David.

kluv58, 2019-04-12:
RD is connected to D43
tft.begin (0x1963) is installed in the code
Last version installed = 2.9.9-Beta

kluv58, 2019-04-12:
This adapter fits my display?
https://www.elechouse.com/elechouse/index.php?main_page=product_info&cPath=148_211&products_id=2251&zenid=2ushkiisg0sjfu6d9jsfiopqb4

prenticedavid, 2019-04-12:
Yes,   the Elechouse Adapter should work.   I think that I have `USE_ELECHOUSE_DUE_16BIT_SHIELD` in mcufriend_special.h

But the existing `USE_MEGA_16BIT_SHIELD` should work.

If you have _RD is connected to D43_ what does tft.readID() say?

David.

kluv58, 2019-04-12:
port monitor responds:
ID = 0x1963
if you don't have to do USE_ELECHOUSE_DUE_16BIT_SHIELD
but, I think it will help a little. after all, in essence, this is a reassignment of pins, which can be done with loops

kluv58, 2019-04-12:
everything earned !!! It was necessary to remove a comment from
define SUPPORT_1963 in MCUFRIEND_kbv.cpp
but there are still questions on image size and color, but it's for tomorrow

prenticedavid, 2019-04-12:
Ah-ha.   I had just gone to the effort of plugging a RM68140 Mega shield into the Due.
I defined USE_SPECIAL and USE_MEGA_16BIT_SHIELD.   I forced tft.begin(0x6814).

It all works fine.

SUPPORT_1963 is enabled by default.   From MCUFRIEND_kbv.cpp:

`#define SUPPORT_1963              //only works with 16BIT bus anyway
`

You will see that there are several initialisation sequences.   If you select 
```

        table8_ads = SSD1963_480_regValues, table_size = sizeof(SSD1963_480_regValues);
        p16 = (int16_t *) & HEIGHT;
        *p16 = 272;
        p16 = (int16_t *) & WIDTH;
        *p16 = 480;

```

It should support your display.    Please let me know if there is a problem with colours, dimensions, scroll directions.
Note that Rotation=0  is 480x272.   I know it looks like Landscape but this means  **vertical** scroll.

David.

kluv58, 2019-04-12:
Thanks for the comment. It's already night. I'll try to do everything
tomorrow. I will definitely write about the results.

kluv58, 2019-04-13:
Dear David.
After adjusting the lines you specified, everything fell into place.
[image]
Now I will try on the screen 7 '
Thanks for participating and your advice.

prenticedavid, 2019-04-13:
Please look back over this thread.   What was your problem?
1.  Verify wiring by testing with UTFT
2.  Then test with LCD_ID_readreg.    Edit the pin defines to match your wiring.
3.  Then edit USE_SPECIAL,  USE_MEGA_16BIT_SHIELD 
4.  Look for orange warning messages when building library.
5.  Run examples.   Copy-Paste the Serial Terminal messages.
6.   I strongly advise buying a Due Adapter shield.   The CTE/Sainsmart adapter is cheap.
7.  The Elechouse adapter looks convenient for accessing other GPIO pins on the Due..

David.

kluv58, 2019-04-13:
ok i'll follow your advice.
once again THANKS for participation!

kluv58, 2019-04-25:
Good evening, David!
I got an adapter for my screens. Everything worked the first time, but another problem arose: on the screen 7 'the image is printed mirror, that is, from the right - to the left. On the screen 4.3' everything is fine. Tell me what and where to fix?

jsweet13, 2019-04-25:
When that happened to my screen, I decided to paint pixels backwards after creating a 64-bit character font to fix my issue.

I started with X = 319 (the 320th position) and subtracted 1 from X per pixel while keeping the Y coordination the same. 

Doing that painted pixels from left to right, top to bottom. Hope that helps, it took me about 4 hours to calibrate my code around writing to the screen backwards so it looked regular.

prenticedavid, 2019-04-25:
You have a 480x272 SSD1963 screen as shown in your photo from 13 April 2019.
Look for this sequence in MCUFRIEND_kbv.cpp :

```
#if defined(SUPPORT_1963) && USING_16BIT_BUS 
    case 0x1963:
        _lcd_capable = AUTO_READINC | MIPI_DCS_REV1 | READ_NODUMMY | INVERT_SS | INVERT_RGB;

```
And remove the `INVERT_SS` attribute.

There are other ways of doing it.  Which "regValue" array are you using ?
I suspect that your 7 inch screen is 800x480

David.

kluv58, 2019-04-25:
Yes, my screen 7 'has a resolution of 800x480.
I am using an array:
table8_ads = SSD1963_NHD_70_regValues, table_size = sizeof (SSD1963_NHD_70_regValues);

 p16 = (int16_t *) & HEIGHT;
        *p16 = 480;
        p16 = (int16_t *) & WIDTH;
        *p16 = 800;
When deleted:

| INVERT_SS

everything fell into place, i.e. The picture is displayed from left to right.
Then another question arises:
When moving from one display to another, will you need to edit the library every time?

prenticedavid, 2019-04-25:
Yes,  you would need to edit.   Or you could invent an "imaginary" 0x1964 ID.   Put your 480x272 into a different `case 0x1964:` block.

Obviously you would need to force `tft.begin(0x1964);` instead of relying on `tft.readID()`

David.

kluv58, 2019-04-25:
Well, tomorrow I will write down all the necessary changes in a notebook for memory and close the question.
 Thanks again for the help !!!

closed 2019-04-27 by kluv58