prenticedavid/MCUFRIEND_kbv #90: show BMP not uno, function ?

PCov3r, 2019-04-13:
Hello, 
I'm trying to make a weather station with a 3,5" tft display. 
I am using an Arduino mega to display bmp thanks to your sketch "Show BMP not uno". 
I was wondering if I could use this sketch as some kind of function to display a BMP picture once and then display other data. In fact, I would like to "transform" that sketch into a function callable with (path of the bmp, bmp name). Is it possible ?

Not sure if I'm very clear, let me know if you don't understand.
Thank you for your work.

prenticedavid, 2019-04-13:
Yes,  of course you can.

`uint8_t showBMP(char *nm, int x, int y);
`
Just call with the pathname e.g. "/somepath/david.bmp" 

David.

PCov3r, 2019-04-13:
Thank you for answering that quick. 
Have a nice day,
Paul

closed 2021-04-17 by PCov3r