prenticedavid/MCUFRIEND_kbv #91: 2.4" display unsupported

tasper, 2019-04-25:
My card isn't well supported. note the ID = 0x1.
When started by tft.begin(0x9320), runs sometimes, sometimes not; but the screen flipped verticallly (running testcard_kbv on UNOr3)

CARD: 2.4" TFT LCD SHIELD www.mcufriend.com 

[image]
[image]

Results of lcd_id_readreg on UNOr3:
reg(0x0000) 00 01	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 01	Get Power Mode
reg(0x000C) 00 01	Get Pixel Format
reg(0x0061) 00 01	RDID1 HX8347-G
reg(0x0062) 00 01	RDID2 HX8347-G
reg(0x0063) 00 01	RDID3 HX8347-G
reg(0x0064) 00 01	RDID1 HX8347-A
reg(0x0065) 00 01	RDID2 HX8347-A
reg(0x0066) 00 01	RDID3 HX8347-A
reg(0x0067) 00 01	RDID Himax HX8347-A
reg(0x0070) 00 01	Panel Himax HX8347-A
reg(0x00A1) 00 01 00 01 00	RD_DDB SSD1963
reg(0x00B0) 00 01	RGB Interface Signal Control
reg(0x00B4) 00 01	Inversion Control
reg(0x00B6) 00 01 00 01 00	Display Control
reg(0x00B7) 00 01	Entry Mode Set
reg(0x00BF) 00 01 00 01 00 01	ILI9481, HX8357-B
reg(0x00C0) 00 01 00 01 00 01 00 01 00	Panel Control
reg(0x00C8) F8 00 F8 00 F8 00 F8 00 F8 00 F8 00 F8	GAMMA
reg(0x00CC) 00 01	Panel Control
reg(0x00D0) 00 01 00	Power Control
reg(0x00D2) F8 00 F8 00 F8	NVM Read
reg(0x00D3) 00 01 00 01	ILI9341, ILI9488
reg(0x00D4) 00 01 00 01	Novatek ID
reg(0x00DA) 00 01	RDID1
reg(0x00DB) 00 01	RDID2
reg(0x00DC) 00 01	RDID3
reg(0x00E0) 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01	GAMMA-P
reg(0x00E1) 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01	GAMMA-N
reg(0x00EF) 00 01 00 01 00 01	ILI9327
reg(0x00F2) 00 01 00 01 00 01 00 01 00 01 00 01	Adjust Control 2
reg(0x00F6) 00 01 00 01	Interface Control

prenticedavid, 2019-04-25:
The 0x0001 controller is supported by the Beta on GitHub.

It is certainly NOT an ILI9341 type.

```
    case 0x0001:
        _lcd_capable = 0 | REV_SCREEN | INVERT_GS; //no RGB bug. thanks Ivo_Deshev
        goto common_9320;

```

It looks like an ILI9320 style controller.

David.

tasper, 2019-04-25:
> The 0x0001 controller is supported by the Beta on GitHub.
> 
> It is certainly NOT an ILI9341 type.
> 
> ```
>     case 0x0001:
>         _lcd_capable = 0 | REV_SCREEN | INVERT_GS; //no RGB bug. thanks Ivo_Deshev
>         goto common_9320;
> 
> 
> It looks like an ILI9320 style controller.
> 
> David.

Thanks for reply.

Tested on  beta with 0x1, testcard_kbv ok without flicks nor the vert flip, once; hard reset buttons don't do the tft to display. Replugging UNO and button reset sometimes started correctly the screen. 

Due to the intermittent response of the display, I think about some wrong timing at init_table16 calling. Perhaps?

 Luis

baqwas, 2019-07-13:
> I appear to have the very same display, and used 0x9341 for the display identifier before it started working.
>

Where does one specify the _identifier_ in one's code? The picture **matches exactly** the display that I have! I cannot get mine to work at all except to flash momentarily during boot.

artdeell, 2020-04-05:
So what about timing fixes? Can i fix it manually?

prenticedavid, 2020-04-05:
> So what about timing fixes? Can i fix it manually?

My motor car is Blue.   What are you going to do about it?

David.

artdeell, 2020-04-05:
I'm going to add a TFTLCD_DELAYs after all register writes, and then remove them one by one.

prenticedavid, 2020-04-05:
Why don't you say what Shield you have ? e.g. link to Ebay sale page
And what ID is reported ?  e.g. from graphictest_kbv.ino or LCD_ID_readreg.ino
And what target ? e.g. Uno, Due, ...

You would get a sensible reply.   (you are an experienced GitHub member)

David.

artdeell, 2020-04-06:
I have the same shield as the issue's author, and this display reports same data when running LCD_id_readreg

Can't provide link to Ebay page, because I bought it in Arduino store in Moscow

I checked that bit of code from beta library, and pasted it to my current version. Display started to initialize, but only on 2nd attempt and only after board was replugged.

prenticedavid, 2020-04-06:
Thanks for the information.   It is not difficult to quote ID, matching messages, photos etc.

I do not have an 0x0001 display.
I can only rely on reports from users.

I suggest that you:
1. leave IDE. 
2. delete any existing MCUFRIEND_kbv directory.
3. download "master" branch as a ZIP.
4. start IDE
5. install from ZIP
6. run **all** the examples.

7. do **not** edit any library files
8. report back with any problems with examples.   quoting example name and problem. 

9. if (6) works 100% but you have a problem with your own sketch code,  attach your complete sketch. 

David.

artdeell, 2020-04-06:
I ran some examlples, and get some interesting results:
1. graphictest_slim can be started, but NOT in 1st or 2nd attempt.
2. graphictest_kbv sometimes start from 2nd attempt, sometimes not. Color band doesn't scroll (the whole display scrolls instead)
3. aspect_kbv just didn't start, but serial monitor outputs `ID = 0x1`
4. LCD_ID_readreg outputs this: 
`Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 01	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 01 00 01	Manufacturer ID
reg(0x0009) 00 01 00 01 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 01	RDID2 HX8347-G
reg(0x0063) 00 01	RDID3 HX8347-G
reg(0x0064) 00 01	RDID1 HX8347-A
reg(0x0065) 00 01	RDID2 HX8347-A
reg(0x0066) 00 01	RDID3 HX8347-A
reg(0x0067) 00 01	RDID Himax HX8347-A
reg(0x0070) 00 01	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 01	RGB Interface Signal Control
reg(0x00B4) 00 01	Inversion Control
reg(0x00B6) 00 01 00 01 00	Display Control
reg(0x00B7) 00 01	Entry Mode Set
reg(0x00BF) 00 01 00 01 00 01	ILI9481, HX8357-B
reg(0x00C0) F8 00 F8 00 F8 00 F8 00 F8	Panel Control
reg(0x00C8) 00 01 00 01 00 01 00 01 00 01 00 01 00	GAMMA
reg(0x00CC) F8 00	Panel Control
reg(0x00D0) F8 00 F8	Power Control
reg(0x00D2) 00 01 00 01 00	NVM Read
reg(0x00D3) F8 00 F8 00	ILI9341, ILI9488
reg(0x00D4) F8 00 F8 00	Novatek ID
reg(0x00DA) F8 00	RDID1
reg(0x00DB) F8 00	RDID2
reg(0x00DC) F8 00	RDID3
reg(0x00E0) 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01	GAMMA-P
reg(0x00E1) 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01	GAMMA-N
reg(0x00EF) 00 01 00 01 00 01	ILI9327
reg(0x00F2) 00 01 00 01 00 01 00 01 00 01 00 01	Adjust Control 2
reg(0x00F6) 00 01 00 01	Interface Control`
5. GLUE_Demo_320x240 started, but after Arduino reset the display refuses to show anything.
6. scroll_kbv: works, but sometimes shows old elements, jumps back to 3 and shows what it's supposed to show. Video: https://www.youtube.com/watch?v=xlNTI5iGRpQ (sorry, but it's flipped)
6. readpixel_kbv: does anything right. No read-back gamma issues. Blue is 001f ,white is FFFF and black is 0
7. Font_simple: started only after I replugged power and presssed Reset in shield.
8. testcard_kbv: white screen, serial: `TFT ID = 0x1`, screen flickers a bit.
9. diagnose_TFT_support: white screen, serial: 
`Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt

tft.readID() finds: ID = 0x1

MCUFRIEND_kbv version: 2.9.9

PORTRAIT is 240 x 320

Run the examples/graphictest_kbv sketch
All colours, text, directions, rotations, scrolls
should work.  If there is a problem,  make notes on paper
Post accurate description of problem to Forum
Or post a link to a video (or photos)

I rely on good information from remote users`

prenticedavid, 2020-04-06:
(1), (2):  I would expect sketch to run reliably at Reset.  e.g. Power cycle, DTR pulse from Serial Terminal
(2):  ILI9320-style controllers can only scroll whole screen
(3):  it should start.   I don't think that I have run this example in 5 years
(4):  your report is slightly different to the first message.   ILI9320-style controllers use 16-bit interface.   This might cause confusion when reading an odd number of 8-bit values.   But the /CS line should always re-synchronise correctly.
(5): should always start reliably.
(6): I presume you mean everything not anything
(7): should start.  needs correct calibration to detect correct button.
(8): should start
(9): seems to read ID correctly.

The unreliable behaviour suggests that it does not read the ID correctly. 
Do you ever get anything other than ID=0x0001 reported on Serial ?

Or it could be that the LCD_RST pin is badly soldered.   ILI9320-style controllers need the hardware RST pin.     Modern controllers have a Software Reset command and can work without the hardware pin.

What target Arduino are you using?  e.g. Uno, Mega, Due, ...
Is the Shield mating as Nature intended ?

Your feedback is very useful.  Often users stop reporting when it works "once".    Software should work reliably "every time"

David.

artdeell, 2020-04-06:
I tested on Mega, but behaviour is the same as Uno. I tried to create reset pulses on A4 pin manually. Display behaves same.
I'm creating a tiny program, whick will read all registers after initialization and fillScreen(0x0).
I will paste two results here: when display is initialized wrongly (white screen), and when initialized OK(black screen)

artdeell, 2020-04-06:
I modified some code in LCD_ID_readreg. Here's the results when initialization is done (screen is filled black):
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 01	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 01 00	Get Power Mode
reg(0x000C) 01 00	Get Pixel Format
reg(0x0061) 01 00	RDID1 HX8347-G
reg(0x0062) 01 00	RDID2 HX8347-G
reg(0x0063) 01 00	RDID3 HX8347-G
reg(0x0064) 01 00	RDID1 HX8347-A
reg(0x0065) 01 00	RDID2 HX8347-A
reg(0x0066) 01 00	RDID3 HX8347-A
reg(0x0067) 01 00	RDID Himax HX8347-A
reg(0x0070) 01 00	Panel Himax HX8347-A
reg(0x00A1) 01 00 01 00 01	RD_DDB SSD1963
reg(0x00B0) 00 01	RGB Interface Signal Control
reg(0x00B4) 00 01	Inversion Control
reg(0x00B6) 00 01 00 01 00	Display Control
reg(0x00B7) 01 00	Entry Mode Set
reg(0x00BF) 01 00 01 00 01 00	ILI9481, HX8357-B
reg(0x00C0) 01 00 01 00 01 00 01 00 01	Panel Control
reg(0x00C8) 00 01 00 01 00 01 00 01 00 01 00 01 00	GAMMA
reg(0x00CC) 01 00	Panel Control
reg(0x00D0) 01 00 01	Power Control
reg(0x00D2) 00 01 00 01 00	NVM Read
reg(0x00D3) 01 00 01 00	ILI9341, ILI9488
reg(0x00D4) 01 00 01 00	Novatek ID
reg(0x00DA) 01 00	RDID1
reg(0x00DB) 01 00	RDID2
reg(0x00DC) 01 00	RDID3
reg(0x00E0) 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00	GAMMA-P
reg(0x00E1) 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00	GAMMA-N
reg(0x00EF) 01 00 01 00 01 00	ILI9327
reg(0x00F2) 01 00 01 00 01 00 01 00 01 00 01 00	Adjust Control 2
reg(0x00F6) 01 00 01 00	Interface Control

I think this shield is read-only.

artdeell, 2020-04-06:
https://github.com/prenticedavid/MCUFRIEND_kbv/issues/91#issuecomment-609667832

About that tests: examples that witescreen can be ran using some random dark magic.
Also, one time the display started to report `ID = 0xD3D3`, but it was back to 0x1 after  power replug. Of course, i didn't ran readreg in that case. 
It's write-only shield, I think.

artdeell, 2020-04-06:
Register data 0-255:
`1
256
768
4144
0
256
256
307
514
0
0
0
0
0
0
0
6064
7
282
6656
784
784
784
784
784
784
784
784
784
784
784
784
0
0
0
0
1028
1028
1028
1028
0
12
191
0
0
0
0
0
0
1285
4
4
773
6
1799
261
2
1799
1799
1799
1796
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
2055
0
239
0
319
319
319
319
319
319
319
319
319
319
319
319
319
42752
1
1
1
289
289
289
289
289
289
0
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
1
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
16
0
0
3
275
272
272
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
65535
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
0
63232`

prenticedavid, 2020-04-07:
Your "unreliability" symptoms imply that the LCD_RST signal is not connected well.
This could be:
1.  a broken track or bad header joint on the Arduino board.  
2.  a broken track on shield or bad joint on header pin.
3.  bad soldering on HC245 buffer
4.  bad ribbon to screen panel

(1) is unlikely because you get the same behaviour on Uno and Mega.
(2), (3) can be seen from the pcb.   tested with a DMM
(4) requires dismantling the panel from the pcb i.e. double-sided sticky tape.

LCD_ID_readreg is designed to read the controller "reset" values.    A hardware reset pulse is sent before each register is read.
This means that a controller can be identified anywhere in the world with any Arduino board.

Yes,  you can initialise the controller registers with tft.begin(0x0001);
Then read the controller registers (and compare with the written values)

I have no intention of reading decimal values. 
The result is only useful if you report register and hex value. e.g.
```
    for(int i = 0; i < 256; i++) {
        uint16_t val = tft.readReg(i, 0);
        if (val == 0) continue;
        Serial.print("reg(0x");
        Serial.print(i, HEX);
        Serial.print(") = 0x");
        Serial.println(val, HEX);
    }
```

artdeell, 2020-04-07:
When display is white: `reg(0x0) = 0x1
reg(0x1) = 0x1
reg(0x2) = 0x1
reg(0x3) = 0x1
reg(0x4) = 0x1
reg(0x5) = 0x1
reg(0x6) = 0x1
reg(0x7) = 0x1
reg(0x8) = 0x1
reg(0x9) = 0x1
reg(0xA) = 0x1
reg(0xB) = 0x1
reg(0xC) = 0x1
reg(0xD) = 0x1
reg(0xE) = 0x1
reg(0xF) = 0x1
reg(0x10) = 0x1
reg(0x11) = 0x1
reg(0x12) = 0x1
reg(0x13) = 0x1
reg(0x14) = 0x1
reg(0x15) = 0x1
reg(0x16) = 0x1
reg(0x17) = 0x1
reg(0x18) = 0x1
reg(0x19) = 0x1
reg(0x1A) = 0x1
reg(0x1B) = 0x1
reg(0x1C) = 0x1
reg(0x1D) = 0x1
reg(0x1E) = 0x1
reg(0x1F) = 0x1
reg(0x20) = 0x1
reg(0x21) = 0x1
reg(0x22) = 0x1
reg(0x23) = 0x1
reg(0x24) = 0x1
reg(0x25) = 0x1
reg(0x26) = 0x1
reg(0x27) = 0x1
reg(0x28) = 0x1
reg(0x29) = 0x1
reg(0x2A) = 0x1
reg(0x2B) = 0x1
reg(0x2C) = 0x1
reg(0x2D) = 0x1
reg(0x2E) = 0x1
reg(0x2F) = 0x1
reg(0x30) = 0x1
reg(0x31) = 0x1
reg(0x32) = 0x1
reg(0x33) = 0x1
reg(0x34) = 0x1
reg(0x35) = 0x1
reg(0x36) = 0x1
reg(0x37) = 0x1
reg(0x38) = 0x1
reg(0x39) = 0x1
reg(0x3A) = 0x1
reg(0x3B) = 0x1
reg(0x3C) = 0x1
reg(0x3D) = 0x1
reg(0x3E) = 0x1
reg(0x3F) = 0x1
reg(0x40) = 0x1
reg(0x41) = 0x1
reg(0x42) = 0x1
reg(0x43) = 0x1
reg(0x44) = 0x1
reg(0x45) = 0x1
reg(0x46) = 0x1
reg(0x47) = 0x1
reg(0x48) = 0x1
reg(0x49) = 0x1
reg(0x4A) = 0x1
reg(0x4B) = 0x1
reg(0x4C) = 0x1
reg(0x4D) = 0x1
reg(0x4E) = 0x1
reg(0x4F) = 0x1
reg(0x50) = 0x1
reg(0x51) = 0x1
reg(0x52) = 0x1
reg(0x53) = 0x1
reg(0x54) = 0x1
reg(0x55) = 0x1
reg(0x56) = 0x1
reg(0x57) = 0x1
reg(0x58) = 0x1
reg(0x59) = 0x1
reg(0x5A) = 0x1
reg(0x5B) = 0x1
reg(0x5C) = 0x1
reg(0x5D) = 0x1
reg(0x5E) = 0x1
reg(0x5F) = 0x1
reg(0x60) = 0x1
reg(0x61) = 0x1
reg(0x62) = 0x1
reg(0x63) = 0x1
reg(0x64) = 0x1
reg(0x65) = 0x1
reg(0x66) = 0x1
reg(0x67) = 0x1
reg(0x68) = 0x1
reg(0x69) = 0x1
reg(0x6A) = 0x1
reg(0x6B) = 0x1
reg(0x6C) = 0x1
reg(0x6D) = 0x1
reg(0x6E) = 0x1
reg(0x6F) = 0x1
reg(0x70) = 0x1
reg(0x71) = 0x1
reg(0x72) = 0x1
reg(0x73) = 0x1
reg(0x74) = 0x1
reg(0x75) = 0x1
reg(0x76) = 0x1
reg(0x77) = 0x1
reg(0x78) = 0x1
reg(0x79) = 0x1
reg(0x7A) = 0x1
reg(0x7B) = 0x1
reg(0x7C) = 0x1
reg(0x7D) = 0x1
reg(0x7E) = 0x1
reg(0x7F) = 0x1
reg(0x80) = 0x1
reg(0x81) = 0x1
reg(0x82) = 0x1
reg(0x83) = 0x1
reg(0x84) = 0x1
reg(0x85) = 0x1
reg(0x86) = 0x1
reg(0x87) = 0x1
reg(0x88) = 0x1
reg(0x89) = 0x1
reg(0x8A) = 0x1
reg(0x8B) = 0x1
reg(0x8C) = 0x1
reg(0x8D) = 0x1
reg(0x8E) = 0x1
reg(0x8F) = 0x1
reg(0x90) = 0x1
reg(0x91) = 0x1
reg(0x92) = 0x1
reg(0x93) = 0x1
reg(0x94) = 0x1
reg(0x95) = 0x1
reg(0x96) = 0x1
reg(0x97) = 0x1
reg(0x98) = 0x1
reg(0x99) = 0x1
reg(0x9A) = 0x1
reg(0x9B) = 0x1
reg(0x9C) = 0x1
reg(0x9D) = 0x1
reg(0x9E) = 0x1
reg(0x9F) = 0x1
reg(0xA0) = 0x1
reg(0xA1) = 0x1
reg(0xA2) = 0x1
reg(0xA3) = 0x1
reg(0xA4) = 0x1
reg(0xA5) = 0x1
reg(0xA6) = 0x1
reg(0xA7) = 0x1
reg(0xA8) = 0x1
reg(0xA9) = 0x1
reg(0xAA) = 0x1
reg(0xAB) = 0x1
reg(0xAC) = 0x1
reg(0xAD) = 0x1
reg(0xAE) = 0x1
reg(0xAF) = 0x1
reg(0xB0) = 0x1
reg(0xB1) = 0x1
reg(0xB2) = 0x1
reg(0xB3) = 0x1
reg(0xB4) = 0x1
reg(0xB5) = 0x1
reg(0xB6) = 0x1
reg(0xB7) = 0x1
reg(0xB8) = 0x1
reg(0xB9) = 0x1
reg(0xBA) = 0x1
reg(0xBB) = 0x1
reg(0xBC) = 0x1
reg(0xBD) = 0x1
reg(0xBE) = 0x1
reg(0xBF) = 0x1
reg(0xC0) = 0x1
reg(0xC1) = 0x1
reg(0xC2) = 0x1
reg(0xC3) = 0x1
reg(0xC4) = 0x1
reg(0xC5) = 0x1
reg(0xC6) = 0x1
reg(0xC7) = 0x1
reg(0xC8) = 0x1
reg(0xC9) = 0x1
reg(0xCA) = 0x1
reg(0xCB) = 0x1
reg(0xCC) = 0x1
reg(0xCD) = 0x1
reg(0xCE) = 0x1
reg(0xCF) = 0x1
reg(0xD0) = 0x1
reg(0xD1) = 0x1
reg(0xD2) = 0x1
reg(0xD3) = 0x1
reg(0xD4) = 0x1
reg(0xD5) = 0x1
reg(0xD6) = 0x1
reg(0xD7) = 0x1
reg(0xD8) = 0x1
reg(0xD9) = 0x1
reg(0xDA) = 0x1
reg(0xDB) = 0x1
reg(0xDC) = 0x1
reg(0xDD) = 0x1
reg(0xDE) = 0x1
reg(0xDF) = 0x1
reg(0xE0) = 0x1
reg(0xE1) = 0x1
reg(0xE2) = 0x1
reg(0xE3) = 0x1
reg(0xE4) = 0x1
reg(0xE5) = 0x1
reg(0xE6) = 0x1
reg(0xE7) = 0x1
reg(0xE8) = 0x1
reg(0xE9) = 0x1
reg(0xEA) = 0x1
reg(0xEB) = 0x1
reg(0xEC) = 0x1
reg(0xED) = 0x1
reg(0xEE) = 0x1
reg(0xEF) = 0x1
reg(0xF0) = 0x1
reg(0xF1) = 0x1
reg(0xF2) = 0x1
reg(0xF3) = 0x1
reg(0xF4) = 0x1
reg(0xF5) = 0x1
reg(0xF6) = 0x1
reg(0xF7) = 0x1
reg(0xF8) = 0x1
reg(0xF9) = 0x1
reg(0xFA) = 0x1
reg(0xFB) = 0x1
reg(0xFC) = 0x1
reg(0xFD) = 0x1
reg(0xFE) = 0x1
reg(0xFF) = 0x1`

artdeell, 2020-04-07:
When display is initialized and filled black: `reg(0x0) = 0x1
reg(0x1) = 0x100
reg(0x2) = 0x300
reg(0x3) = 0x1030
reg(0x5) = 0x100
reg(0x6) = 0x100
reg(0x7) = 0x133
reg(0x8) = 0x202
reg(0x10) = 0x17B0
reg(0x11) = 0x7
reg(0x12) = 0x11A
reg(0x13) = 0x1A00
reg(0x14) = 0x310
reg(0x15) = 0x310
reg(0x16) = 0x310
reg(0x17) = 0x310
reg(0x18) = 0x310
reg(0x19) = 0x310
reg(0x1A) = 0x310
reg(0x1B) = 0x310
reg(0x1C) = 0x310
reg(0x1D) = 0x310
reg(0x1E) = 0x310
reg(0x1F) = 0x310
reg(0x24) = 0x404
reg(0x25) = 0x404
reg(0x26) = 0x404
reg(0x27) = 0x404
reg(0x29) = 0xC
reg(0x31) = 0x505
reg(0x32) = 0x4
reg(0x33) = 0x4
reg(0x34) = 0x105
reg(0x35) = 0x6
reg(0x36) = 0x707
reg(0x37) = 0x105
reg(0x38) = 0x2
reg(0x39) = 0x707
reg(0x3A) = 0x707
reg(0x3B) = 0x707
reg(0x3C) = 0x704
reg(0x3D) = 0x807
reg(0x3E) = 0x807
reg(0x3F) = 0x807
reg(0x40) = 0x807
reg(0x41) = 0x807
reg(0x42) = 0x807
reg(0x43) = 0x807
reg(0x44) = 0x807
reg(0x45) = 0x807
reg(0x46) = 0x807
reg(0x47) = 0x807
reg(0x48) = 0x807
reg(0x49) = 0x807
reg(0x4A) = 0x807
reg(0x4B) = 0x807
reg(0x4C) = 0x807
reg(0x4D) = 0x807
reg(0x4E) = 0x807
reg(0x4F) = 0x807
reg(0x51) = 0xEF
reg(0x53) = 0x13F
reg(0x54) = 0x13F
reg(0x55) = 0x13F
reg(0x56) = 0x13F
reg(0x57) = 0x13F
reg(0x58) = 0x13F
reg(0x59) = 0x13F
reg(0x5A) = 0x13F
reg(0x5B) = 0x13F
reg(0x5C) = 0x13F
reg(0x5D) = 0x13F
reg(0x5E) = 0x13F
reg(0x5F) = 0x13F
reg(0x60) = 0xA700
reg(0x61) = 0x1
reg(0x62) = 0x1
reg(0x63) = 0x1
reg(0x64) = 0x120
reg(0x65) = 0x120
reg(0x66) = 0x120
reg(0x67) = 0x120
reg(0x68) = 0x120
reg(0x69) = 0x120
reg(0x6B) = 0x1
reg(0x6C) = 0x1
reg(0x6D) = 0x1
reg(0x6E) = 0x1
reg(0x6F) = 0x1
reg(0x70) = 0x1
reg(0x71) = 0x1
reg(0x72) = 0x1
reg(0x73) = 0x1
reg(0x74) = 0x1
reg(0x75) = 0x1
reg(0x76) = 0x1
reg(0x77) = 0x1
reg(0x78) = 0x1
reg(0x79) = 0x1
reg(0x7A) = 0x1
reg(0x7B) = 0x1
reg(0x7C) = 0x1
reg(0x7D) = 0x1
reg(0x7E) = 0x1
reg(0x7F) = 0x1
reg(0x90) = 0x10
reg(0x93) = 0x3
reg(0x94) = 0x113
reg(0x95) = 0x110
reg(0x96) = 0x110
reg(0xFF) = 0xF700`

prenticedavid, 2020-04-07:
There is a distinct absence of newlines.
But the initialised registers appear to hold the values that I would expect.

This is academic since you have not checked for intermittent LCD_RST connection.

David.

artdeell, 2020-04-08:
I checked LCD_RST soldering, and anything is okay as I can see.

artdeell, 2020-04-08:
Photo:
[image]

prenticedavid, 2020-04-08:
Have you checked for broken tracks with a DMM ?

You photo "looks" ok.  I presume that the HC245 soldering is ok.
Only you can see the quality of soldering, broken tracks,  solder wisps, ...

I can only suggest that you go back to the Moscow shop and ask for a replacement.

I suspect that the LCD_RST signal does not always get to the controller (i.e. intermittent break in pcb trace / joint)
You can always force the ID i.e. ` tft.begin(0x0001);`
This will attempt to initialise in software.

An intermittent short in LCD_RST pcb trace would make the running program die 

Of course it could be an intermittent break/short in RD, WR, RS, CS pcb trace and not the RST signal.

David.

rovind, 2021-06-29:
Hi,
I had bought this Shield quite a while back and since then I have tried a lot with different drives from MCUFRIEND_kbv.
I am not even sure if it is working or broken as I have never seen anything being displayed, other than just the WHITE Screen.
I am just a beginner and trying to use this in some display projects.

   
[image]

To Test, I have run the LCD_ID_readreg.ino program and this is the output I am getting:

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 01 14	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 1E 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

I have noticed "0154" on shield. but the ID shown in the program out is "01 14" which is confusing.
Also I could not find any support or help for this ID on the forum.

Please can you help/guide me in the right direction.

Regards,

Rovin D.

prenticedavid, 2021-06-29:
I suspect that it is a Samsung S6D0154 but your wiring is not so good.   Bit 6 is reading 0 instead of 1.
I hope that the Shield is plugged into a Uno as Nature intended.   And not flying wires.

There is a Samsung S6D0114 but it is a 132x176 controller.   Which is unlikely to be on a 2.4 inch screen.
Anyway,  I don't support the S6D114.

I suggest that you check your wiring.  `SUPPORT_0154` is enabled by default.  (I used to have a 0x0154 screen)

David.

rovind, 2021-06-30:
Thank you David for the quick response and for the clarification.

The shield fits nicely on top of the Arduino and so I have not done any wiring, but plugged the shield on top of the UNO.
Can you let me know what wiring you referring to when you suggest - check wiring? Do you mean, the shield's track.

Regards,

Rovin D.

[image]
[image]

prenticedavid, 2021-06-30:
> The shield fits nicely on top of the Arduino and so I have not done any wiring, but plugged the shield on top of the UNO.
> Can you let me know what wiring you referring to when you suggest - check wiring? Do you mean, the shield's track.
You would be amazed by how many people do not plug the shield into the Arduino.

Shield is generally well soldered.   Arduino headers are well soldered.  
Swap with a different Arduino if you think that a port pin has been damaged.
The current Mcufriend pcb is wrong.   The backlight is powered from the Uno 3.3V pin.   Does the screen feel warm / hot ?   The 3.3V regulator will be damaged on the 3.5 inch shields.  Less likely on a 2.4 inch shield.

I can only suggest forcing `tft.begin(0x0154);` but if there is a damaged pin(s) the initialisation will not work.

You can run LCD_ID_readnew.ino displaying the `read_9320("ILI9320") `report.
It might reveal some more S6D0154 registers (S6D0154 has 9320-style registers but some have different addresses)

David.

rovind, 2021-06-30:
Thanks David .... Really appreciate the inputs your are providing...:)

I kept the shield powered ON for around 5 mins and it was a bit warm, but not getting hot.
Just as you suggested, I have also changed the UNO and tried the below 2 program (on each of the UNO), the results are same as given below.

I tried forcing  `tft.begin(0x0154);` in the diagnose_TFT_support.ino, just before the `uint16_t ID = tft.readID();` 
The output is still the same...  ID = 0x114

And finally, I ran the LCD_ID_readnew.ino displaying the read_9320("ILI9320")  for report and the output of this is  :

⸮Read Special Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 01 14	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 1E 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0030) 00 00 00 00 00	PTLAR
reg(0x0033) 00 00 00 00 00 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B3) 00 00 00 00 00	Frame Memory
reg(0x00B4) 00 00	Frame Mode
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C1) 00 00 00 00	Display Timing
reg(0x00C5) 00 00	Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00 00	Power Control
reg(0x00D1) 00 00 00 00	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
ILI9320
reg(0x0000) 01 14	Device Code Read
reg(0x0001) 00 E8	Drive Output Control
reg(0x0002) 01 00	LCD Waveform Control
reg(0x0003) C0 F0	Entry Mode
reg(0x0007) 00 00	Display Control 1
reg(0x0008) 08 F8	Display Control 2
reg(0x0009) 00 1E	Display Control 3
reg(0x000A) 00 00	Display Control 4
reg(0x0010) 22 00	Power Control 1
reg(0x0011) 00 00	Power Control 2
reg(0x0012) 00 00	Power Control 3
reg(0x0013) 00 00	Power Control 4
reg(0x0014) 00 00	Power Control 5
reg(0x0015) 00 00	Power Control 6
reg(0x0029) 00 00	Power Control 7
reg(0x002B) 00 00	Frame Rate

Does this help in anyway? I am stating to wonder is this shield busted ...:(

Regards,

Rovin D.

prenticedavid, 2021-06-30:
```
ILI9320
reg(0x0000) 01 14 Device Code Read
reg(0x0001) 00 E8 Drive Output Control
reg(0x0002) 01 00 LCD Waveform Control
reg(0x0003) C0 F0 Entry Mode
reg(0x0007) 00 00 Display Control 1
reg(0x0008) 08 F8 Display Control 2
reg(0x0009) 00 1E Display Control 3
reg(0x000A) 00 00 Display Control 4
reg(0x0010) 22 00 Power Control 1
reg(0x0011) 00 00 Power Control 2
reg(0x0012) 00 00 Power Control 3
reg(0x0013) 00 00 Power Control 4
reg(0x0014) 00 00 Power Control 5
reg(0x0015) 00 00 Power Control 6
reg(0x0029) 00 00 Power Control 7
reg(0x002B) 00 00 Frame Rate
```

Well this shows thatthere is nothing wrong with bit6.  There are several registers with bit6 set.

So perhaps you have a genuine S6D0114.   I could compare those values with the S6D0114 datasheet if I am feeling enterprising.

On the other hand,  you could just go and buy another cheap display.

I could send you a specific read_0114() function.   But regs 01, 03, 09 have bits that are not in the datasheet.   And you would only have a 132x176 display at best.   The panel is probably narrower than 132.

David.

rovind, 2021-06-30:
Thanks David for the reply and also for letting me know that I really have a shield which is not usually normal.
If I get a solution, I would like to try and see what this display would look like in 132x176 ... though narrower.
Guess time to get a new display anyway...:(

But said that, I really appreciate the support/inputs provided by you. 

Regards

Rovin D.

closed 2021-08-24 by prenticedavid